dune3d/dune3d #1: Linker failure when building

kozross, 2023-10-23:
When I tried to build, I got this link error at the end:

```
[139/139] Linking target dune3d
FAILED: dune3d 
c++  -o dune3d dune3d.p/meson-generated_.._version_gen.cpp.o dune3d.p/meson-generated_.._resources.c.o dune3d.p/meson-generated_.._icon_texture_map.cpp.o dune3d.p/src_main.cpp.o dune3d.p/src_dune3d_application.cpp.o dune3d.p/src_dune3d_appwindow.cpp.o dune3d.p/src_canvas_canvas.cpp.o dune3d.p/src_canvas_gl_util.cpp.o dune3d.p/src_canvas_base_renderer.cpp.o dune3d.p/src_canvas_background_renderer.cpp.o dune3d.p/src_canvas_face_renderer.cpp.o dune3d.p/src_canvas_point_renderer.cpp.o dune3d.p/src_canvas_line_renderer.cpp.o dune3d.p/src_canvas_glyph_renderer.cpp.o dune3d.p/src_canvas_bitmap_font_util.cpp.o dune3d.p/src_canvas_bitmap_font_bitmap_font_desc.c.o dune3d.p/src_canvas_bitmap_font_bitmap_font_img.c.o dune3d.p/src_canvas_icon_renderer.cpp.o dune3d.p/src_canvas_appearance.cpp.o dune3d.p/src_import_step_step_importer.cpp.o dune3d.p/src_import_step_step_import_manager.cpp.o dune3d.p/src_util_uuid.cpp.o dune3d.p/src_document_document.cpp.o dune3d.p/src_document_entity.cpp.o dune3d.p/src_document_entity_and_point.cpp.o dune3d.p/src_document_entity_line3d.cpp.o dune3d.p/src_document_entity_line2d.cpp.o dune3d.p/src_document_entity_arc2d.cpp.o dune3d.p/src_document_entity_circle2d.cpp.o dune3d.p/src_document_entity_circle3d.cpp.o dune3d.p/src_document_entity_arc3d.cpp.o dune3d.p/src_document_entity_workplane.cpp.o dune3d.p/src_document_entity_step.cpp.o dune3d.p/src_document_constraint.cpp.o dune3d.p/src_document_constraint_points_coincident.cpp.o dune3d.p/src_document_constraint_parallel.cpp.o dune3d.p/src_document_constraint_equal_length.cpp.o dune3d.p/src_document_constraint_equal_radius.cpp.o dune3d.p/src_document_constraint_point_on_line.cpp.o dune3d.p/src_document_constraint_point_on_circle.cpp.o dune3d.p/src_document_constraint_same_orientation.cpp.o dune3d.p/src_document_constraint_hv.cpp.o dune3d.p/src_document_constraint_point_distance.cpp.o dune3d.p/src_document_constraint_point_distance_hv.cpp.o dune3d.p/src_document_constraint_workplane_normal.cpp.o dune3d.p/src_document_constraint_midpoint.cpp.o dune3d.p/src_document_constraint_diameter_radius.cpp.o dune3d.p/src_document_constraint_arc_line_tangent.cpp.o dune3d.p/src_document_constraint_arc_arc_tangent.cpp.o dune3d.p/src_document_constraint_line_points_perpendicular.cpp.o dune3d.p/src_document_solid_model.cpp.o dune3d.p/src_document_solid_model_occ.cpp.o dune3d.p/src_document_group.cpp.o dune3d.p/src_document_group_sketch.cpp.o dune3d.p/src_document_group_reference.cpp.o dune3d.p/src_document_group_extrude.cpp.o dune3d.p/src_document_group_local_operation.cpp.o dune3d.p/src_document_group_fillet.cpp.o dune3d.p/src_document_group_chamfer.cpp.o dune3d.p/src_document_group_lathe.cpp.o dune3d.p/src_document_group_sweep.cpp.o dune3d.p/src_system_system.cpp.o dune3d.p/src_logger_logger.cpp.o dune3d.p/src_render_renderer.cpp.o dune3d.p/src_util_util.cpp.o dune3d.p/src_util_fs_util.cpp.o dune3d.p/src_util_json_util.cpp.o dune3d.p/src_util_glm_util.cpp.o dune3d.p/src_util_selection_util.cpp.o dune3d.p/src_util_file_version.cpp.o dune3d.p/src_core_core.cpp.o dune3d.p/src_core_tool.cpp.o dune3d.p/src_core_create_tool.cpp.o dune3d.p/src_core_tools_tool_common.cpp.o dune3d.p/src_core_tools_tool_draw_line_3d.cpp.o dune3d.p/src_core_tools_tool_draw_line_2d.cpp.o dune3d.p/src_core_tools_tool_draw_arc_2d.cpp.o dune3d.p/src_core_tools_tool_draw_circle_2d.cpp.o dune3d.p/src_core_tools_tool_delete.cpp.o dune3d.p/src_core_tools_tool_move.cpp.o dune3d.p/src_core_tools_tool_constrain_coincident.cpp.o dune3d.p/src_core_tools_tool_constrain_hv.cpp.o dune3d.p/src_core_tools_tool_constrain_distance.cpp.o dune3d.p/src_core_tools_tool_constrain_same_orientation.cpp.o dune3d.p/src_core_tools_tool_constrain_parallel.cpp.o dune3d.p/src_core_tools_tool_constrain_workplane_normal.cpp.o dune3d.p/src_core_tools_tool_constrain_midpoint.cpp.o dune3d.p/src_core_tools_tool_constrain_equal_length.cpp.o dune3d.p/src_core_tools_tool_constrain_equal_radius.cpp.o dune3d.p/src_core_tools_tool_constrain_diameter_radius.cpp.o dune3d.p/src_core_tools_tool_constrain_perpendicular.cpp.o dune3d.p/src_core_tools_tool_enter_datum.cpp.o dune3d.p/src_core_tools_tool_draw_workplane.cpp.o dune3d.p/src_core_tools_tool_add_anchor.cpp.o dune3d.p/src_core_tools_tool_toggle_construction.cpp.o dune3d.p/src_core_tools_tool_import_step.cpp.o dune3d.p/src_core_tools_tool_select_edges.cpp.o dune3d.p/src_util_history_manager.cpp.o dune3d.p/src_action_action.cpp.o dune3d.p/src_action_action_catalog.cpp.o dune3d.p/src_in_tool_action_in_tool_action_catalog.cpp.o dune3d.p/src_util_str_util.cpp.o dune3d.p/src_util_lut.cpp.o dune3d.p/src_preferences_preferences.cpp.o dune3d.p/src_util_action_label.cpp.o dune3d.p/src_tool_popover.cpp.o dune3d.p/src_workspace_browser.cpp.o dune3d.p/src_constraints_box.cpp.o dune3d.p/src_document_view.cpp.o dune3d.p/src_group_editor.cpp.o dune3d.p/src_dialogs_dialogs.cpp.o dune3d.p/src_dialogs_enter_datum_window.cpp.o dune3d.p/src_dialogs_tool_window.cpp.o dune3d.p/src_widgets_spin_button_dim.cpp.o dune3d.p/src_util_gtk_util.cpp.o dune3d.p/src_preferences_preferences_window.cpp.o dune3d.p/src_preferences_preferences_window_keys.cpp.o dune3d.p/src_preferences_preferences_window_canvas.cpp.o dune3d.p/src_widgets_capture_dialog.cpp.o dune3d.p/src_widgets_axes_lollipop.cpp.o dune3d.p/src_widgets_about_dialog.cpp.o dune3d.p/src_util_version.cpp.o -Wl,--as-needed -Wl,--no-undefined -Wl,--start-group 3rd_party/solvespace/libsolvespace.a 3rd_party/Clipper2Lib/libclipper.a /usr/lib/libgtk-4.so /usr/lib/libpangocairo-1.0.so /usr/lib/libpango-1.0.so /usr/lib/libharfbuzz.so /usr/lib/libgdk_pixbuf-2.0.so /usr/lib/libcairo-gobject.so /usr/lib/libcairo.so /usr/lib/libgraphene-1.0.so /usr/lib/libgio-2.0.so /usr/lib/libgobject-2.0.so /usr/lib/libglib-2.0.so /usr/lib/libgtkmm-4.0.so /usr/lib/libpangomm-2.48.so /usr/lib/libgiomm-2.68.so /usr/lib/libglibmm-2.68.so /usr/lib/libcairomm-1.16.so /usr/lib/libsigc-3.0.so /usr/lib/libepoxy.so -lTKSTEP -lTKernel -lTKXCAF -lTKXSBase -lTKBRep -lTKCDF -lTKXDESTEP -lTKLCAF -lTKMath -lTKMesh -lTKTopAlgo -lTKPrim -lTKBO -lTKShHealing -lTKBRep -lTKG3d -lTKGeomBase -lTKHLR /usr/lib/libuuid.so /usr/lib/libmimalloc.so -Wl,--end-group
/usr/bin/ld: dune3d.p/src_document_solid_model.cpp.o: warning: relocation against `_ZTV16ChFi3d_ChBuilder' in read-only section `.text._ZN16ChFi3d_ChBuilderD2Ev[_ZN16ChFi3d_ChBuilderD5Ev]'
/usr/bin/ld: dune3d.p/src_document_solid_model.cpp.o: in function `dune3d::SolidModel::create(dune3d::Document const&, dune3d::GroupFillet const&)':
/home/koz/dune3d/build/../src/document/solid_model.cpp:732:(.text+0x3188): undefined reference to `BRepFilletAPI_MakeFillet::BRepFilletAPI_MakeFillet(TopoDS_Shape const&, ChFi3d_FilletShape)'
/usr/bin/ld: /home/koz/dune3d/build/../src/document/solid_model.cpp:741:(.text+0x3249): undefined reference to `BRepFilletAPI_MakeFillet::Add(double, TopoDS_Edge const&)'
/usr/bin/ld: /home/koz/dune3d/build/../src/document/solid_model.cpp:750:(.text+0x32d3): undefined reference to `BRepFilletAPI_MakeFillet::Build(Message_ProgressRange const&)'
/usr/bin/ld: dune3d.p/src_document_solid_model.cpp.o: in function `dune3d::SolidModel::create(dune3d::Document const&, dune3d::GroupChamfer const&)':
/home/koz/dune3d/build/../src/document/solid_model.cpp:772:(.text+0x355d): undefined reference to `BRepFilletAPI_MakeChamfer::BRepFilletAPI_MakeChamfer(TopoDS_Shape const&)'
/usr/bin/ld: /home/koz/dune3d/build/../src/document/solid_model.cpp:781:(.text+0x361e): undefined reference to `BRepFilletAPI_MakeChamfer::Add(double, TopoDS_Edge const&)'
/usr/bin/ld: /home/koz/dune3d/build/../src/document/solid_model.cpp:790:(.text+0x36a8): undefined reference to `BRepFilletAPI_MakeChamfer::Build(Message_ProgressRange const&)'
/usr/bin/ld: dune3d.p/src_document_solid_model.cpp.o: in function `ChFi3d_FilBuilder::~ChFi3d_FilBuilder()':
/usr/include/opencascade/ChFi3d_FilBuilder.hxx:45:(.text._ZN17ChFi3d_FilBuilderD2Ev[_ZN17ChFi3d_FilBuilderD5Ev]+0xf): undefined reference to `vtable for ChFi3d_FilBuilder'
/usr/bin/ld: /usr/include/opencascade/ChFi3d_FilBuilder.hxx:45:(.text._ZN17ChFi3d_FilBuilderD2Ev[_ZN17ChFi3d_FilBuilderD5Ev]+0x22): undefined reference to `ChFi3d_Builder::~ChFi3d_Builder()'
/usr/bin/ld: dune3d.p/src_document_solid_model.cpp.o: in function `BRepFilletAPI_MakeFillet::~BRepFilletAPI_MakeFillet()':
[... cut ...]
Program python3 found: YES (/usr/bin/python3)
Message: including git commit
Found pkg-config: /usr/bin/pkg-config (1.8.1)
Build-time dependency gio-2.0 found: YES 2.78.0
Program /usr/bin/glib-compile-resources found: YES (/usr/bin/glib-compile-resources)
Run-time dependency uuid found: YES 2.39.2
Dependency eigen3 found: YES 3.4.0 (cached)
Run-time dependency mimalloc found: YES 2.1
Build targets in project: 7
NOTICE: Future-deprecated features used:
 * 0.56.0: {'gui_app arg in executable'}

Found ninja-1.11.1 at /usr/bin/ninja
```

bl00dy1837, 2023-10-23:
Have the same error on archlinux.

Jookia, 2023-10-23:
Try commenting out this section in meson.build:

```
if not cxx.links('int main(void) {return 0;}', dependencies: opencascade)
    message('opencascade link args are broken, replacing')
    # on ubuntu 20.04, the link args are broken and are missing the -l for some reason??
    opencascade = declare_dependency(dependencies: opencascade.partial_dependency(compile_args: true, includes:true), link_args:['-lTKSTEP', '-lTKernel', '-lTKXCAF', '-lTKXSBase', '-lTKBRep', '-lTKCDF', '-lTKXDESTEP', '-lTKLCAF', '-lTKMath', '-lTKMesh', '-lTKTopAlgo', '-lTKPrim', '-lTKBO', '-lTKShHealing', '-lTKBRep', '-lTKG3d', '-lTKGeomBase', '-lTKHLR'])
endif
```

carrotIndustries, 2023-10-23:
Fixed in afdf07e715b8f263b90bacf68e344e8574779dd4

closed 2023-10-23 by carrotIndustries