dune3d/dune3d #3: document debian dependencies

khimaros, 2023-10-23:
this seemed to work in my debian trixie environment:

```
apt install libgtkmm-4.0-dev libeigen3-dev libglm-dev librange-v3-dev libocct-foundation-dev libmimalloc-dev
```

khimaros, 2023-10-23:
however, it appears that [gtk_list_box_remove_all](https://docs.gtk.org/gtk4/method.ListBox.remove_all.html) was not added until gtkmm 4.12 which means it is not able to be built as is even in debian testing

khimaros, 2023-10-23:
after working around this with a manual iteration over the children, it seems like there is a link failure related to libocct:

```
/usr/bin/ld: dune3d/dune3d/build/../src/import_step/step_importer.cpp:66:(.text+0x78): undefined reference to `STEPCAFControl_Reader::ReadFile(char const*)'
/usr/bin/ld: dune3d/dune3d/build/../src/import_step/step_importer.cpp:72:(.text+0x9f): undefined reference to `Interface_Static::SetIVal(char const*, int)'
/usr/bin/ld: dune3d/dune3d/build/../src/import_step/step_importer.cpp:77:(.text+0xcb): undefined reference to `Interface_Static::SetRVal(char const*, double)'
/usr/bin/ld: dune3d/dune3d/build/../src/import_step/step_importer.cpp:81:(.text+0xf0): undefined reference to `STEPCAFControl_Reader::SetColorMode(bool)'
/usr/bin/ld: dune3d/dune3d/build/../src/import_step/step_importer.cpp:82:(.text+0x104): undefined reference to `STEPCAFControl_Reader::SetNameMode(bool)'
/usr/bin/ld: dune3d/dune3d/build/../src/import_step/step_importer.cpp:83:(.text+0x118): undefined reference to `STEPCAFControl_Reader::SetLayerMode(bool)'
```

cam72cam, 2023-10-23:
I also needed to install `cmake librsvg2-dev python3-cairo-dev` on trixie.  I should probably start in a freshly bootstrapped container to find the full list.

cam72cam, 2023-10-23:
I wrote some terrible code that patches around the issue for the moment:
```patch
diff --git a/src/preferences/preferences_window_keys.cpp b/src/preferences/preferences_window_keys.cpp
index 3d32081..fd50245 100644
--- a/src/preferences/preferences_window_keys.cpp
+++ b/src/preferences/preferences_window_keys.cpp
@@ -50,6 +50,20 @@ std::vector<KeySequence> &get_keys() override
 };
 */
 
+
+void gtk_list_box_remove_all (Gtk::ListBox* box)
+{
+  for (auto row = box->get_row_at_index(0); row != NULL; row = box->get_row_at_index(0)) {
+    box->remove(*row);
+  }
+}
+void gtk_flow_box_remove_all (Gtk::FlowBox* box)
+{
+  for (auto row = box->get_child_at_index(0); row != NULL; row = box->get_child_at_index(0)) {
+    box->remove(*row);
+  }
+}
+
 class KeySequencesPreferencesEditor::ActionItem : public Glib::Object {
 public:
     static Glib::RefPtr<ActionItem> create()
@@ -331,7 +345,8 @@ public:
     void update()
     {
 
-        m_lb->remove_all();
+        //m_lb->remove_all();
+        gtk_list_box_remove_all(m_lb);
         size_t i = 0;
         if (m_prefs.key_sequences.keys.contains(m_action)) {
             auto &keys = m_prefs.key_sequences.keys.at(m_action);
@@ -373,10 +388,12 @@ private:
     Gtk::ListBox *m_lb = nullptr;
 };
 
+
 void KeySequencesPreferencesEditor::update_action_editors()
 {
 
-    m_action_editors->remove_all();
+    //m_action_editors->remove_all();
+    gtk_flow_box_remove_all(m_action_editors);
 
 
     auto row = std::dynamic_pointer_cast<Gtk::TreeListRow>(m_selection_model->get_selected_item());
```

It's been quite a few years since I've done any C++

carrotIndustries, 2023-10-23:
Fixed by https://github.com/dune3d/dune3d/commit/2d2d303cac66933d976d7d32d38f5cd026d0f07e The github actions configuration has all of the package names.

closed 2023-10-23 by carrotIndustries