dune3d/dune3d #6: Crash with `what: pick not found` when trying to run

tachiniererin, 2023-10-24:
When trying to run the application (Arch Linux + KDE), i get this crash:

```log
** (process:197198): WARNING **: 09:47:20.245: Can't set the global locale to the user's preferred locale.
   locale::facet::_S_create_c_locale name not valid
   The environment variable LANG may be wrong.

(process:197198): Gtk-WARNING **: 09:47:20.247: Locale not supported by C library.
        Using the fallback 'C' locale.
clear
realize

(dune3d:197198): Gtk-WARNING **: 09:47:20.443: GtkGizmo 0x5434a234c90 (slider) reported min width -2, but sizes must be >= 0

(dune3d:197198): Gtk-WARNING **: 09:47:20.443: GtkGizmo 0x5434a234910 (slider) reported min height -2, but sizes must be >= 0
resize 348x278

** (dune3d:197198): ERROR **: 09:47:20.455: 
unhandled exception (type std::exception) in signal handler:
what: pick not found

[1]    197198 trace trap (core dumped)  ./build/dune3d
```
Meson configure output:

```log
The Meson build system
Version: 1.2.3
Source dir: /home/elen/code/dune3d
Build dir: /home/elen/code/dune3d/build
Build type: native build
Project name: dune3d
Project version: 0.1.0
C compiler for the host machine: cc (gcc 13.2.1 "cc (GCC) 13.2.1 20230801")
C linker for the host machine: cc ld.bfd 2.41.0
C++ compiler for the host machine: c++ (gcc 13.2.1 "c++ (GCC) 13.2.1 20230801")
C++ linker for the host machine: c++ ld.bfd 2.41.0
Host machine cpu family: x86_64
Host machine cpu: x86_64
Found pkg-config: /usr/bin/pkg-config (1.8.1)
Run-time dependency gtk4 found: YES 4.12.3
Run-time dependency gtkmm-4.0 found: YES 4.12.0
Run-time dependency epoxy found: YES 1.5.10
Run-time dependency eigen3 found: YES 3.4.0
Run-time dependency glm found: YES 0.9.9.8
Found CMake: /usr/bin/cmake (3.27.7)
Run-time dependency opencascade found: YES 7.6.3
Run-time dependency range-v3 found: YES 0.12.0
Program python3 found: YES (/usr/bin/python3)
Message: including git commit
Found pkg-config: /usr/bin/pkg-config (1.8.1)
Build-time dependency gio-2.0 found: YES 2.78.0
Program /usr/bin/glib-compile-resources found: YES (/usr/bin/glib-compile-resources)
Run-time dependency uuid found: YES 2.39.2
Dependency eigen3 found: YES 3.4.0 (cached)
Run-time dependency mimalloc found: YES 2.1
Build targets in project: 7
NOTICE: Future-deprecated features used:
 * 0.56.0: {'gui_app arg in executable'}

Found ninja-1.11.1 at /usr/bin/ninja
```
Not quite sure what this is about if it works on Arch in CI tbh.

carrotIndustries, 2023-10-24:
Is this reproducible every time?

I've added some more debug output on the https://github.com/dune3d/dune3d/tree/pick-debug branch. Try running this and post the output.

tachiniererin, 2023-10-24:
Yep, happens every time. The value for pick varies, but dunno if that's relevant.

```log
➜  dune3d git:(pick-debug) ✗ ./build/dune3d

** (process:424169): WARNING **: 22:48:03.704: Can't set the global locale to the user's preferred locale.
   locale::facet::_S_create_c_locale name not valid
   The environment variable LANG may be wrong.

(process:424169): Gtk-WARNING **: 22:48:03.706: Locale not supported by C library.
        Using the fallback 'C' locale.
clear
realize

(dune3d:424169): Gtk-WARNING **: 22:48:03.901: GtkGizmo 0x2a45a234c90 (slider) reported min width -2, but sizes must be >= 0

(dune3d:424169): Gtk-WARNING **: 22:48:03.901: GtkGizmo 0x2a45a234910 (slider) reported min height -2, but sizes must be >= 0
resize 348x278
pick=65391
type=4 1:0

** (dune3d:424169): ERROR **: 22:48:03.914: 
unhandled exception (type std::exception) in signal handler:
what: pick not found

[1]    424169 trace trap (core dumped)  ./build/dune3d
```

carrotIndustries, 2023-10-24:
> Yep, happens every time. The value for pick varies, but dunno if that's relevant.

Looks like it's reading back garbage from the pick buffer for some reason. Probably has some relation to GPU drivers or me not using OpenGL exactly right. Which GPU are you using?

The linked branch now ignores bogus pick values rather than crashing. Does the selection work in a new document?

It's also grown a button that dumps the pick buffer into `/tmp/pick.txt` for further debugging. If selecting things still doesn't work, press that button and either attach the file or plot it with gnuplot using `plot '/tmp/pick.txt' matrix with image` and paste the plot.

tachiniererin, 2023-10-25:
The image:
[image]

Selecting still doesn't quite work. As for GPU, I have a RX 7900XT and a nvidia T400 in the same PC. But only the AMD card is connected to an output. The driver is latest mesa-git. I just tried on my laptop which has a 5850U with a Vega iGPU and latest mesa in arch repos (non -git) but still no dice.

carrotIndustries, 2023-10-25:
Hm, are you using a HiDPI setup with 2× scaling or so? I just pushed a commit onto `main` that fixes picking in that case.

tachiniererin, 2023-10-25:
Yep, that did the trick. I have fractional scaling on all systems but the fix on main works now.

closed 2023-10-25 by tachiniererin