dune3d/dune3d #8: Compiling on macOS

vespakoen, 2023-10-24:
I tried to compile dune3d on macOS to see how far it would get.

Here are the dependencies that I needed:

```shell
brew install meson eigen opencascade pkg-config gtk4 gtkmm4 glm range-v3 mimalloc pygobject3
```

I was unable to get `std::format` to work, I updated XCode and tried different C++ standards:

Output from `clang --version`:
```
clang --version
Apple clang version 15.0.0 (clang-1500.0.40.1)
Target: arm64-apple-darwin22.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

Snippet from `man clang`:
```
                 c++20

                    ISO C++ 2020 with amendments
                 gnu++20

                    ISO C++ 2020 with amendments and GNU extensions
                 c++2b

                    Working draft for ISO C++ 2023
                 gnu++2b

                    Working draft for ISO C++ 2023 with GNU extensions
```

I then replaced the `std::format` calls with a `"dummy"` string, and it compiled.

Launching the app crashes at `GL_CHECK_ERROR` in `canvas.cpp`

Commenting those checks out crashes a bit later at:

```
    if (glCheckFramebufferStatus(GL_FRAMEBUFFER) != GL_FRAMEBUFFER_COMPLETE) {
```

Full crash log:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               dune3d [98812]
Path:                  /Users/USER/*/dune3d
Identifier:            dune3d
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        zsh [47248]
Responsible:           iTerm2 [710]
User ID:               501

Date/Time:             2023-10-24 13:24:59.7137 +0200
OS Version:            macOS 13.5.1 (22G90)
Report Version:        12
Anonymous UUID:        DEB45627-BEA3-C886-1EA6-7EC11AF8556F

Sleep/Wake UUID:       6B0BFED2-B2BE-45E5-A984-FF944320AC5F

Time Awake Since Boot: 75000 seconds
Time Since Wake:       866 seconds
[... 314 lines cut ...]
       0x10baf4000 -        0x10bbc7fff libX11.6.dylib (*) <8d0d951d-ed42-32c3-b92c-ba0259c3dc5f> /opt/homebrew/*/libX11.6.dylib
       0x105bd0000 -        0x105bdbfff libXext.6.dylib (*) <62c931e5-f6d1-39df-9a9a-e8275b3e5c87> /opt/homebrew/*/libXext.6.dylib
       0x105e60000 -        0x105e67fff libXrender.1.dylib (*) <c6647c64-5365-30b5-9f7b-51fad293188b> /opt/homebrew/*/libXrender.1.dylib
       0x10b1cc000 -        0x10b1dffff libxcb.1.1.0.dylib (*) <ce7ef4f1-e5c8-373d-8e74-ec5af408822f> /opt/homebrew/*/libxcb.1.1.0.dylib
       0x109b34000 -        0x109b3bfff libxcb-render.0.0.0.dylib (*) <1a87ff6b-a668-3fa2-ac2a-3d7cc02676e8> /opt/homebrew/*/libxcb-render.0.0.0.dylib
       0x105994000 -        0x105997fff libxcb-shm.0.0.0.dylib (*) <0830787c-4131-3d0b-a0ca-50c537f2ca5b> /opt/homebrew/*/libxcb-shm.0.0.0.dylib
       0x10b9f8000 -        0x10ba67fff libpixman-1.0.42.2.dylib (*) <84f24785-ac6b-3da1-9dc7-aaec0b30aff6> /opt/homebrew/*/libpixman-1.0.42.2.dylib
       0x1057b8000 -        0x1057bbfff libXau.6.0.0.dylib (*) <af8acecd-49d7-34fb-9329-a8d20247afab> /opt/homebrew/*/libXau.6.0.0.dylib
       0x1057cc000 -        0x1057cffff libXdmcp.6.dylib (*) <f5f47c63-68c9-3673-95c0-7257dbf8f6d8> /opt/homebrew/*/libXdmcp.6.dylib
       0x10bcac000 -        0x10bd4ffff libzstd.1.5.5.dylib (*) <ab1456bc-240e-3a03-af02-06c1e94e6b0f> /opt/homebrew/*/libzstd.1.5.5.dylib
       0x10b7ac000 -        0x10b7cbfff liblzma.5.dylib (*) <305eb634-5675-3b96-953b-d35c657ca786> /opt/homebrew/*/liblzma.5.dylib
       0x10add0000 -        0x10ade7fff liblzo2.2.dylib (*) <500b9d12-d339-3013-b5bf-90120f99948e> /opt/homebrew/*/liblzo2.2.dylib
       0x1114fc000 -        0x111503fff libpixbufloader-svg.so (*) <f802ce0e-0e86-3a30-9447-41cb66c181c3> /opt/homebrew/*/libpixbufloader-svg.so
       0x1119fc000 -        0x111d57fff librsvg-2.2.dylib (*) <08deb726-139a-3671-bc94-41af9c785994> /opt/homebrew/*/librsvg-2.2.dylib
       0x111560000 -        0x11156bfff libobjc-trampolines.dylib (*) <8e928412-9e96-32d4-b173-d99beb9fed0b> /usr/lib/libobjc-trampolines.dylib
       0x1116b0000 -        0x11171bfff com.apple.AppleMetalOpenGLRenderer (1.0) <76243a19-d6d2-32ee-84d3-7bcfce93d66e> /System/Library/Extensions/AppleMetalOpenGLRenderer.bundle/Contents/MacOS/AppleMetalOpenGLRenderer
       0x196377000 -        0x1963b0ff7 libsystem_kernel.dylib (*) <08c5fe2a-b0bf-3ab6-bb42-460c18917d33> /usr/lib/system/libsystem_kernel.dylib
       0x1963b1000 -        0x1963bdfff libsystem_pthread.dylib (*) <1f30fb9a-bdf9-32db-a709-8417666a7e45> /usr/lib/system/libsystem_pthread.dylib
       0x19624f000 -        0x1962cdff7 libsystem_c.dylib (*) <949943e2-52cb-3c95-ab08-7ed984333d03> /usr/lib/system/libsystem_c.dylib
       0x19605a000 -        0x1960e8587 dyld (*) <e7a99595-e0f8-34af-be8b-9347d0d658a4> /usr/lib/dyld

--- SNIP ---
```

Hope this helps others trying the same ;)

vespakoen, 2023-10-24:
Forgot to post the error ;)

```
gl error 1,282 in ../src/canvas/canvas.cpp:435
```

vespakoen, 2023-10-24:
I also forgot to add that I had to add constructors for a couple of classes, here is the patch:

```diff
diff --git a/src/action/action.hpp b/src/action/action.hpp
index 7d2e6a8..4e7f5e1 100644
--- a/src/action/action.hpp
+++ b/src/action/action.hpp
@@ -31,6 +31,7 @@ enum class ActionSource {
 struct KeySequenceItem {
     unsigned int key;
     Gdk::ModifierType mod;
+    KeySequenceItem(unsigned int k, Gdk::ModifierType m) : key(k), mod(m) {}
     friend bool operator==(const KeySequenceItem &, const KeySequenceItem &) = default;
 };
 using KeySequence = std::vector<KeySequenceItem>;
diff --git a/src/canvas/canvas.hpp b/src/canvas/canvas.hpp
index 2c0f2cb..affb4e4 100644
--- a/src/canvas/canvas.hpp
+++ b/src/canvas/canvas.hpp
@@ -300,6 +300,16 @@ private:
         uint32_t bits;
 
         VertexFlags flags = VertexFlags::DEFAULT;
+
+        GlyphVertex(float x)
+            : x0(x), y0(0.0f), z0(0.0f), xs(0.0f), ys(0.0f), scale(1.0f), bits(0)
+        {}
+
+        GlyphVertex(float x, float y, float z,
+                    float xs_in, float ys_in,
+                    float scale_in, uint32_t bits_in)
+            : x0(x), y0(y), z0(z), xs(xs_in), ys(ys_in), scale(scale_in), bits(bits_in)
+        {}
     };
 
     std::vector<GlyphVertex> m_glyphs;
@@ -318,6 +328,10 @@ private:
         uint16_t icon_y;
 
         VertexFlags flags = VertexFlags::DEFAULT;
+
+        IconVertex(float x, float y, float z, float xs_in, float ys_in, uint16_t icon_x_in, uint16_t icon_y_in)
+            : x0(x), y0(y), z0(z), xs(xs_in), ys(ys_in), icon_x(icon_x_in), icon_y(icon_y_in)
+        {}
     };
 
     std::vector<IconVertex> m_icons;
diff --git a/src/document/group.hpp b/src/document/group.hpp
index 74f12dc..4f2b929 100644
--- a/src/document/group.hpp
+++ b/src/document/group.hpp
@@ -26,6 +26,8 @@ struct GroupStatusMessage {
     Status status;
     std::string message;
 
+    GroupStatusMessage(Status status_in, std::string message_in) : status(status_in), message(message_in) {}
+
     static GroupStatusMessage::Status summarize(const std::list<GroupStatusMessage> &msgs);
 };
 
diff --git a/src/preferences/preferences_window_canvas.cpp b/src/preferences/preferences_window_canvas.cpp
index 0456caf..d72b69b 100644
--- a/src/preferences/preferences_window_canvas.cpp
+++ b/src/preferences/preferences_window_canvas.cpp
@@ -82,7 +82,7 @@ static void spinbutton_set_px(Gtk::SpinButton &sp)
 {
     sp.signal_output().connect(
             [&sp] {
-                sp.set_text(std::format("{:.1f} px", sp.get_value()));
+                sp.set_text("test px"); //std::format("{:.1f} px", sp.get_value())
                 return true;
             },
             true);
diff --git a/src/render/renderer.cpp b/src/render/renderer.cpp
index 3567315..cdba923 100644
--- a/src/render/renderer.cpp
+++ b/src/render/renderer.cpp
@@ -379,7 +379,7 @@ void Renderer::visit(const ConstraintPointDistance &constr)
     m_ca.add_selectable(m_ca.draw_line(from, p1), sr);
     m_ca.add_selectable(m_ca.draw_line(to, p2), sr);
 
-    std::string label = std::format(" {:.3f}", constr.m_distance);
+    std::string label = "test"; // std::format(" {:.3f}", constr.m_distance);
     for (auto vr : m_ca.draw_bitmap_text(p, 1, label, 0)) {
         m_ca.add_selectable(vr, sr);
     }
@@ -420,7 +420,7 @@ void Renderer::visit(const ConstraintDiameterRadius &constr)
     m_ca.add_selectable(m_ca.draw_screen_line(to, (n + l * 1.5f) * 1.5f * scale), sr);
     m_ca.add_selectable(m_ca.draw_screen_line(to, (-n + l * 1.5f) * 1.5f * scale), sr);
 
-    std::string label = std::format(" {:.3f}", constr.m_distance);
+    std::string label = "test"; // "test"; // std::format(" {:.3f}", constr.m_distance);
     for (auto vr : m_ca.draw_bitmap_text(p, 1, label, 0)) {
         m_ca.add_selectable(vr, sr);
     }
@@ -477,7 +477,7 @@ void Renderer::visit(const ConstraintPointDistanceHV &constr)
 
     m_ca.add_selectable(m_ca.draw_line(wrkpl.transform(pt), wrkpl.transform(to)), sr);
 
-    std::string label = std::format(" {:.3f}", constr.m_distance);
+    std::string label = "test"; // std::format(" {:.3f}", constr.m_distance);
     for (auto vr : m_ca.draw_bitmap_text(wrkpl.transform(p), 1, label, 0)) {
         m_ca.add_selectable(vr, sr);
     }
```

Probably the issue now is locating the shaders etc, will look into it again another time.

rlatowicz, 2023-10-30:
Hi,
Don't know if this will help you but we experienced the same error regarding **std::format** on linux with **gcc/g++ v11.xx**

Upgrading to **gcc/g++ v13.xx** fixed it.

Try upgrading to **Clang 17** which claims to implement most features of **c++20**

**std::format** is a **c++20** feature,

ref. [https://en.cppreference.com/w/cpp/utility/format/format](url)

best of luck.

tkornack, 2023-11-14:
macOS only implements OpenGL 4.1. Perhaps this is using features from a more advanced version of OpenGL?

vespakoen, 2023-11-14:
That could be correct, I added some more `GL_CHECK_ERROR` statements and narrowed it down to this line:

```
 glRenderbufferStorageMultisample(GL_RENDERBUFFER, samples, GL_R16UI, m_width * get_scale_factor(),
                                      m_height * get_scale_factor());
```

And I figured out that `GL_R16UI` does not seem to be supported on macOS, so I tried changing it to: `GL_RGBA16F` which got me a bit further, it then got stuck at:

```
glBlitFramebuffer(0, 0, m_width * get_scale_factor(), m_height * get_scale_factor(), 0, 0,
                          m_width * get_scale_factor(), m_height * get_scale_factor(), GL_COLOR_BUFFER_BIT, GL_NEAREST);
```

Which, after commenting it out, got me this:

[image]

Promising! =)

It doesn't resize and crashes when doing things, but it's a start I guess...

vespakoen, 2023-11-14:
As promised: https://github.com/dune3d/dune3d/pull/25

It's a start, I tried a bunch of things and it didn't crash, but I am unable to select points / make extrudes / fillets etc, this could be due to me not understanding the program or maybe those things are not implemented yet, who knows...

Anyways, since I commented out some OpenGL calls, it makes sense that stuff is broken.
Hopefully someone else with more experience can make things work even better ;)

tkornack, 2024-01-05:
For the future: to make it right at home on macOS, consider https://github.com/vinceliuice/WhiteSur-gtk-theme

vespakoen, 2024-01-06:
@tkornack thanks, I'll take a look if I can use some CSS property to remove the padding.

Since macOS compilation should work now, I'll close this issue (see updated README for instructions).

closed 2024-01-06 by vespakoen