dune3d/dune3d #10: Better documentation for how to build on Windows

matt-roberts, 2023-10-25:
It took me a little while to work out how to build dune3d under windows.  Clearly it would have been quicker if there were a set of building instructions like there are for horizon-eda.  So I've written down what I did in that style:

# Building on Windows

## Install MSYS2

Download and run the msys2 installer from http://msys2.github.io/ 
It is probably a good idea to pick the 64bit version and to make sure that the path you select for installation doesn’t contain any spaces.

## Start MSYS console

Launch the Start Menu item “MSYS2 mingw 64 bit” you should be greeted with a console window. All steps below refer to what you should type into that window.

## Install updates

```
pacman -Syu
```

if it tells you to close restart msys, close the console window and start it again. Then run `pacman -Syu` again.

## Install dependencies

```
pacman -S \
	mingw-w64-x86_64-gcc \
	mingw-w64-x86_64-pkgconf \
	mingw-w64-x86_64-gtkmm4 \
	mingw-w64-x86_64-glm \
	mingw-w64-x86_64-opencascade \
	mingw-w64-x86_64-mimalloc \
	mingw-w64-x86_64-eigen3 \
	mingw-w64-x86_64-meson \
	mingw-w64-x86_64-cmake \
	mingw-w64-x86_64-python \
	mingw-w64-x86_64-python-cairo \
	mingw-w64-x86_64-python-gobject \
	mingw-w64-x86_64-range-v3 \
	zip \
	unzip \
	git \
	dos2unix \
	--needed
```

## Clone Dune3D

```
git clone https://github.com/dune3d/dune3d
cd dune3d
```

## Build it

```
meson setup build
meson compile -C build
```

## Running 

```
build/dune3d.exe
```

carrotIndustries, 2023-10-25:
Thanks, now please paste what you wrote here into a PR for the README.

ruevs, 2023-10-25:
The build process on Windows is also documented - in a rather roundabout way :-) - here: https://github.com/dune3d/dune3d/blob/main/.github/workflows/win32.yml

matt-roberts, 2023-10-26:
Hopefully this pull request is reasonable: https://github.com/dune3d/dune3d/pull/12

matt-roberts, 2023-10-26:
Just to let you know: I've put the build documentation in its own file because I think it'll expand a bitto become fairly large when (other) people add the instuctions for building on Mac and Linux.  And it feels that all three sets of instructions shouldn't clutter the main README that is shown on the main github page for dune3d.

ruevs, 2023-10-27:
With these instructions (and referring to this build https://github.com/VSngg/dune3d/actions/runs/6644145205/job/18052668872 as an extra check) I successfully built Dune 3D on Windows (10).

Here is my first small, completely pointless) model:
[Dune3D_MyFirstExperiment.zip](https://github.com/dune3d/dune3d/files/13186744/Dune3D_MyFirstExperiment.zip)

carrotIndustries, 2023-11-07:
Solved by #12

closed 2023-11-07 by carrotIndustries