dune3d/dune3d #11: + -> Sketch -> `unhandled exception (type std::exception) in signal handler: what: map::at`

q3k, 2023-10-25:
When pressing '+' on the left toolbar, then 'Sketch', I get the following crash on 74eea3740049c2329160f1419e343ea29e10d01e:

```
** (dune3d:95006): ERROR **: 16:25:41.932: 
unhandled exception (type std::exception) in signal handler:
what: map::at
```

GDB Backtrace:

```
Thread 1 "dune3d" received signal SIGTRAP, Trace/breakpoint trap.
0x00007ffff7333a77 in g_log_structured_array ()
   from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libglib-2.0.so.0
(gdb) bt
#0  0x00007ffff7333a77 in g_log_structured_array () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libglib-2.0.so.0
#1  0x00007ffff7333ece in g_log_default_handler () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libglib-2.0.so.0
#2  0x00007ffff7334138 in g_logv () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libglib-2.0.so.0
#3  0x00007ffff73343ef in g_log () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libglib-2.0.so.0
#4  0x00007ffff7ed02cd in Glib::exception_handlers_invoke() [clone .cold] () from /nix/store/5kml9hm2qhjrix6zy24rq1i0q2dh2qhw-glibmm-2.76.0/lib/libglibmm-2.68.so.1
#5  0x00007ffff693cc59 in (anonymous namespace)::SimpleAction_signal_activate_callback(_GSimpleAction*, _GVariant*, void*) [clone .cold] () from /nix/store/5kml9hm2qhjrix6zy24rq1i0q2dh2qhw-glibmm-2.76.0/lib/libgiomm-2.68.so.1
#6  0x00007ffff7f764c0 in g_closure_invoke () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#7  0x00007ffff7f89844 in signal_emit_unlocked_R.isra.0 () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#8  0x00007ffff7f9048d in g_signal_emit_valist () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#9  0x00007ffff7f9070f in g_signal_emit () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#10 0x00007ffff75041dd in g_simple_action_activate () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgio-2.0.so.0
#11 0x00007ffff79051e6 in gtk_menu_tracker_item_activated () from /nix/store/snfsharv7pljnr2b9a7ji6i1dzsqwadw-gtk4-4.10.3/lib/libgtk-4.so.1
#12 0x00007ffff7901608 in gtk_popover_item_activate () from /nix/store/snfsharv7pljnr2b9a7ji6i1dzsqwadw-gtk4-4.10.3/lib/libgtk-4.so.1
#13 0x00007ffff7f764c0 in g_closure_invoke () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#14 0x00007ffff7f89576 in signal_emit_unlocked_R.isra.0 () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#15 0x00007ffff7f9048d in g_signal_emit_valist () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#16 0x00007ffff7f9070f in g_signal_emit () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#17 0x00007ffff76ccb99 in _gtk_marshal_VOID__INT_DOUBLE_DOUBLEv () from /nix/store/snfsharv7pljnr2b9a7ji6i1dzsqwadw-gtk4-4.10.3/lib/libgtk-4.so.1
#18 0x00007ffff7f766b9 in _g_closure_invoke_va () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#19 0x00007ffff7f90457 in g_signal_emit_valist () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#20 0x00007ffff7f9070f in g_signal_emit () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#21 0x00007ffff7776d3e in gtk_gesture_click_end () from /nix/store/snfsharv7pljnr2b9a7ji6i1dzsqwadw-gtk4-4.10.3/lib/libgtk-4.so.1
#22 0x00007ffff7f79592 in g_cclosure_marshal_VOID__BOXEDv () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#23 0x00007ffff7f766b9 in _g_closure_invoke_va () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#24 0x00007ffff7f90457 in g_signal_emit_valist () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#25 0x00007ffff7f9070f in g_signal_emit () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#26 0x00007ffff7773d43 in _gtk_gesture_check_recognized () from /nix/store/snfsharv7pljnr2b9a7ji6i1dzsqwadw-gtk4-4.10.3/lib/libgtk-4.so.1
#27 0x00007ffff777546b in gtk_gesture_handle_event () from /nix/store/snfsharv7pljnr2b9a7ji6i1dzsqwadw-gtk4-4.10.3/lib/libgtk-4.so.1
#28 0x00007ffff7778343 in gtk_gesture_single_handle_event () from /nix/store/snfsharv7pljnr2b9a7ji6i1dzsqwadw-gtk4-4.10.3/lib/libgtk-4.so.1
#29 0x00007ffff7745e66 in gtk_event_controller_handle_event () from /nix/store/snfsharv7pljnr2b9a7ji6i1dzsqwadw-gtk4-4.10.3/lib/libgtk-4.so.1
#30 0x00007ffff7886868 in gtk_widget_run_controllers () from /nix/store/snfsharv7pljnr2b9a7ji6i1dzsqwadw-gtk4-4.10.3/lib/libgtk-4.so.1
#31 0x00007ffff788dfa4 in _gtk_widget_captured_event () from /nix/store/snfsharv7pljnr2b9a7ji6i1dzsqwadw-gtk4-4.10.3/lib/libgtk-4.so.1
#32 0x00007ffff77b077e in gtk_propagate_event_internal () from /nix/store/snfsharv7pljnr2b9a7ji6i1dzsqwadw-gtk4-4.10.3/lib/libgtk-4.so.1
#33 0x00007ffff77b0e3b in gtk_main_do_event () from /nix/store/snfsharv7pljnr2b9a7ji6i1dzsqwadw-gtk4-4.10.3/lib/libgtk-4.so.1
#34 0x00007ffff77daffc in surface_event () from /nix/store/snfsharv7pljnr2b9a7ji6i1dzsqwadw-gtk4-4.10.3/lib/libgtk-4.so.1
#35 0x00007ffff7a06029 in _gdk_marshal_BOOLEAN__POINTER () from /nix/store/snfsharv7pljnr2b9a7ji6i1dzsqwadw-gtk4-4.10.3/lib/libgtk-4.so.1
#36 0x00007ffff7a7d19f in gdk_surface_event_marshaller () from /nix/store/snfsharv7pljnr2b9a7ji6i1dzsqwadw-gtk4-4.10.3/lib/libgtk-4.so.1
#37 0x00007ffff7f764c0 in g_closure_invoke () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#38 0x00007ffff7f89576 in signal_emit_unlocked_R.isra.0 () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#39 0x00007ffff7f8fd75 in g_signal_emit_valist () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#40 0x00007ffff7f9070f in g_signal_emit () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgobject-2.0.so.0
#41 0x00007ffff7a807f5 in gdk_surface_handle_event () from /nix/store/snfsharv7pljnr2b9a7ji6i1dzsqwadw-gtk4-4.10.3/lib/libgtk-4.so.1
#42 0x00007ffff7a4ee9a in gdk_event_source_dispatch () from /nix/store/snfsharv7pljnr2b9a7ji6i1dzsqwadw-gtk4-4.10.3/lib/libgtk-4.so.1
#43 0x00007ffff732bdbd in g_main_context_dispatch () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libglib-2.0.so.0
#44 0x00007ffff732c068 in g_main_context_iterate.constprop () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libglib-2.0.so.0
#45 0x00007ffff732c0fc in g_main_context_iteration () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libglib-2.0.so.0
#46 0x00007ffff74fb53d in g_application_run () from /nix/store/kaq6wabyc6bgj71ryv4wk7nb824h9kih-glib-2.76.2/lib/libgio-2.0.so.0
#47 0x0000000000449940 in main (argc=1, argv=0x7ffffffd79d8) at ../src/main.cpp:11
```

To reproduce / versions / log:
```
$ cat shell.nix
with import (builtins.fetchTarball {
  url = "https://github.com/NixOS/nixpkgs/archive/39506d70cbdcf23b4d91f99a3a44f11ea113a9e7.tar.gz";
[... 30 lines cut ...]
Found pkg-config: /nix/store/7l5gbdz3rcab57r24arg61i4vavpf2cz-pkg-config-wrapper-0.29.2/bin/pkg-config (0.29.2)
Run-time dependency gtk4 found: YES 4.10.3
Run-time dependency gtkmm-4.0 found: YES 4.10.0
Run-time dependency epoxy found: YES 1.5.10
Run-time dependency eigen3 found: YES 3.4.0
Run-time dependency glm found: YES 0.9.9.8
Found CMake: /nix/store/h8q4r7f1j906rlffj96dlbbmwc9zbirk-cmake-3.25.3/bin/cmake (3.25.3)
Run-time dependency opencascade found: YES 7.6.2
Run-time dependency range-v3 found: YES 0.12.0
Program python3 found: YES (/nix/store/4wm5m2zmjs6vwlqx4aqrgjl6s7hwwcva-python3-3.10.13-env/bin/python3)
Message: including git commit
Found pkg-config: /nix/store/7l5gbdz3rcab57r24arg61i4vavpf2cz-pkg-config-wrapper-0.29.2/bin/pkg-config (0.29.2)
Program glib-compile-resources found: YES (/nix/store/g1ms0l8g8mykc19m8h2y3b4hzblipd5z-glib-2.76.2-dev/bin/glib-compile-resources)
Run-time dependency uuid found: YES 2.38.1
Dependency eigen3 found: YES 3.4.0 (cached)
Run-time dependency mimalloc found: YES 2.1
Build targets in project: 7
NOTICE: Future-deprecated features used:
 * 0.56.0: {'gui_app arg in executable'}

Found ninja-1.11.1 at /nix/store/04p9vxaplyd4a1m4q8qyh371dfmny3pl-ninja-1.11.1/bin/ninja
$ meson compile -C build
[... snip ...]
$ build/dune3d
```

q3k, 2023-10-25:
(this happens on selecting any of the options in the '+' menu)

q3k, 2023-10-25:
This might be a Nix issue, let me investigate further and see if I can figure this out myself.

carrotIndustries, 2023-10-25:
That thing should be insensitive if there's no document.

commit 4c1c0413 (2023-10-25): workspace browser: keep insensitive after opening

it should be insensitive when there are no documents

fixes #11
closed 2023-10-25 by carrotIndustries