dune3d/dune3d #13: Linking fails

jue89, 2023-10-31:
Staring with de51e81a49ea42227352fefd4f18042818afb3ab I'm not able to link Dune3D. Am I holding it wrong?

```
The Meson build system
Version: 1.2.1
Source dir: /build/source
Build dir: /build/source/build
Build type: native build
Project name: dune3d
Project version: 0.1.0
C compiler for the host machine: gcc (gcc 13.2.0 "gcc (GCC) 13.2.0")
C linker for the host machine: gcc ld.bfd 2.40
C++ compiler for the host machine: g++ (gcc 13.2.0 "g++ (GCC) 13.2.0")
C++ linker for the host machine: g++ ld.bfd 2.40
Host machine cpu family: x86_64
Host machine cpu: x86_64
Library m found: YES
Found pkg-config: /nix/store/ifr6srqgpvygd5vp14748d109ri31isv-pkg-config-wrapper-0.29.2/bin/pkg-config (0.29.2)
Run-time dependency gtk4 found: YES 4.10.4
Run-time dependency gtkmm-4.0 found: YES 4.10.0
Run-time dependency epoxy found: YES 1.5.10
Run-time dependency eigen3 found: YES 3.4.0
Run-time dependency glm found: YES 0.9.9.8
Found CMake: /nix/store/5h0akwq4cwlc3yp92i84nfgcxpv5xv79-cmake-3.26.4/bin/cmake (3.26.4)
Run-time dependency opencascade found: YES 7.6.2
Run-time dependency range-v3 found: YES 0.12.0
Program python3 found: YES (/nix/store/qnpajglqj86jm1pr6q9mwsyncvbfanl3-python3-3.11.5-env/bin/python3)
Found pkg-config: /nix/store/ifr6srqgpvygd5vp14748d109ri31isv-pkg-config-wrapper-0.29.2/bin/pkg-config (0.29.2)
Program glib-compile-resources found: YES (/nix/store/9mfjj1p9xkx1dzpbd20jfd5xbj0l5gan-glib-2.76.4-dev/bin/glib-compile-resources)
Fetching value of define "__GLIBCXX__" : 20230727 
Message: using libstdc++
Library stdc++ found: YES
Library stdc++fs found: YES
Run-time dependency uuid found: YES 2.39.2
Dependency eigen3 found: YES 3.4.0 (cached)
Run-time dependency mimalloc found: YES 2.1
Build targets in project: 7
NOTICE: Future-deprecated features used:
 * 0.56.0: {'gui_app arg in executable'}

dune3d 0.1.0

  User defined options
    auto_features: enabled
    bindir       : /nix/store/qdkyg86khhcfhg9k70rvw696if2jaxf4-dune3d-0.1.0/bin
    buildtype    : plain
    includedir   : /nix/store/qdkyg86khhcfhg9k70rvw696if2jaxf4-dune3d-0.1.0/include
    infodir      : /nix/store/qdkyg86khhcfhg9k70rvw696if2jaxf4-dune3d-0.1.0/share/info
    libdir       : /nix/store/qdkyg86khhcfhg9k70rvw696if2jaxf4-dune3d-0.1.0/lib
    libexecdir   : /nix/store/qdkyg86khhcfhg9k70rvw696if2jaxf4-dune3d-0.1.0/libexec
    localedir    : /nix/store/qdkyg86khhcfhg9k70rvw696if2jaxf4-dune3d-0.1.0/share/locale
    mandir       : /nix/store/qdkyg86khhcfhg9k70rvw696if2jaxf4-dune3d-0.1.0/share/man
    prefix       : /nix/store/qdkyg86khhcfhg9k70rvw696if2jaxf4-dune3d-0.1.0
    sbindir      : /nix/store/qdkyg86khhcfhg9k70rvw696if2jaxf4-dune3d-0.1.0/sbin
    wrap_mode    : nodownload

Found ninja-1.11.1 at /nix/store/jyk3j4632ldxsikjq0syjlj16i12jfsx-ninja-1.11.1/bin/ninja
meson: enabled parallel building
meson: enabled parallel installing
---SNIP---
[145/148] Compiling C++ object dune3d.p/src_preferences_preferences_window_keys_base.cpp.o
[146/148] Compiling C++ object dune3d.p/src_preferences_preferences_window_in_tool_keys.cpp.o
[147/148] Compiling C++ object dune3d.p/src_preferences_preferences_window_canvas.cpp.o
[148/148] Linking target dune3d
FAILED: dune3d 
g++  -o dune3d dune3d.p/meson-generated_.._version_gen.cpp.o dune3d.p/meson-generated_.._resources.c.o dune3d.p/meson-generated_.._icon_texture_map.cpp.o dune3d.p/src_main.cpp.o dune3d.p/src_dune3d_application.cpp.o dune3d.p/src_dune3d_appwindow.cpp.o dune3d.p/src_editor_editor.cpp.o dune3d.p/src_editor_editor_workspace_browser.cpp.o dune3d.p/src_canvas_canvas.cpp.o dune3d.p/src_canvas_gl_util.cpp.o dune3d.p/src_canvas_base_renderer.cpp.o dune3d.p/src_canvas_background_renderer.cpp.o dune3d.p/src_canvas_face_renderer.cpp.o dune3d.p/src_canvas_point_renderer.cpp.o dune3d.p/src_canvas_line_renderer.cpp.o dune3d.p/src_canvas_glyph_renderer.cpp.o dune3d.p/src_canvas_bitmap_font_util.cpp.o dune3d.p/src_canvas_bitmap_font_bitmap_font_desc.c.o dune3d.p/src_canvas_bitmap_font_bitmap_font_img.c.o dune3d.p/src_canvas_icon_renderer.cpp.o dune3d.p/src_canvas_appearance.cpp.o dune3d.p/src_canvas_selectable_ref.cpp.o dune3d.p/src_import_step_step_importer.cpp.o dune3d.p/src_import_step_step_import_manager.cpp.o dune3d.p/src_util_uuid.cpp.o dune3d.p/src_document_document.cpp.o dune3d.p/src_document_entity_entity.cpp.o dune3d.p/src_document_entity_entity_and_point.cpp.o dune3d.p/src_document_entity_entity_line3d.cpp.o dune3d.p/src_document_entity_entity_line2d.cpp.o dune3d.p/src_document_entity_entity_arc2d.cpp.o dune3d.p/src_document_entity_entity_circle2d.cpp.o dune3d.p/src_document_entity_entity_circle3d.cpp.o dune3d.p/src_document_entity_entity_arc3d.cpp.o dune3d.p/src_document_entity_entity_workplane.cpp.o dune3d.p/src_document_entity_entity_step.cpp.o dune3d.p/src_document_constraint_constraint.cpp.o dune3d.p/src_document_constraint_constraint_points_coincident.cpp.o dune3d.p/src_document_constraint_constraint_parallel.cpp.o dune3d.p/src_document_constraint_constraint_equal_length.cpp.o dune3d.p/src_document_constraint_constraint_equal_radius.cpp.o dune3d.p/src_document_constraint_constraint_point_on_line.cpp.o dune3d.p/src_document_constraint_constraint_point_on_circle.cpp.o dune3d.p/src_document_constraint_constraint_same_orientation.cpp.o dune3d.p/src_document_constraint_constraint_hv.cpp.o dune3d.p/src_document_constraint_constraint_point_distance.cpp.o dune3d.p/src_document_constraint_constraint_point_distance_hv.cpp.o dune3d.p/src_document_constraint_constraint_workplane_normal.cpp.o dune3d.p/src_document_constraint_constraint_midpoint.cpp.o dune3d.p/src_document_constraint_constraint_diameter_radius.cpp.o dune3d.p/src_document_constraint_constraint_arc_line_tangent.cpp.o dune3d.p/src_document_constraint_constraint_arc_arc_tangent.cpp.o dune3d.p/src_document_constraint_constraint_line_points_perpendicular.cpp.o dune3d.p/src_document_solid_model.cpp.o dune3d.p/src_document_solid_model_occ.cpp.o dune3d.p/src_document_group_group.cpp.o dune3d.p/src_document_group_group_sketch.cpp.o dune3d.p/src_document_group_group_reference.cpp.o dune3d.p/src_document_group_group_extrude.cpp.o dune3d.p/src_document_group_group_local_operation.cpp.o dune3d.p/src_document_group_group_fillet.cpp.o dune3d.p/src_document_group_group_chamfer.cpp.o dune3d.p/src_document_group_group_lathe.cpp.o dune3d.p/src_document_group_group_sweep.cpp.o dune3d.p/src_system_system.cpp.o dune3d.p/src_logger_logger.cpp.o dune3d.p/src_render_renderer.cpp.o dune3d.p/src_util_util.cpp.o dune3d.p/src_util_fs_util.cpp.o dune3d.p/src_util_json_util.cpp.o dune3d.p/src_util_glm_util.cpp.o dune3d.p/src_util_selection_util.cpp.o dune3d.p/src_util_file_version.cpp.o dune3d.p/src_core_core.cpp.o dune3d.p/src_core_tool.cpp.o dune3d.p/src_core_create_tool.cpp.o dune3d.p/src_core_tools_tool_common.cpp.o dune3d.p/src_core_tools_tool_draw_line_3d.cpp.o dune3d.p/src_core_tools_tool_draw_line_2d.cpp.o dune3d.p/src_core_tools_tool_draw_arc_2d.cpp.o dune3d.p/src_core_tools_tool_draw_circle_2d.cpp.o dune3d.p/src_core_tools_tool_delete.cpp.o dune3d.p/src_core_tools_tool_move.cpp.o dune3d.p/src_core_tools_tool_constrain_coincident.cpp.o dune3d.p/src_core_tools_tool_constrain_hv.cpp.o dune3d.p/src_core_tools_tool_constrain_distance.cpp.o dune3d.p/src_core_tools_tool_constrain_same_orientation.cpp.o dune3d.p/src_core_tools_tool_constrain_parallel.cpp.o dune3d.p/src_core_tools_tool_constrain_workplane_normal.cpp.o dune3d.p/src_core_tools_tool_constrain_midpoint.cpp.o dune3d.p/src_core_tools_tool_constrain_equal_length.cpp.o dune3d.p/src_core_tools_tool_constrain_equal_radius.cpp.o dune3d.p/src_core_tools_tool_constrain_diameter_radius.cpp.o dune3d.p/src_core_tools_tool_constrain_perpendicular.cpp.o dune3d.p/src_core_tools_tool_enter_datum.cpp.o dune3d.p/src_core_tools_tool_draw_workplane.cpp.o dune3d.p/src_core_tools_tool_add_anchor.cpp.o dune3d.p/src_core_tools_tool_toggle_construction.cpp.o dune3d.p/src_core_tools_tool_import_step.cpp.o dune3d.p/src_core_tools_tool_select_edges.cpp.o dune3d.p/src_core_tools_tool_set_workplane.cpp.o dune3d.p/src_core_tools_tool_helper_constrain.cpp.o dune3d.p/src_util_history_manager.cpp.o dune3d.p/src_action_action.cpp.o dune3d.p/src_action_action_catalog.cpp.o dune3d.p/src_in_tool_action_in_tool_action_catalog.cpp.o dune3d.p/src_util_str_util.cpp.o dune3d.p/src_util_lut.cpp.o dune3d.p/src_preferences_preferences.cpp.o dune3d.p/src_util_action_label.cpp.o dune3d.p/src_editor_tool_popover.cpp.o dune3d.p/src_editor_workspace_browser.cpp.o dune3d.p/src_widget [... cut ...]
/nix/store/p58l5qmzifl20qmjs3xfpl01f0mqlza2-binutils-2.40/bin/ld: dune3d.p/src_core_tools_tool_helper_constrain.cpp.o: in function `dune3d::ToolHelperConstrain::set_constrain_tip(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
tool_helper_constrain.cpp:(.text+0x576): undefined reference to `dune3d::Entity& dune3d::ToolCommon::get_entity<dune3d::Entity>(dune3d::UUID const&)'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

ruevs, 2023-10-31:
Works for me (Windows with MSYS2).

carrotIndustries, 2023-10-31:
I have no idea why this worked in the CI, but not on nixos.

closed 2023-10-31 by carrotIndustries

jue89, 2023-10-31:
Yes, Dune3D links perfectly, now. Thanks!

commit eb62dd2e (2023-10-31): tool helper constrain: add missing include

should fix #13