dune3d/dune3d #19: Version 1.0 TODOs

carrotIndustries, 2023-11-12:
So I don't forget what should go into version 1.0:

# Packaging
- [ ] Install target 
- [ ] Desktop file
- [ ] Flatpak
- [ ] AUR
- [ ] Windows ZIPs
- [ ] MSI installer

# Features
## Constraints
 - [ ] Perpendicular
 - [ ] Angle
 - [ ] Parallel

## Groups
 - [ ] Rectangular array
 - [ ] (Revolve) Is this really must-have?

## UI
 - [ ] Persp/ortho projection
 - [ ] Smooth zooming
 - [ ] Align camera to workplane
 - [ ] Horizontal/vertical tool bar preference
 - [ ] Select connected entities action
 - [ ] Make default/import export in keys preferences functional
 - [ ] Color themes, automatic dark/light
 
## Export
 - [ ] DXF / SVG export of contours

JB63134, 2023-11-25:
(Revolve) Is this really must-have?  Absolutely!  Bolt hole patterns on flanges being one example. gear teeth, etc

carrotIndustries, 2023-11-27:
> Bolt hole patterns on flanges being one example. gear teeth, etc

That's not what revolve does, that's a circular array.

carrotIndustries, 2024-01-05:
Documentation: https://docs.dune3d.org/en/latest/index.html

closed 2024-02-01 by carrotIndustries