dune3d/dune3d #23: FYI: a purple theme

jue89, 2023-11-13:
... if you like it, you may add it to the color themes ;-)

[image]

[image]

<details>
<summary>prefs.json</summary>

```json
    "canvas": {
        "colors": {
            "BACKGROUND_BOTTOM": {
                "b": 0.08915555477142334,
                "g": 0.1133333370089531,
                "r": 0.1133333370089531
            },
            "BACKGROUND_TOP": {
                "b": 0.3566666543483734,
                "g": 0.0939222127199173,
                "r": 0.20427489280700684
            },
            "CONSTRAINT": {
                "b": 0.6800000071525574,
                "g": 0.25,
                "r": 0.6899999976158142
            },
            "CONSTRUCTION_ENTITY": {
                "b": 0.36000001430511475,
                "g": 0.20999999344348907,
                "r": 0.36000001430511475
            },
            "CONSTRUCTION_POINT": {
                "b": 0.7066666483879089,
                "g": 0.3038666546344757,
                "r": 0.31997865438461304
            },
            "ENTITY": {
                "b": 1.0,
                "g": 0.7366666793823242,
                "r": 0.9315333962440491
            },
            "HIGHLIGHT": {
                "b": 1.0,
                "g": 0.6166666746139526,
                "r": 0.7700000405311584
            },
            "HOVER": {
                "b": 1.0,
                "g": 0.8957332968711853,
                "r": 0.7733333110809326
            },
            "INACTIVE_ENTITY": {
                "b": 0.3400000035762787,
                "g": 0.24480000138282776,
                "r": 0.3158447742462158
            },
            "INACTIVE_POINT": {
                "b": 0.42666667699813843,
                "g": 0.024177780374884605,
                "r": 0.2898204028606415
            },
            "POINT": {
                "b": 1.0,
                "g": 0.0,
                "r": 0.619999885559082
            },
            "SELECTED": {
                "b": 0.7399997711181641,
                "g": 0.0,
                "r": 1.0
            },
            "SELECTED_HOVER": {
                "b": 0.559999942779541,
                "g": 0.0,
                "r": 1.0
            },
            "SELECTION_BOX": {
                "b": 1.0,
                "g": 1.0,
                "r": 1.0
            },
            "SOLID_MODEL": {
                "b": 0.9766666889190674,
                "g": 0.8790000081062317,
                "r": 0.8848600387573242
            }
        },
        "line_width": 1.5,
        "msaa": 4
    }
```

</details>

carrotIndustries, 2023-11-14:
Look good, thanks! I'll eventually add this once I replace the user-definable colors with selectable color themes, of course retaining the custom option.

That way a color theme will be actually two sets of colors, a dark and a light variant, so the application can pick the right one depending on the user's system theme.

carrotIndustries, 2023-11-21:
included as the default dark theme since 9c6945ed9a9c85fb6135baa6344f0662a8b7a1c4

closed 2023-11-21 by carrotIndustries

jue89, 2023-11-21:
Nice! I feel honored ☺️