dune3d/dune3d #26: 3D mouse support

thirtythreeforty, 2023-11-18:
Please implement support for 3DConnexion input devices.

On Linux, [libspnav](https://github.com/FreeSpacenav/libspnav) can talk to spacenavd.  I am not super familiar with the official 3DxWare SDK on other platforms, though.

thirtythreeforty, 2023-11-18:
If you give me some pointers about where this should be integrated, I might be able to take a crack at a libspnav integration.

carrotIndustries, 2023-11-19:
Good news, the same feature already exists in Horizon EDA and Dune 3D's `Canvas`  is mostly a copy/paste from `Canvas3D` in Horizon EDA. Have a look at https://github.com/horizon-eda/horizon/blob/b78d8063cde1101a56a26562a8206e51e3e45add/src/canvas3d/canvas3d.cpp#L126

closed 2024-01-24 by carrotIndustries