dune3d/dune3d #34: UX: deselect everything after creating a new constrain

jue89, 2023-12-08:
I find myself clicking a lot because of deselecting everything after I've constrained parts of my model. I'd say UX would improve (at least for me) if this happens automatically.

carrotIndustries, 2023-12-11:
Hm, right now it's the same across all tools in that the selection and selection mode are retained by the tool.

So should constrain tools then reset the selection mode to hover and clear the selection? Or just clear the selection and keep the selection mode?

jue89, 2023-12-12:
TBH I haven't figured out how these selection modes work and how to switch between them :D Could you enlighten me?

carrotIndustries, 2023-12-12:
> TBH I haven't figured out how these selection modes work and how to switch between them :D Could you enlighten me?

They work pretty much the same way as they do in Horizon EDA: There's click select and hover select mode, as indicated in the status bar. To enter hover select mode, press escape. Click select is activated by clicking on something selectable.

In hover select mode, the hover selection is what's selected. In click select mode, one has to click on things for them to be selected.

jue89, 2023-12-12:
Wow, I haven't realized that, yet! Thanks a lot :)

> So should constrain tools then reset the selection mode to hover and clear the selection? Or just clear the selection and keep the selection mode?

I think it should clear the selection and go back to hover mode.

closed 2023-12-17 by carrotIndustries

jue89, 2023-12-18:
Awesome! Thank you :)

commit 2dcc129d (2023-12-17): constraint tools: reset selection mode

fixes #34