dune3d/dune3d #35: UX: draw contour doesn't create parallel constrain between arc and following line

jue89, 2023-12-08:
Drawing an arc after a line adds the *parallel constrain*. But drawing a line after an arc doesn't. I would like to see this constrain 
being created automatically.

Created constrains:
[image]

Expected constrains:
[image]

closed 2023-12-11 by carrotIndustries

jue89, 2023-12-12:
Thanks for addressing this!

Another improvement: Is it possible to add the tangential constrain *after* placing the second point of the following line? That would give more freedom to adjust the angle using the line tool.

jue89, 2023-12-18:
@carrotIndustries - I don't want to be pushy, but have you seen the comment above?

carrotIndustries, 2023-12-18:
I don't quite understand how that should work, can you elaborate a bit more?

jue89, 2023-12-21:
1. click for creating the first point for a line
2. click for creating the second point for the line (the line may get's a horizontal or vertical constrain with that)
3. switch to arc tool
4. click for creating the second point for the arc

Now you're back to the line tool. In the current version (b498e54) it sticks to the tangent of the second point of the arc. That feels cumbersome to me, since I can't be *sloppy* while placing the second point of the arc. Otherwise I have problems creating the line at that point i planned placing it because it's far of the tangent's ray.

I would suggest not to restrict that and allow the user to place the second point of the subsequent line freely. Instead, I would apply the tangent constrain between the arc and second line after the second point of the second line has been places.

Does this make sense?

carrotIndustries, 2023-12-23:
> I would suggest not to restrict that and allow the user to place the second point of the subsequent line freely.

One can always turn off the tangent constraint by pressing <kbd>t</kbd>.

jue89, 2023-12-25:
True. But then we're back to the old behavior.

If you intend it to work like this, I rest my case. Just wanted to point out what I think would be a significant improvement for my workflow.

carrotIndustries, 2023-12-25:
I think now I got what you meant: Moving the end point (top right in this case):
[image]
Should move the and point of the arc to keep the current line and the arc tangential.

If I got this right however, this would have one degree of freedom: the length of the line. Should the radius of the arc stay fixed?

jue89, 2023-12-27:
Thanks for sticking to this conversation :)

> Should the radius of the arc stay fixed?

I'd say yes. I guess that would feel kinda *intuitive*.

carrotIndustries, 2024-01-02:
> I'd say yes. I guess that would feel kinda intuitive.

Done in the commit above.

jue89, 2024-01-03:
Wow, this is perfect now! Thank you a lot!

commit 7a83e11d (2023-12-11): tool draw contour: second pass

 - create parallel constraint
 - be more smart about enabling tangent constraint

tangent constraints are now always created except for line-line

fixes #35
commit bd935b9e (2024-01-02): tool draw contour: move last arc's end point

fixes #35