dune3d/dune3d #38: Basic usage video

nh2, 2023-12-23:
Hi, currently trying dune3d out via your Nix build.

I cannot find the button to import at STEP file. I only managed to do that by setting a keybinding to it, but nothing appears when I select my STEP file.

Looks like this:

[image]

In contrast, if I load the sample [case.d3ddoc](https://github.com/dune3d/samples/blob/main/case.d3ddoc), there is a STEP `Path` selector on the left, like this:

[image]

If I put my STEP file in there, it works.

So I don't really get if I'm using dune3d correctly. (Note I don't really have experience with parametric CAD, beyod OpenSCAD.)

Could you make a small video that explains basic usage?

nh2, 2023-12-23:
Another thing (which I think is a bug) is that when in the second screenshot (example file), I switch to the different "layers" in the top left panel, then the lower panel with "Path" gets greyed out (deactivated), until I switch from `Selection` to `Group` and back.

Also, the checkboxes don't behave like checkboxes, but they "stack" (clicking one enables all checkboxes down to that one). Maybe that's expected, but checkboxes as UI elements don't make that super obvious.

Maybe the video could shortly explain the basics as well.

Thanks!

nh2, 2023-12-23:
> I cannot find the button to import at STEP file

Ah, from https://librearts.org/2023/11/radical-redesign-of-gimp/#enter-dune-3d, I can press `Space` to find the actions I can do?

How is one supposed to know that -- I would never have guessed.

If this dialogue is the main way one should interact with the program, it would be nice to add a button that the user can actually see, so that it is more discoverable.

Or a menu bar.

carrotIndustries, 2023-12-23:
> How is one supposed to know that -- I would never have guessed.

It'll eventually part of the to-be done user documentation and it's shown in the status bar "space for menu".

Anyhow, there indeed was a bug when importing step models if the document hasn't been saved yet.

carrotIndustries, 2024-01-05:
While not a video, the docs include a getting-started guide: https://docs.dune3d.org/en/latest/simple-part.html

closed 2024-01-05 by carrotIndustries

nh2, 2024-01-07:
Awesome!