dune3d/dune3d #40: UX: selecting certain points/lines is difficult when they overlap with other lines/points

jue89, 2023-12-30:
[image]

Selecting the currently highlighted line (purple) is difficult since the white line overlaps it.

Maybe a selection pop-over (like Horizon EDA?) could help. The pop-over allows the user to select a certain feature out of the group of overlapping features. The selection could be aided by highlighting the group in the tree-view the feature belongs to and the feature itself.

[image]

jue89, 2024-01-15:
I'd say https://github.com/dune3d/dune3d/commit/08a4f674f31b64ae0de6a576442d5c494c5f95bd improved the situation significantly! But the selection aid would be helpful, though

carrotIndustries, 2024-01-15:
This is a rather fundamental limitation of how the selection is implemented right now: During rendering, it creates a second buffer that contains 32 bit indices of the items. That buffer is read back to figure out what's at a certain pixel. While this is really easy to implement correctly it obviously makes it impossible to select things that are behind the topmost item.

For that to work, we need to look at the items themselves and have to do some math. It's something that'll eventually come, but not for 1.0

X-Ryl669, 2024-01-16:
You can try the ray-casting solution, but unless you scene is already well organized in a space structured tree (octree/KDTree/etc...), it'll be very slow since it's a O(N²) algorithm at best.

Another solution that might work quite well and simple to implement is to perform a n pass picking.
The first pass is the color picking you're already doing (that it, assigning a unique item ID as the color for each pixel drawn for this object and let the rendering sort out the front one).

The second pass only need to be done upon picking/casting the clicked position to the 3D scene. You only need to read the depth buffer for the area around the clicked point (let say you read a Z of 0.13 for example). That number matches the last triangle that has written the final color you've already read in step 1.
Then you'll render the scene again to a small viewport (can be as small as 1x1 but more likely you'll want some tolerance here for user selection, like 32x32). Set the Z near clipping range to 0.13 (or a bit less) so the previous triangle will be culled from rendering again). The color buffer will now contain the next item ID that was behind the former.

Repeat until you are reaching the clear color ID used for the background (or a fixed number of iteration, since in most case, you won't have more than 2 or 3 overlapping objects anyway)

If you don't have hundred of overlapping objects, this is going to be way faster and doesn't imply rewriting your scene to use a complex space partitioning structure.

carrotIndustries, 2024-06-21:
> Another solution that might work quite well and simple to implement is to perform a n pass picking.

Thanks for the suggestion, that's roughly what I implemented.

commit 760c8510 (2024-06-21): Canvas: add selection peeling

changelog: Enhancements/Editor: add menu for selecting obscured items

closes #40
closed 2024-06-21 by carrotIndustries