dune3d/dune3d #44: UX: end contour tool only if a coincide constrain is created to the first point of the currently drawn contour

jue89, 2024-01-04:
When drawing contours relative to points of other groups/contours the tool ends after every created coincide constrain and has to be restarted, even though the contour hasn't been closed, yet.

If it's possible to tell whether a coincide constrain has been created to a foreign point or the beginning of the currently drawn contour, it would be helpful to end the tool only if the latter is true.

carrotIndustries, 2024-01-04:
I see. I initially added this to better cover the use case of just drawing a single line from point A to point B.

So the tradeoff is
 - having to dismiss the extra line when just wanting to draw a single line
 - restarting the draw countour tool

Rethinking this, the first one is definitely the lesser evil.

I might also add a "draw line" tool that's a variation of the draw contour tool that automatically finishes after the first segment.

closed 2024-01-04 by carrotIndustries

jue89, 2024-01-04:
Nice! Thank you!

commit 67a562cd (2024-01-04): tool draw contour: only end when constraining to point of drawn path

closes #44