dune3d/dune3d #49: Linking error when building on MacOS

CreativeQuantum, 2024-01-15:
```
[187/187] Linking target dune3d
FAILED: dune3d
/usr/local/opt/llvm/bin/clang++  -o dune3d dune3d.p/meson-generated_.._version_gen.cpp.o dune3d.p/meson-generated_.._resources.c.o dune3d.p/meson-generated_.._icon_texture_map.cpp.o dune3d.p/meson-generated_.._color_presets.cpp.o dune3d.p/src_main.cpp.o dune3d.p/src_dune3d_application.cpp.o dune3d.p/src_dune3d_appwindow.cpp.o dune3d.p/src_editor_editor.cpp.o dune3d.p/src_editor_editor_workspace_browser.cpp.o dune3d.p/src_canvas_canvas.cpp.o dune3d.p/src_canvas_gl_util.cpp.o dune3d.p/src_canvas_base_renderer.cpp.o dune3d.p/src_canvas_background_renderer.cpp.o dune3d.p/src_canvas_face_renderer.cpp.o dune3d.p/src_canvas_point_renderer.cpp.o dune3d.p/src_canvas_line_renderer.cpp.o dune3d.p/src_canvas_glyph_renderer.cpp.o dune3d.p/src_canvas_glyph_3d_renderer.cpp.o dune3d.p/src_canvas_box_selection.cpp.o dune3d.p/src_canvas_bitmap_font_util.cpp.o dune3d.p/src_canvas_bitmap_font_bitmap_font_desc.c.o dune3d.p/src_canvas_bitmap_font_bitmap_font_img.c.o dune3d.p/src_canvas_icon_renderer.cpp.o dune3d.p/src_canvas_appearance.cpp.o dune3d.p/src_canvas_selectable_ref.cpp.o dune3d.p/src_import_step_step_importer.cpp.o dune3d.p/src_import_step_step_import_manager.cpp.o dune3d.p/src_util_uuid.cpp.o dune3d.p/src_document_document.cpp.o dune3d.p/src_document_entity_entity.cpp.o dune3d.p/src_document_entity_entity_and_point.cpp.o dune3d.p/src_document_entity_entity_line3d.cpp.o dune3d.p/src_document_entity_entity_line2d.cpp.o dune3d.p/src_document_entity_entity_arc2d.cpp.o dune3d.p/src_document_entity_entity_circle2d.cpp.o dune3d.p/src_document_entity_entity_circle3d.cpp.o dune3d.p/src_document_entity_entity_arc3d.cpp.o dune3d.p/src_document_entity_entity_workplane.cpp.o dune3d.p/src_document_entity_entity_step.cpp.o dune3d.p/src_document_entity_entity_point2d.cpp.o dune3d.p/src_document_constraint_constraint.cpp.o dune3d.p/src_document_constraint_constraint_points_coincident.cpp.o dune3d.p/src_document_constraint_constraint_parallel.cpp.o dune3d.p/src_document_constraint_constraint_equal_length.cpp.o dune3d.p/src_document_constraint_constraint_equal_radius.cpp.o dune3d.p/src_document_constraint_constraint_point_on_line.cpp.o dune3d.p/src_document_constraint_constraint_point_on_line_base.cpp.o dune3d.p/src_document_constraint_constraint_point_on_circle.cpp.o dune3d.p/src_document_constraint_constraint_same_orientation.cpp.o dune3d.p/src_document_constraint_constraint_hv.cpp.o dune3d.p/src_document_constraint_constraint_point_distance.cpp.o dune3d.p/src_document_constraint_constraint_point_distance_hv.cpp.o dune3d.p/src_document_constraint_constraint_workplane_normal.cpp.o dune3d.p/src_document_constraint_constraint_midpoint.cpp.o dune3d.p/src_document_constraint_constraint_diameter_radius.cpp.o dune3d.p/src_document_constraint_constraint_arc_line_tangent.cpp.o dune3d.p/src_document_constraint_constraint_arc_arc_tangent.cpp.o dune3d.p/src_document_constraint_constraint_line_points_perpendicular.cpp.o dune3d.p/src_document_constraint_constraint_angle.cpp.o dune3d.p/src_document_constraint_constraint_point_in_plane.cpp.o dune3d.p/src_document_constraint_constraint_point_line_distance.cpp.o dune3d.p/src_document_constraint_constraint_point_plane_distance.cpp.o dune3d.p/src_document_constraint_constraint_lock_rotation.cpp.o dune3d.p/src_document_solid_model.cpp.o dune3d.p/src_document_solid_model_occ.cpp.o dune3d.p/src_document_solid_model_array.cpp.o dune3d.p/src_document_solid_model_local_operation.cpp.o dune3d.p/src_document_solid_model_lathe.cpp.o dune3d.p/src_document_solid_model_extrude.cpp.o dune3d.p/src_document_solid_model_util.cpp.o dune3d.p/src_document_group_group.cpp.o dune3d.p/src_document_group_group_sketch.cpp.o dune3d.p/src_document_group_group_reference.cpp.o dune3d.p/src_document_group_group_extrude.cpp.o dune3d.p/src_document_group_group_local_operation.cpp.o dune3d.p/src_document_group_group_fillet.cpp.o dune3d.p/src_document_group_group_chamfer.cpp.o dune3d.p/src_document_group_group_lathe.cpp.o dune3d.p/src_document_group_group_sweep.cpp.o dune3d.p/src_document_group_group_linear_array.cpp.o dune3d.p/src_document_group_group_polar_array.cpp.o dune3d.p/src_document_group_group_array.cpp.o dune3d.p/src_document_export_paths.cpp.o dune3d.p/src_system_system.cpp.o dune3d.p/src_logger_logger.cpp.o dune3d.p/src_logger_log_dispatcher.cpp.o dune3d.p/src_render_renderer.cpp.o dune3d.p/src_util_util.cpp.o dune3d.p/src_util_fs_util.cpp.o dune3d.p/src_util_json_util.cpp.o dune3d.p/src_util_glm_util.cpp.o dune3d.p/src_util_selection_util.cpp.o dune3d.p/src_util_file_version.cpp.o dune3d.p/src_core_core.cpp.o dune3d.p/src_core_tool.cpp.o dune3d.p/src_core_create_tool.cpp.o dune3d.p/src_core_tools_tool_common.cpp.o dune3d.p/src_core_tools_tool_draw_line_3d.cpp.o dune3d.p/src_core_tools_tool_draw_point_2d.cpp.o dune3d.p/src_core_tools_tool_draw_circle_2d.cpp.o dune3d.p/src_core_tools_tool_draw_contour.cpp.o dune3d.p/src_core_tools_tool_delete.cpp.o dune3d.p/src_core_tools_tool_move.cpp.o dune3d.p/src_core_tools_tool_constrain_coincident.cpp.o dune3d.p/src_core_tools_tool_constrain_hv.cpp.o dune3d.p/src_core_tools_tool_constrain_distance.cpp.o dune3d.p/src_core_tools_tool_constrain_same_orientation.cpp.o dune3d.p/src_core_tools_tool_constrain_parallel.cpp.o dune3d.p/src_core_tools_tool_constrain_workplane_normal.cpp.o dune3d.p/src_core_tools_tool_constrain_midpoint.cpp.o dune3d.p/src_core_tools_tool_constrain_equal_length.cpp.o dune3d.p/src_core_tools_tool_constrain_equal_radius.cpp.o dune3d.p/src_core_tools_tool_constrain_diameter_radius.cpp.o dune3d.p/src_core_tools_tool_constrain_perpendicular.cpp.o dune3d.p/src_core_tools_tool_constrain_point_in_plane.cpp.o dune3d.p/src_core_tools_tool_constrain_lock_rotation.cpp.o dune3d.p/src_core_tools_tool_enter_datum.cpp.o dune3d.p/src_core_tools_tool_draw_workplane.cpp.o dune3d.p/src_core_tools_tool_add_anchor.cpp.o dune3d.p/src_core_tools_tool_move_anchor.cpp.o dune3d.p/src_core_tools_tool_toggle_construction.cpp.o dune3d.p/src_core_tools_tool_import_step.cpp.o dune3d.p/src_core_tools_tool_select_edges.cpp.o dune3d.p/src_core_tools_tool_set_workplane.cpp.o dune3d.p/src_core_tools_tool_helper_constrain.cpp.o dune3d.p/src_core_tools_tool_rotate.cpp.o dune3d.p/src_core_tools_tool_draw_regular_polygon.cpp.o dune3d.p/src_core_tools_tool_draw_rectangle.cpp.o dune3d.p/src_core_tools_tool_flip_arc.cpp.o dune3d.p/src_util_history_manager.cpp.o dune3d.p/src_action_action.cpp.o dune3d.p/src_action_action_catalog.cpp.o dune3d.p/src_in_tool_action_in_tool_action_catalog.cpp.o dune3d.p/src_util_str_util.cpp.o dune3d.p/src_util_lut.cpp.o dune3d.p/src_preferences_preferences.cpp.o dune3d.p/src_util_action_label.cpp.o dune3d.p/src_editor_tool_popover.cpp.o dune3d.p/src_editor_workspace_browser.cpp.o dune3d.p/src_widgets_constraints_box.cpp.o dune3d.p/src_editor_document_view.cpp.o dune3d.p/src_editor_group_editor.cpp.o dune3d.p/src_editor_selection_editor.cpp.o dune3d.p/src_dialogs_dialogs.cpp.o dune3d.p/src_dialogs_enter_datum_window.cpp.o dune3d.p/src_dialogs_rotate_window.cpp.o dune3d.p/src_dialogs_tool_window.cpp.o dune3d.p/src_widgets_spin_button_dim.cpp.o dune3d.p/src_widgets_spin_button_angle.cpp.o dune3d.p/src_util_gtk_util.cpp.o dune3d.p/src_preferences_preferences_window.cpp.o dune3d.p/src_preferences_action_editor.cpp.o dune3d.p/src_preferences_preferences_window_keys.cpp.o dune3d.p/src_preferences_preferences_window_in_tool_keys.cpp.o dune3d.p/src_preferences_preferences_window_keys_base.cpp.o dune3d.p/src_preferences_preferences_window_canvas.cpp.o dune3d.p/src_preferences_preferences_row.cpp.o dune3d.p/src_preferences_preferences_window_misc.cpp.o dune3d.p/src_widgets_capture_dialog.cpp.o dune3d.p/src_widgets_axes_lollipop.cpp.o dune3d.p/src_widgets_about_dialog.cpp.o dune3d.p/src_widgets_recent_item_box.cpp.o dune3d.p/src_widgets_log_view.cpp.o dune3d.p/src_widgets_log_window.cpp.o dune3d.p/src_widgets_clipping_plane_window.cpp.o dune3d.p/src_util_version.cpp.o dune3d.p/src_util_msd.cpp.o dune3d.p/src_util_msd_animator.cpp.o dune3d.p/src_util_exception_util.cpp.o -I/usr/local/opt/openjdk@17/include -Wl,-dead_strip_dylibs -Wl,-headerpad_max_install_names -Wl,-rpath,/usr/local/Cellar/gtk4/4.12.4/lib -Wl,-rpath,/usr/local/Cellar/pango/1.50.14/lib -Wl,-rpath,/usr/local/Cellar/harfbuzz/8.3.0/lib -Wl,-rpath,/usr/local/Cellar/gdk-pixbuf/2.42.10_1/lib -Wl,-rpath,/usr/local/Cellar/cairo/1.18.0/lib -Wl,-rpath,/usr/local/Cellar/graphene/1.10.8/lib -Wl,-rpath,/usr/local/Cella [... cut ...]
      Processor Name: Dual-Core Intel Core i5
      Processor Speed: 2,7 GHz
      Number of Processors: 1
      Total Number of Cores: 2
      L2 Cache (per Core): 256 KB
      L3 Cache: 3 MB
      Hyper-Threading Technology: Enabled
      Memory: 8 GB
      System Firmware Version: 476.0.0.0.0
      OS Loader Version: 540.120.3~22
      SMC Version (system): 2.28f7
      Serial Number (system): C02QCGUCFVH3
      Hardware UUID: DDD404CD-A218-5952-8F38-16F434CE1148
      Provisioning UDID: DDD404CD-A218-5952-8F38-16F434CE1148
```

CreativeQuantum, 2024-01-15:
Tried to recompile with this script:

```sh
#!/usr/bin/env bash

llvm_prefix="$(brew --prefix llvm)"
export CC="$llvm_prefix/bin/clang"
export CXX="$llvm_prefix/bin/clang++"

export LDFLAGS="-L/usr/local/opt/llvm/lib/,-L/usr/local/opt/llvm/lib/c++ -Wl,-rpath,/usr/local/opt/llvm/lib/c++"
export CPPFLAGS="-I/usr/local/opt/llvm/include"
meson setup --wipe build
meson compile -C build
```

following brew instructions:

```
To use the bundled libc++ please add the following LDFLAGS:
  LDFLAGS="-L/usr/local/opt/llvm/lib/c++ -Wl,-rpath,/usr/local/opt/llvm/lib/c++"

llvm is keg-only, which means it was not symlinked into /usr/local,
because macOS already provides this software and installing another version in
parallel can cause all kinds of trouble.

If you need to have llvm first in your PATH, run:
  echo 'export PATH="/usr/local/opt/llvm/bin:$PATH"' >> ~/.zshrc

For compilers to find llvm you may need to set:
  export LDFLAGS="-L/usr/local/opt/llvm/lib"
  export CPPFLAGS="-I/usr/local/opt/llvm/include"
```

Same problem.

vespakoen, 2024-01-15:
You are missing std::format, which is not available in your macOS version, you can:
- update your mac
- integrate a std:formatter library [fmt](https://github.com/fmtlib/fmt)
- change the std::format calls to something else (like over here: https://github.com/dune3d/dune3d/pull/25/files)

Please note that at the moment selection is not working because lines can only be 1px wide on macOS, and an image from the canvas is used for a selection lookup table, so don't get your hopes up ;)

CreativeQuantum, 2024-01-15:
Got it working:

```
[187/187] Linking target dune3d
ld: warning: directory not found for option '-L/usr/local/opt/llvm/lib/,-L/usr/local/opt/llvm/lib/c++'
ld: warning: dylib (/usr/local/opt/llvm/lib/c++/libc++.dylib) was built for newer macOS version (12.6) than being linked (12.0)
```
The LDFLAGS weren't picked up by meson, so I updated `meson.build`

```diff
diff --git a/meson.build b/meson.build
index 2ac4bb7..79b98c3 100644
--- a/meson.build
+++ b/meson.build
@@ -21,7 +21,7 @@ if target_machine.system() == 'darwin'
   message('meson\'s framework handling on macos with the cmake method is broken, replacing')
   brew_prefix_cmd = run_command('brew', '--prefix', check: true)
   brew_prefix = brew_prefix_cmd.stdout().strip() + '/lib'
-  opencascade = declare_dependency(dependencies: opencascade.partial_dependency(compile_args: true, includes:true), link_args:['-Wl,-L' + brew_prefix, '-lTKSTEP', '-lTKernel', '-lTKXCAF', '-lTKXSBase', '-lTKBRep', '-lTKCDF', '-lTKXDESTEP', '-lTKLCAF', '-lTKMath', '-lTKMesh', '-lTKTopAlgo', '-lTKPrim', '-lTKBO', '-lTKShHealing', '-lTKG3d', '-lTKGeomBase', '-lTKHLR', '-lTKSTL', '-lTKFillet'])
+  opencascade = declare_dependency(dependencies: opencascade.partial_dependency(compile_args: true, includes:true), link_args:['-Wl,-L' + brew_prefix, '-lTKSTEP', '-lTKernel', '-lTKXCAF', '-lTKXSBase', '-lTKBRep', '-lTKCDF', '-lTKXDESTEP', '-lTKLCAF', '-lTKMath', '-lTKMesh', '-lTKTopAlgo', '-lTKPrim', '-lTKBO', '-lTKShHealing', '-lTKG3d', '-lTKGeomBase', '-lTKHLR', '-lTKSTL', '-lTKFillet', '-L/usr/local/opt/llvm/lib/c++'])
 endif

 spnav = dependency('spnav', required: false)
```

Let me know if you want me to open a new PR.

vespakoen, 2024-01-15:
Not sure why they were not picked up on your side, it seems to work correctly in CI and "over here".

Adding them to the opencascade dep is not really clean either so let's maybe wait until more reports come in to see if this needs fixing?

viernullvier, 2024-02-09:
> so let's maybe wait until more reports come in to see if this needs fixing?

FWIW, I've managed to build the app successfully without any changes to the build script on two different MacOS devices (Intel and ARM)

calumk, 2024-02-26:
I have succsessfull built on M1 mac

forivall, 2024-03-11:
I'm not sure why, but i'm getting this error. Idk if it's the same. x86_64

```
ld: Undefined symbols:
  Glib::DateTime::create_now_local(long), referenced from:
      dune3d::Dune3DApplication::UserConfig::load(std::__1::__fs::filesystem::path const&) in src_dune3d_application.cpp.o
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

<details>
<summary><code>brew ls --verbose glib glibmm gtk4 gtkmm4</code></summary>

```
/usr/local/Cellar/glib/2.80.0/INSTALL_RECEIPT.json
/usr/local/Cellar/glib/2.80.0/bin/gtester
/usr/local/Cellar/glib/2.80.0/bin/gdbus-codegen
/usr/local/Cellar/glib/2.80.0/bin/gsettings
/usr/local/Cellar/glib/2.80.0/bin/glib-compile-schemas
/usr/local/Cellar/glib/2.80.0/bin/gobject-query
/usr/local/Cellar/glib/2.80.0/bin/gi-decompile-typelib
/usr/local/Cellar/glib/2.80.0/bin/gio-querymodules
/usr/local/Cellar/glib/2.80.0/bin/glib-compile-resources
/usr/local/Cellar/glib/2.80.0/bin/gresource
/usr/local/Cellar/glib/2.80.0/bin/gtester-report
/usr/local/Cellar/glib/2.80.0/bin/glib-gettextize
/usr/local/Cellar/glib/2.80.0/bin/gdbus
/usr/local/Cellar/glib/2.80.0/bin/glib-genmarshal
/usr/local/Cellar/glib/2.80.0/bin/gi-compile-repository
/usr/local/Cellar/glib/2.80.0/bin/glib-mkenums
/usr/local/Cellar/glib/2.80.0/bin/gi-inspect-typelib
/usr/local/Cellar/glib/2.80.0/bin/gio
/usr/local/Cellar/glib/2.80.0/.brew/glib.rb
/usr/local/Cellar/glib/2.80.0/libexec/bin/python3
/usr/local/Cellar/glib/2.80.0/libexec/bin/python
/usr/local/Cellar/glib/2.80.0/libexec/bin/python3.12
/usr/local/Cellar/glib/2.80.0/libexec/pyvenv.cfg
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging/tags.py
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging/_musllinux.py
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging/metadata.py
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging/version.py
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging/__init__.py
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging/_parser.py
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging/utils.py
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging/requirements.py
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging/_structures.py
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging/markers.py
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging/py.typed
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging/_manylinux.py
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging/_tokenizer.py
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging/specifiers.py
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging/_elffile.py
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging-23.2.dist-info/RECORD
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging-23.2.dist-info/LICENSE
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging-23.2.dist-info/LICENSE.APACHE
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging-23.2.dist-info/WHEEL
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging-23.2.dist-info/LICENSE.BSD
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging-23.2.dist-info/REQUESTED
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging-23.2.dist-info/INSTALLER
/usr/local/Cellar/glib/2.80.0/libexec/lib/python3.12/site-packages/packaging-23.2.dist-info/METADATA
/usr/local/Cellar/glib/2.80.0/include/gio-unix-2.0/gio/gunixoutputstream.h
/usr/local/Cellar/glib/2.80.0/include/gio-unix-2.0/gio/gunixinputstream.h
/usr/local/Cellar/glib/2.80.0/include/gio-unix-2.0/gio/gunixmounts.h
/usr/local/Cellar/glib/2.80.0/include/gio-unix-2.0/gio/gunixfdmessage.h
/usr/local/Cellar/glib/2.80.0/include/gio-unix-2.0/gio/gfiledescriptorbased.h
/usr/local/Cellar/glib/2.80.0/include/glib-2.0/glib-object.h
/usr/local/Cellar/glib/2.80.0/include/glib-2.0/gobject/gparam.h
/usr/local/Cellar/glib/2.80.0/include/glib-2.0/gobject/gparamspecs.h
/usr/local/Cellar/glib/2.80.0/include/glib-2.0/gobject/gsignal.h
/usr/local/Cellar/glib/2.80.0/include/glib-2.0/gobject/genums.h
[... 2135 lines cut ...]
/usr/local/Cellar/gtkmm4/4.12.0/include/gtkmm-4.0/gtkmm/cellrendereraccel.h
/usr/local/Cellar/gtkmm4/4.12.0/include/gtkmm-4.0/gtkmm/treeiter.h
/usr/local/Cellar/gtkmm4/4.12.0/include/gtkmm-4.0/gtkmm/boxlayout.h
/usr/local/Cellar/gtkmm4/4.12.0/include/gtkmm-4.0/gtkmm/passwordentry.h
/usr/local/Cellar/gtkmm4/4.12.0/include/gtkmm-4.0/gtkmm/textview.h
/usr/local/Cellar/gtkmm4/4.12.0/include/gtkmm-4.0/gtkmm/cellareabox.h
/usr/local/Cellar/gtkmm4/4.12.0/COPYING.tools
/usr/local/Cellar/gtkmm4/4.12.0/README.md
/usr/local/Cellar/gtkmm4/4.12.0/COPYING
/usr/local/Cellar/gtkmm4/4.12.0/NEWS
/usr/local/Cellar/gtkmm4/4.12.0/lib/pkgconfig/gtkmm-4.0.pc
/usr/local/Cellar/gtkmm4/4.12.0/lib/libgtkmm-4.0.dylib
/usr/local/Cellar/gtkmm4/4.12.0/lib/libgtkmm-4.0.0.dylib
/usr/local/Cellar/gtkmm4/4.12.0/lib/gtkmm-4.0/proc/m4/convert_gtkmm.m4
/usr/local/Cellar/gtkmm4/4.12.0/lib/gtkmm-4.0/proc/m4/class_gtkobject.m4
/usr/local/Cellar/gtkmm4/4.12.0/lib/gtkmm-4.0/proc/m4/convert_gdk.m4
/usr/local/Cellar/gtkmm4/4.12.0/lib/gtkmm-4.0/proc/m4/convert_gtk.m4
/usr/local/Cellar/gtkmm4/4.12.0/lib/gtkmm-4.0/proc/m4/class_shared.m4
/usr/local/Cellar/gtkmm4/4.12.0/lib/gtkmm-4.0/proc/m4/convert.m4
/usr/local/Cellar/gtkmm4/4.12.0/lib/gtkmm-4.0/include/gdkmmconfig.h
/usr/local/Cellar/gtkmm4/4.12.0/lib/gtkmm-4.0/include/gtkmmconfig.h

```

</details>

sparkylein, 2024-03-26:
@forivall 
I got the same message on M1 (ARM) with macOS 14.3.1.
All libraries are installed via homebrew.
```ld: Undefined symbols:
  Glib::DateTime::create_now_local(long long), referenced from:
      dune3d::Dune3DApplication::UserConfig::load(std::__1::__fs::filesystem::path const&) in src_dune3d_application.cpp.o
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

JacksonCampolattaro, 2024-03-28:
> @forivall I got the same message on M1 (ARM) with macOS 14.3.1. All libraries are installed via homebrew.

I have the same situation on 14.4.1; also an M1 mac with all libraries installed with brew (and fully updated). I tried a few of the fixes listed above, but no luck. `Glib::DateTime::create_now_local` hasn't changed recently, but if that's the first function used it's possible the entirety of Glib can't be linked against.

ronangaillard, 2024-05-01:
Hi,

I have a Mac M2 fully up to date. I hav the exact same problem while building (ie. the Glib::DateTime::create_now_local issue).

carrotIndustries, 2024-05-02:
I have no idea about mac OS, but can someone do (the equivalent) of `objdump -C -T /usr/lib/libglibmm-2.68.so`  to see which symbols are actually exported?

JacksonCampolattaro, 2024-05-02:
Here's the output of `nm -g /opt/homebrew/Cellar/glibmm/2.80.0/lib/libglibmm-2.68.dylib | c++filt > glibmm-symbols.txt`, I hope it helps!

[glibmm-symbols.txt](https://github.com/dune3d/dune3d/files/15189517/glibmm-symbols.txt)

carrotIndustries, 2024-05-02:
>  but there is no create_now_local(long long).

The function is called `create_now_local(gint64 t)`  on linux, `gint64` is defined as `signed long` in glibconfig.h.
What is it defined as on mac os?

JacksonCampolattaro, 2024-05-02:
From glibconfig.h, it looks like it's a `signed long long`:
```
G_GNUC_EXTENSION typedef signed long long gint64;
```
The headers are the same, too:
```
static DateTime create_now_local();
static DateTime create_now_local(gint64 t);
```
This makes it really weird that the packaged binary doesn't match.

carrotIndustries, 2024-05-02:
> This makes it really weird that the packaged binary doesn't match.

Yeah, maybe try rebuilding glibmm?

ronangaillard, 2024-05-02:
Where is that `glibconfig.h` located on macos ?

JacksonCampolattaro, 2024-05-02:
> Where is that `glibconfig.h` located on macos ? (I have no such file in `/opt/homebrew/opt/glib/include/glib-2.0` neither in ` /opt/homebrew/Cellar/glibmm/2.80.0/include`)

On my computer, the full path was `/opt/homebrew/Cellar/glib/2.80.0_2/lib/glib-2.0/include/glibconfig.h`

ronangaillard, 2024-05-02:
Thanks, it's now building with your fix and this fix : https://github.com/dune3d/dune3d/issues/49#issuecomment-1892276583

JacksonCampolattaro, 2024-05-02:
Glad to hear! This seems like a more widespread issue with the glib's homebrew distribution, but some people built without needing the patch, so maybe it only affects certain configurations.

rkarlsba, 2024-05-05:
I'm seeing the same error 
ld: Undefined symbols:
  Glib::DateTime::create_now_local(long long), referenced from:
      dune3d::Dune3DApplication::UserConfig::load(std::__1::__fs::filesystem::path const&) in src_dune3d_application.cpp.o
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.

The directory /usr/local/opt/llvm/lib/c++ does not exist on my computer. It's an Macbook pro with an M2 pro running 14.4.1 (23E224)

JacksonCampolattaro, 2024-05-05:
@rkarlsba It might be worth trying the glib patch anyway; in my case changing the LDFLAGS wasn't necessary.

rkarlsba, 2024-05-05:
@JacksonCampolattaro thanks, that worked, but I'm quite sure it might break other things later. Perhaps it'd be better to fix the source of dune3d? ;)

rkarlsba, 2024-05-05:
#ifdef __APPLE__
# warn "fixing stupid macOS problem"
// do something smart
#endif // __APPLE__

JacksonCampolattaro, 2024-05-05:
As far as I can tell, the issue is actually with brew's distribution of Glib and not Dune3d itself! Changing the headers won't affect anything that's already compiled, but it might prevent this same bug from breaking other builds (I can't think of any scenarios where a program would _depend_ on the headers being mismatched with the dylibs). 

I've been using Gtk/Gtkmm on macos for a while and I haven't encountered this before, but I think that's only because I've never needed to use a `guint64`.

rkarlsba, 2024-05-05:
I wonder why they don't use a uint64_t

rkarlsba, 2024-05-05:
(that is, int64_t, sorry)

JacksonCampolattaro, 2024-05-05:
Historical reasons, actually -- Gtk came before fixed-width types! As for why they haven't changed the typedefs, I couldn't say.

lessbones, 2024-05-06:
anybody got a binary for mac intel they could share?  I'm getting this error when trying to compile:

`Run-time dependency gtk4 found: NO (tried framework and cmake)

meson.build:12:7: ERROR: Dependency lookup for gtk4 with method 'pkgconfig' failed: Could not generate cflags for gtk4:
Package 'gobject-2.0' requires 'libffi >= 3.0.0' but version of libffi is 2.1
`
I've tried

`export PKG_CONFIG_PATH=/usr/local/Cellar/libffi/3.4.6/lib/pkgconfig/`

to point to the newer version of libffi, but I have very little experience with compilation, so I don't really know what I'm doing.  But I love solvespace, and I've been hoping that they would update it to add chamfers/fillets etc. and this looks like essentially the true evolution of that program...

forivall, 2024-05-08:
With the information that the problem is homebrew's glib distribution, I ran `brew reinstall --build-from-source glib glibmm gtk4 gtkmm4`, and building works now! Intel mac, MacOS 14.4.1 .

[image]

lessbones, 2024-05-11:
@forivall I tried building the packages from scratch, and tried using your binary, but neither is working properly (on OS 14.2.1) I'm still getting the same error with the build, probably because I haven't linked the newly compiled versions to the right place... as for the binary, it opens and I can see the GUI, but nothing appears when I try to draw lines or shapes-- clearly still something GTK related as per the terminal output
[image]

lvillani, 2024-05-20:
I'm experiencing the same issues with bottled (precompiled) glibmm that others have reported. In my case I just needed to rebuild glibmm from sources (I didn't have to rebuild Gtk or any other package). The same work-around seems to fix the build in CI.

Here's the issue I opened on Homebrew's side: https://github.com/Homebrew/homebrew-core/issues/172204

JacksonCampolattaro, 2024-05-21:
> I'm experiencing the same issues with bottled (precompiled) glibmm that others have reported. In my case I just needed to rebuild glibmm from sources (I didn't have to rebuild Gtk or any other package). The same work-around seems to fix the build in CI.
> 
> Here's the issue I opened on Homebrew's side: [Homebrew/homebrew-core#172204](https://github.com/Homebrew/homebrew-core/issues/172204)

Thanks for opening that issue! I wasn't sure where to report it.

carrotIndustries, 2024-07-05:
The recent update to opencascade 7.8.1 requires fixing the link args in https://github.com/dune3d/dune3d/blob/1c0d90f1ba2eff47965b5ecbfe1b6705cff747ba/meson.build#L29

This is really tedious to debug using github actions alone. Can someone with a mac have a look at this? I guess it's mostly replacing `-lTKSTEP` with `-lTKDESTEP` and the like.

tostr7191, 2024-11-26:
I just tried to build current head `7b1dd70` on MacOS 15.1.1 and it worked for me. What I needed to do was this:

```
brew install gtkmm4 cairomm opencascade libepoxy pkgconf librsvg glm cmake meson eigen pygobject3 py3cairo

rm -rf ./build

./scripts/build_macos.sh
```

Maybe that helps someone.

carrotIndustries, 2025-03-15:
@viernullvier @skurfuerst  Seems like this is no longer an issue and can be closed?

viernullvier, 2025-03-15:
The linker step has been working reliably for me, the only thing I noticed a long time ago was that the build setup will become tainted when running `meson setup build` without properly pointing `CC` and `CXX` to the homebrew-provided clang binaries. Adding `--wipe` to the build script helps, but I don't think it's a good idea to make it a default.

As of today, all of the latest dependency versions distributed via homebrew seem to be compatible, so I guess this can be considered fixed for now.

closed 2025-03-17 by carrotIndustries