dune3d/dune3d #50: [Win] Black screen flickering when drawing menus when fullscreen on multi-monitor setups

biggestsonicfan, 2024-01-23:
It's a bit of a long title and I apologize but I'm encountering an issue where the screen flickers black when mousing over menus within dune3d on Windows. This is a fresh compile from MSYS2-MINGW64 on latest git 1d45ed443ef1cef063089e6691b7d9064393bd9d (as of this issue posting) following the instructions [here](https://docs.dune3d.org/en/latest/build-win32.html).

https://github.com/dune3d/dune3d/assets/9744201/8d1d5bc5-d3bf-4dcf-9733-dae5b206bb32

OBS goes black when the menus are on screen, but what I'm seeing is a lot of flickering, possibly due to the 144hz refresh rate of my monitor and the 60 fps capture of OBS.

If I turn off or disable my second monitor, there is no flickering on the 144hz display.

carrotIndustries, 2024-01-23:
Maybe try recording with a video camera so we get to see what you're seeing as well.

Seems like gtk bug to me. Try the Shadertoy demo from `tk4-demo` , it has a context menu in the text are. Do you see the same behavior there as well?

biggestsonicfan, 2024-01-23:
> Maybe try recording with a video camera so we get to see what you're seeing as well.

See attached:

https://github.com/dune3d/dune3d/assets/9744201/33340a5d-8ea8-4a7b-8e3c-6c69ffaebbe8

> Try the Shadertoy demo from tk4-demo , it has a context menu in the text are. Do you see the same behavior there as well?

I could not locate `tk4-demo` but the `gtk4-demo` application's "Shadertoy" demo worked fine. There was only a singular flicker when opening/closing a window, not navigating any boxes.

carrotIndustries, 2024-01-24:
> See attached:

Oh that's awful. Does it also happen with the right-click context menu?

Even though I don't like generic advice like this: Have you tried updating your GPU drivers? This has helped some folks with similar driver-related issues on Horizon EDA which uses Gtk 3.

What's the make and model of your GPU?

biggestsonicfan, 2024-01-24:
> Does it also happen with the right-click context menu?

Yes it does.

> Have you tried updating your GPU drivers?

I am on a fresh Win10 install less than 2 weeks old using latest drivers from two weeks ago.

I have updated to a more "up to date" driver and there was no change in the flickering.
> What's the make and model of your GPU?

I am running an AMD Radeon RX 6750 XT.

bob-u, 2024-02-01:
I've had something similar with nvidia drivers.  Try setting refresh rate to all of your displays to 60Hz and see if issue goes away.

biggestsonicfan, 2024-02-01:
> I've had something similar with nvidia drivers. Try setting refresh rate to all of your displays to 60Hz and see if issue goes away.

Yes that's one way to mitigate the issue but it doesn't really solve the problem. Simply turning off the second monitor as stated in the initial post is a quicker solution than adjusting settings while keeping my 144Hz refresh rate.

biggestsonicfan, 2024-03-30:
My phone's camera actually hasn't worked properly since recording this. I think the constant attempt to focus on the flashes killed some component. The flash and the actual shutter close no longer sync :(