dune3d/dune3d #51: [Win] Item toggle has odd behavior

biggestsonicfan, 2024-01-27:
If opening the sample project with latest git as of this issue 290d011468720513ed6dc8ab44bcbb82c914f9ec and I highlight a Chamfer item:
[image]

Then proceed to click an item above the highlighted item (not clicking the checkbox), all items between the two are untoggled
[image]

Next, clicking on a specific item on the checkbox of the untoggled items now re-toggles all items between the two...
[image]

I've also taken a video to show the behavior in motion
https://github.com/dune3d/dune3d/assets/9744201/6b1524bd-ba9d-400a-9440-ae0f9180893f

Should the default behavior of clicking an item toggle all items below it?

carrotIndustries, 2024-01-27:
This is works as intended: https://docs.dune3d.org/en/latest/usage.html#groups

When looking at a group, all groups after it are out of scope.

closed 2024-01-27 by carrotIndustries

biggestsonicfan, 2024-01-28:
I'd really wish that were an option specifically, it's very jarring to work with in my opinion. But then again, I haven't gotten quite far with the program.