dune3d/dune3d #52: Feature request: Ability to center/align

biggestsonicfan, 2024-01-29:
I've imported my own .step file and adjusted a workspace area, however when drawing a reference rectangle there appears to be zero way to align the rectangle to the center of the workspace.

[image]

Ideally, you should be able to take an item/object/point and select another item/object/point and tell it to align with the second point either vertically, horizontally, or both.

There just appears to be no way to precisely do this without drawing a bunch of constrains then praying your X and Y constraint lines match exactly with the middle.

carrotIndustries, 2024-01-29:
> however when drawing a reference rectangle there appears to be zero way to align the rectangle to the center of the workspace.

There is: In the draw rectangle tool press <kbd>m</kbd> before placing the first point. This switches it to the center/corner mode.

Alternatively, you can do the same by drawing a diagonal construction line and constraining its center to be on the workplane's origin.

biggestsonicfan, 2024-01-29:
This seemingly only works if you want a rectangle aligned along two axis, not one?

For example:
[image]

While the rectangle is now center with my part, the sketch is now longer than it needs to be on the pinout side. I can delete the construction line and adjust, but there is no way to fine tune the length without using the mouse, which adjusts both the x and y axis, I can't choose a single axis to adjust.

carrotIndustries, 2024-01-29:
Well, then you have to use a different approach:

[image]

It's the same thing as in any other parametric CAD.