dune3d/dune3d #55: std::exception: pick not found

tkornack, 2024-02-02:
Note: this could be macOS-specific. 

1. New document
2. Centered rectangle (place first at origin)
3. Right-click->constrain horizontal 
4. Enter a value (here, 5 mm)

Obtain "std::exception: pick not found" 

[image]

ruevs, 2024-02-02:
Works for me on Windows - self compiled ae9fdb41088 (which is version 1.0.0).

osresearch, 2024-02-24:
[image]

I'm getting this on Ubuntu 20.04 with the 1.0.0 flatpak installed as well.  I can click `Ignore` and keep working without crashing the tool.

carrotIndustries, 2024-02-24:
Can you check if this happens with latest `main` as well?