dune3d/dune3d #56: Radius for Chamfer/fillet is always 0.100 mm

mfdewit, 2024-02-02:
Hi, 

I'm on version 1.0 and have trouble with the Chamfer and Fillet tool; despite setting a higher radius it always reverts to 0.100 mm after selecting the edges and ending the tool with the right mouse click.  Is this a bug? Or is it me not understanding the tool?

(On a side note: I stumbled across this tool by accident on flathub and it looks really promising. It takes some getting used to as a Freecad user, specifically the cleaner interface. But so far, I really like it. Great job!)

ruevs, 2024-02-02:
You need to press `Enter` after typing the number. Then it is remembered. However chamfers and fillets still do not work for me.  I get a "not done" and they do not appear:
[image]

Perhaps I am also missing something. Here is an example model:
[ParallelogramChamferedNOT.zip](https://github.com/dune3d/dune3d/files/14139655/ParallelogramChamferedNOT.zip)

I've read https://docs.dune3d.org/en/latest/groups.html#fillet-chamfer

mfdewit, 2024-02-02:
Ahem, you're right. Somehow I didn't thought of pressing _enter_. Now it works correctly. Thank you!

Regrading your error: are you sure you model is large enough to apply the Chamfer of 5 mm? Judging from the size of your workplanes, it is not and you have a tiny model. Or can the workplanes be resized as well?

mfdewit, 2024-02-02:
I'll close this ticket now.

closed 2024-02-02 by mfdewit

ruevs, 2024-02-02:
> Regrading your error: are you sure you model is large enough to apply the Chamfer of 5 mm?

Indeed - ot works just fine!

Being used to [SolveSpace](https://solvespace.com) - which starts with about 120mm on the screen as a default zoom level for a new document - I did not even think to check the size of the model :-)

[ParallelogramChamferedFilleted.zip](https://github.com/dune3d/dune3d/files/14139945/ParallelogramChamferedFilleted.zip)

So presumably this issue can be closed?