dune3d/dune3d #57: [Mac] Selection picks wrong entity

viernullvier, 2024-02-02:
#### Steps to reproduce:

1. Draw a circle centered on origin
2. Draw a line from left intersection of circle with ZX plane to upper intersection of circle with YZ plane
3. Draw a line from upper intersection of circle with YZ plane to right intersection of circle with ZX plane
4. Try to click and delete both lines

### Expected behaviour:
- Clicking the left line highlights it, pressing Delete removes it
- Clicking the right line highlights it, pressing Delete removes it
- Clicking on any of the reference planes selects it

### Actual behaviour:
- Clicking the left line highlights it, pressing Delete removes it ✅
- Clicking the right line does not highlight anything ❌
- After pressing Delete, the _left_ line gets removed ❌

[image]

Note 1: It appears that some entities do in fact receive the hover highlight when hovering over presumably empty areas of the canvas. Clicking over there will actually select the highlighted entity.
Note 2: The erroneous behaviour might be related to the zoom level. Changing zoom level moves the invisible hover areas a bit.
Note 3: The issue mostly affects lines. Selecting points seems to be almost reliable, there are just some invisible hover areas that will select points as well.
Note 4: I was first suspecting that the issue might be related to high-DPI screens, but I could also reproduce it on an external screen at native resolution.
Note 5: At some point I was not able to select any of the reference planes either (it would select the circle instead), but I can't reproduce it reliably yet.

ruevs, 2024-02-03:
Selection is known to be... hard... on macOS since the line width can only be 1 pixel - see here:
https://github.com/dune3d/dune3d/pull/25#issuecomment-1875853026

But there could be other problems as well.

carrotIndustries, 2024-02-06:
The latest commit makes it possible to dump the pick buffer that is used to figure out what to select.

How to enable it:

 - In the `prefs.json` set `pick_path` as such:

````json
...
    "canvas": {
        "pick_path" : "/tmp/pick.txt",
...
````

After restarting the application, a "Pick" button should appear in the header bar. Clicking it will dump the contents of the pick buffer to the set path.

To show its contents, use gnuplot as such: `plot '/tmp/pickz.txt' matrix with image`

This should look roughly like this:
[image]

viernullvier, 2024-02-08:
@carrotIndustries Thank you, that's quite helpful already! I've set up a sample file, opened it on both Linux and Mac and resized the window to the smallest possible viewport.

Linux (working as expected, used as reference):

[image]

Mac, version 1.0ish:

[image]

Mac, latest `main` (I noticed that you've just introduced some changes to the pick logic):

[image]

The pick buffer dump explains some odd behaviours that I've been seeing. In the last dump, both the XZ and YZ planes have a colour that looks very similar to some of the circle's colours - and sure enough, hovering them highlights the circle instead. In the same vein, the right diagonal looks very similar to the left point; hovering the diagonal does indeed highlight that point.

carrotIndustries, 2024-02-09:
> hovering the diagonal does indeed highlight that point.

I think it makes sense to reduce the test case even more.

The pickbuffer works like that:

Each type of vertex gets drawn separately, these are:
 - Lines (including circles, since they're currently made of line segments)
 - Points (at the end of lines, etc)
 - Icons, used for constraint icons
 - 2D glyphs, used for constraint numbers such as distance
 - 3D glyphs, used for workplane labels

In each draw call, the pick buffer is set to the `gl_VertexID` plus an offset. The offset is used to disambiguate the individual vertex types since the vertex id will be 0 for both the first line and the first icon.
https://github.com/dune3d/dune3d/blob/d679ead3ad5c8a5cd46e6999ee2382c498f919d1/src/canvas/base_renderer.cpp#L92-L94

I think it makes sense to add some more printfs to dump the contents of `m_vertex_type_picks` and manually correlate them with the contents of the pickbuffer.

viernullvier, 2024-03-07:
OK, I think I've narrowed it down and actually fixed it. It looks like there's some antialiasing going on which doesn't cause issues on Linux or Windows, but leads to inaccurate pick buffer data on Mac.

Disabling MSAA via `glDisable(GL_MULTISAMPLE);` fixed the most severe aspects of the issue for me. Without multisampling, I'm now able to pick every entity _almost_ reliably. Will set up a PR now.

One remaining minor issue though: Picking lines will occasionally fail on certain zoom levels. I could reproduce it with an empty document and by trying to pick the XZ or YZ plane. Zooming in or out by a very tiny bit will make the line selectable again.

Side note: I wasn't really happy with gnuplot, so I tried to find a way to get the pick buffer data into GIMP. It turns out that the buffer dump is already in the right format to be parsed as a PGM bitmap, it just needs a header (and is flipped upside-down).

carrotIndustries, 2024-03-07:
> It looks like there's some antialiasing going on

Interesting, it uses nearest-neighbor for downsampling:

https://github.com/dune3d/dune3d/blob/6b14f51febf5a51bc6cd7345f4cdfe62828e7e94/src/canvas/canvas.cpp#L921-L922

Maybe that's broken on apple's GL implementation.

Can you show a closeup view of the pickbuffer for such a problem?

viernullvier, 2024-03-07:
> Can you show a closeup view of the pickbuffer for such a problem?

Sure! I've applied an arbitrary colour palette in order to highlight the effect.

Full pick buffer - MSAA active:
[image]
Detail 1:
[image]
Detail 2:
[image]

Full pick buffer - MSAA disabled:
[image]
Detail 1:
[image]
Detail 2:
[image]

PR #79 (2024-04-12): Canvas: Disable MSAA on Mac to fix picking issues
Fixes #57.

### The issue

On macOS, rendering the pick buffer is subject to antialiasing artifacts, causing the rendered pick indices to be blended instead of just passed through. As a result, placing the mouse over an entity might not actually select it, but a different entity (or none at all) depending on the blended pick index.

### The fix

This PR will disable multisampling specifically on macOS and therefore ensure that the pick indices will be passed accurately. Note that the lack of MSAA does not appear to degrade rendering quality.
closed 2024-04-12 by carrotIndustries