dune3d/dune3d #61: Symmetric constraint

fruchti, 2024-02-04:
I’ve been playing with Dune3D for a bit and found it overall quite nice! There naturally are some small rough edges (and I might open a few issues as soon as I can actually put into words what exactly I expect in these cases).

One of these is the missing symmetric constraint, which I use quite a lot in SolveSpace. It is of course not strictly necessary as you can do the same thing with construction lines and other constraints, but I personally find the symmetric constraint much more convenient.

carrotIndustries, 2024-02-11:
https://github.com/solvespace/solvespace/blob/master/exposed/DOC.txt#L343 which one of these were you thinking of?

bob-u, 2024-02-24:
Probably for this:
@6:49  https://www.youtube.com/watch?v=MVp8fF2LmM8&t=423s

valpackett, 2024-03-02:
> Probably for this

Hah, I've seen that video, Michael actually missed the existence of the symmetric constraint in FreeCAD..

> which one of these were you thinking of?

`SLVS_C_SYMMETRIC_LINE` and `SLVS_C_SYMMETRIC_HORIZ/VERT` (special cased in the solver, but UI-wise the axes of the plane we're sketching on are also lines) are the most necessary ones. `SLVS_C_SYMMETRIC` seems to be the 3D case.

fruchti, 2024-03-03:
> https://github.com/solvespace/solvespace/blob/master/exposed/DOC.txt#L343 which one of these were you thinking of?

Sorry for the late reply! I tried to look up which ones I used the most; I haven’t got *that may* SolveSpace projects, though.

```console
$ fd .slvs -x grep 'Constraint.type=60' {} \; | wc -l # SYMMETRIC
8
$ fd .slvs -x grep 'Constraint.type=61' {} \; | wc -l # SYMMETRIC_HORIZ
2
$ fd .slvs -x grep 'Constraint.type=62' {} \; | wc -l # SYMMETRIC_VERT
4
$ fd .slvs -x grep 'Constraint.type=63' {} \; | wc -l # SYMMETRIC_LINE
3
```

I checked the `SYMMETRIC` manually and all of them are in a sketch and using a plane as reference. These could just be `SYMMETRIC_HORIZ`/`SLVS_C_SYMMETRIC_VERT`.

commit 5437a8c1 (2024-04-10): add symmetry constraints

changelog: New Features/Editor: add symmetry constraints

closes #61
closed 2024-04-10 by carrotIndustries