dune3d/dune3d #62: Drawing lines on top of construction lines with constraints

fruchti, 2024-02-04:
When I draw a contour or line on top of an existing construction line, results are a bit unpredictable. Sometimes, segments seem to go missing, sometimes the whole workplane disappears:
[image]

This seems to only happen when the construction line has a horizontal/vertical constraint, so I guess this might be due to the automatically added constraints for the new line segment over-constraining it?

Drawing the new line next to the construction line and adding a coincident constraint next to it manually works as expected.

carrotIndustries, 2024-02-04:
Thanks for not finding this before the release, that would have caused some sleepless nights…

This particular sketch overconstrains the system which breaks my optimization to solve parts of it symbolically. Even after applying some duct-tape fixes the DOF calculation is still broken and it doesn't know anymore that it's overconstrained.

I'll have to see how to fix this while still keeping the required optimisation where needed.

blathers123, 2024-02-04:
@carrotIndustries just wanted to say thanks for your hard work on this and don't burn out trying to solve everything at once

ruevs, 2024-02-05:
> This particular sketch overconstrains the system which breaks my optimization to solve parts of it symbolically. Even after applying some duct-tape fixes the DOF calculation is still broken and it doesn't know anymore that it's overconstrained.

@carrotIndustries I'm not at the PC where I build DuneD3 but wondering - have you bisect-ed which of [these changes](https://github.com/dune3d/dune3d/commits/main/3rd_party/solvespace) that you made to SolveSpace breaks the DOF/Rank calculation?
I would say not acdb01a2630c0f085a98fc887e274f2c253fc1fe and not 2c8512f41caf06b3c6c27b5b5b72ce8ca3bccf7c.
It is possible that it is 16485b35e92fae1202d87a854f864fd43b479e00 or 74eea3740049c2329160f1419e343ea29e10d01e.

At any rate I think that you are doing unnecessary work here. Your `system.cpp` is "ancient". We started using Eigen in the solver [towards the end of 2021](https://github.com/solvespace/solvespace/commits/master/src/system.cpp) and it is now much, much faster and I suspect does not have the problems you were trying to solve any more.

I my opinion you should update your copy of SolveSpace.

P.S. Yesterday I saw the [slides for your presentation at Fosdem ](https://fosdem.org/2024/events/attachments/fosdem-2024-2763-dune-3d-the-making-of-a-maker-s-tool/slides/21665/presi_hdCy8cW.pdf) - nice :-)
Unfortunately I missed the live stream, so I'm looking forward to the recording to be uploaded somewhere.

ruevs, 2024-02-05:
Yes definitely "ancient". Your `system.cpp` is basically this https://github.com/solvespace/solvespace/blob/668fe6f493eb96e9b4add37b6dbe9b82070cd9cc/src/system.cpp and does not include any of the performance optimizations that went into the solver since then.

If you are interested in the history of how it happened this is a good overview https://github.com/solvespace/solvespace/issues?q=+label%3Aperformance+.

ruevs, 2024-02-05:
I updated SolveSpace to the latest one  - it was very easy and fixes this bug.

fruchti, 2024-02-06:
> Thanks for not finding this before the release, that would have caused some sleepless nights…

Oh well… Just to be clear, I’m nowhere near depending on Dune3D and am just experimenting a bit. Nothing I report comes with an expectation of you fixing it immediately (or at all). Take care, no need to feel pressured!

Without knowing anything about the code base: Would it be easier to detect this particular case and skip the automatic horizontal/vertical constraint for the newly drawn line or are there to many ways of automatic constraints producing a similar result? Sure, the underlying issue is still there but if you can only break the solver by adding constraints manually, it’s at least somewhat less pressing.

carrotIndustries, 2024-02-06:
The above fix is no panacea, but I currently can't think of a case where the remaining symbolically-solved equations should cause any problems. After all, these were the only equations this optimization was meant for.

> Would it be easier to detect this particular case and skip the automatic horizontal/vertical constraint

That's something that'll probably land when checking the solver result when adding constraints as it's the plan for reference dimensions.

closed 2024-02-06 by carrotIndustries

ruevs, 2024-02-08:
@carrotIndustries Sorry for my comments and the pull request. I was comparing "wrong" copies of system.cpp and this is why I though yours was "ancient". In fact it is current.

When I have some time to think carefully, I will look at your changes for solving equations with a single unknown symbolically. Perhaps they are worth back-porting into SolveSpace as well.

> The key difference between my usage of the solver and solvespace's usage is that solvespace doesn't generate solver equations for extrusions and arrays.

Out of curiosity I will also have to look at what type of extra equations you are adding and why.

commit c0e7a4cd (2024-02-06): solver: only symbolically solve equations from extrude/arrays

since these are the ones hurting performance

in some cases, this lead to equations being solved symbolically even if
it doesn't really make sense

changelog: Bugfixes/Solver: properly handle redundant constraints in some cases

fixes #62