dune3d/dune3d #69: Double clicking x-z plane only allows model to rotate around z axis

mrlinuxfish, 2024-02-09:
I clicking around to get used to the model rotations and found that double clicking the x-z plane functions differently to double clicking the x-y or x-z plane.

### Expected behaviour
Navigation after double clicking any of the plans functions identically to any other plane

### Actual behaviour
Double clicking the x-z plane only allows for rotation around the z axis until another plane is double clicked

https://github.com/dune3d/dune3d/assets/444722/f1775495-f331-4471-9fb4-cfa8f39536c7

carrotIndustries, 2024-02-10:
This is due to the navigation scheme, I've added some hints how to get out of this situation to the documentation: https://docs.dune3d.org/en/latest/usage.html

On the long run, we might explore other navigation schemes that hopefully don't have this problem.

Imarok, 2024-05-23:
This looks like using quaternions to represent the view orientation could fix it. ;)

carrotIndustries, 2024-05-23:
> This looks like using quaternions to represent the view orientation could fix it. ;)

It's already using quaternions, the problem is just in the translation from mouse movements to quaternions.

carrotIndustries, 2024-05-24:
Try this branch: https://github.com/dune3d/dune3d/tree/navigation2 and tell me if that navigation scheme makes more sense.

Imarok, 2024-05-26:
> Try this branch: https://github.com/dune3d/dune3d/tree/navigation2 and tell me if that navigation scheme makes more sense.

That feels better.

commit a3fc0d3d (2024-06-09): Canvas: avoid gimbal-lock like situation when rotating

by not making the rotation axis absolute

the old behaviour still is available as the legacy rotation scheme

closes #69
closed 2024-06-09 by carrotIndustries