dune3d/dune3d #80: Non-integer angle constraint after editing

Cjen1, 2024-03-08:
The input field for the angle constraint changes non-integer inputs into integers.
The constraint itself seems to support non-integer constraints, however upon editing them the decimal places are removed.

viernullvier, 2024-03-10:
Had a quick look at this - the issue is here:
https://github.com/dune3d/dune3d/blob/main/src/widgets/spin_button_angle.cpp#L52

Even though the input value is parsed as a `double`, it's converted to an `int64_t` before being passed as a `double` again. Changing the type of `va` to `double` would fix the issue, but then again I'm not familiar enough with the codebase to understand why the extra roundtrip through a local variable happens in the first place. @carrotIndustries are you able to shed some light here?

carrotIndustries, 2024-03-10:
> are you able to shed some light here?

Copy/paste mistake from Horizon EDA: https://github.com/horizon-eda/horizon/blob/master/src/widgets/spin_button_angle.cpp#L47

commit 73033ce8 (2024-03-10): SpinButtonAngle: keep digits after the comma

changelog: Bugfixes/Editor: keep digits after the comma when entering angles

fixes #80
closed 2024-03-10 by carrotIndustries