dune3d/dune3d #81: Cannot extrude several shapes at one ?

qntnlq, 2024-03-08:
Hello everyone,

I'm only starting to learn how to use Dune3D so maybe I'm just making a mistake here.
Trying to create a token for one of my board games, I created several shapes in a workplane to extrude. However, only one of them seems to be able to get extruded. 
[image]
Is it a software limitation that one can only extrude one shape at a time, or am I just missing something ?

In case it may be of use, I've linked the file. I know I have a redundant constraint (at least one ^^) but I've haven't been able to identify it. Thus, a subsidiary question, is the highlighting of redundant constraints a planned feature ? It would be of great use for beginners like me ^^

Thanks in advance !

[patissorciers_token.zip](https://github.com/dune3d/dune3d/files/14542454/patissorciers_token.zip)

carrotIndustries, 2024-03-08:
Are you using the released version? That should be fixed by https://github.com/dune3d/dune3d/commit/f777807b41c4dd87c25fe9cddf7e65adc06924dc here's how it looks on latest `main` 
[image]

Finding redundant constraints is also already on latest main, check "show only redundant constraints" in the constraints list.

qntnlq, 2024-03-08:
I'm using the flathub verison. Is it up to date ?

closed 2024-03-09 by qntnlq