dune3d/dune3d #82: Fillet fails on specific edges for unknown reasons

viernullvier, 2024-03-10:
### Steps to reproduce

**Note:** This is still WIP, I've encountered this problem on two nontrivial models but haven't created a minimal example yet.

- Create a model with a couple of (curved) edges
- Add a new fillet group
- Use the edge picker to select one edge, a couple of edges, all edges

### Expected behaviour

- Fillets should be applied to all edges (where it reasonably makes sense)

### Actual behaviour

- When selecting a single "unproblematic" edge, the fillet gets applied correctly ✅
- When selecting multiple "unproblematic" edges, the fillet gets applied correctly as well ✅
- When selecting a single "problematic" edge, the fillet won't be applied with the error hint "Not done" ❌
- When selecting a single "problematic" edge in addition to other "unproblematic" edges, no fillets will be applied at all with the error hint "Not done" ❌

### Example

Here's the first model in which I could reproduce the problem reliably. Each of the eight notches should have the same fillet applied, but one of them fails reliably.

Source document:
[oscoin.d3ddoc.json](https://github.com/dune3d/dune3d/files/14549872/oscoin.d3ddoc.json)

Screenshot - unproblematic edges only:
[image]

Screenshot - with the problematic edge added:
[image]

viernullvier, 2024-03-16:
Minor update on the document I uploaded above: It appears that the fillet will only fail if the offending notch is exactly aligned with the YZ plane. Displacing the polar array a tiny bit or rotating all notches to keep them from intersecting the planes will get rid of this issue, which makes it even weirder.

dwbrite, 2024-09-21:
Fillets are thin wrappers around OCCT, so certain issues probably can't be worked around.
I've found a few more fillet-breaking issues that result in "not done". These are all best guesses based on playing around with my issues.

- bezier curves where the control points are on anchor points. I can imagine the zero-length vector could muss up the math there, or floating point errors could create non-manifold extrusions. I imagine this could be detectable and warned about by Dune3D.
- fillets with a larger radius than the extrusion you're working with. Pretty obvious when fillets work at certain radii and not at others.

smeikx, 2026-10-28:
This problem still persists. I even encountered a glitch while trying to figure out how to reproduce this bug:

[image]

This hole was not just a rendering glitch, it also existed in the exported STL.

Also, I failed in my initial attempt of detecting any pattern. Sometimes fillets could be created, sometimes not, sometimes only if the radius was small enough.

This is the test file I created: [failing-fillet.d3ddoc.json](https://github.com/user-attachments/files/34087002/failing-fillet.d3ddoc.json)

I had the most success reproducing the bug with the inner vertical edges of the two extrusions (see selection):

[image]

carrotIndustries, 2026-10-28:
Exported the group before the failing fillet as a STEP, imported that one into FreeCAD and applied the fillet there:

[image]

same result, so this is an opencascade issue