dune3d/dune3d #84: Distances somtimes get clipped behind solid objects

yonatan8070, 2024-03-15:
Distances can get clipped if a solid object is in the way. For example, I made a cube, extruded it, and then extruded a new cube into it to make a hole, I then applied a distance constraint to one of the vertical lines of the hole, and the number can get partially or fully clipped behind the solid. Here are screenshots:

Text is partially clipped from here
[image]

Text is entirely invisible from here
[image]

I know that the constraint can be moved, but until I figured that out, the default position for it was hidden from almost any direction
[image]

carrotIndustries, 2024-03-15:
Yeah, this is low-key bothering me as well, but so far I couldn't come up with a good solution to fix this problem

In the meanwhile, the recommended workaround is to temporarily turn off the solid model and move the dimension somewhere it's not clipped.

yonatan8070, 2024-03-16:
I don't know if this is possible with Dune3D's architecture. Maybe you could move all the constraints, points, dimensions etc. to their own rendering layer and overlay that on top of the solid graphics, and while rendering have a check for whether the object a constraint is attached to is visible, and only render the constraint if it is

dwbrite, 2024-09-23:
Until a rendering solution can be figured out, you can use alt+s to disable solid geometry.