dune3d/dune3d #87: error with shader compilation under Arch linux

Sivecano, 2024-03-25:
application crashes on startup when trying to run on archlinux using the `nvidia-open` driver with the following error:
```
❯ dune3d

(dune3d:17553): Gtk-WARNING **: 02:43:38.733: Theme parser error: gtk.css:5604:3-9: No property named "height"
realize
Compile failure in vertex shader: 0(2) : error C0201: unsupported version 330
0(3) : error C7532: global type vec2 requires "#version 100" or later
0(4) : error C7532: global type vec3 requires "#version 100" or later
0(4) : error C5060: out can't be used with non-varying color_to_fragment
0(9) : error C7532: global variable gl_VertexID requires "#version 130" or later
0(9) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
0(10) : error C7532: global function pow requires "#version 100" or later
0(15) : error C7532: global variable gl_Position requires "#version 100" or later
0(15) : error C7532: global type vec4 requires "#version 100" or later
 
Compile failure in vertex shader: 0(2) : error C0201: unsupported version 330
0(3) : error C7532: global type vec3 requires "#version 100" or later
0(7) : error C5060: out can't be used with non-varying normal_to_fragment
0(8) : error C5060: out can't be used with non-varying color_to_fragment
0(9) : error C5060: out can't be used with non-varying pos_to_fragment
0(11) : error C7532: global type mat4 requires "#version 100" or later
0(14) : error C7532: global type mat3 requires "#version 100" or later
0(21) : error C7532: global function isnan requires "#version 130" or later
0(23) : error C7532: global type vec4 requires "#version 100" or later
0(26) : error C7532: global variable gl_Position requires "#version 100" or later
0(28) : error C7532: global function normalize requires "#version 100" or later
 
Compile failure in vertex shader: 0(2) : error C0201: unsupported version 330
0(3) : error C7532: global type vec3 requires "#version 100" or later
0(4) : error C7532: global type uint requires "#version 130" or later
0(4) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
0(5) : error C7532: global type mat4 requires "#version 100" or later
0(7) : error C7532: global type float requires "#version 100" or later
0(9) : error C5060: out can't be used with non-varying pick_to_frag
0(10) : error C5060: out can't be used with non-varying color_to_frag
0(24) : error C0000: syntax error, unexpected '(', expecting "::" at token "("
0(32) : error C1503: undefined variable "colors"
0(32) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '|' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '|' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '|' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '|' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '|' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(46) : error C7548: '>>' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(46) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(46) : error C7011: implicit cast from "uint" to "float"
0(47) : error C7548: '>>' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(47) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(47) : error C7011: implicit cast from "uint" to "float"
0(48) : error C7548: '>>' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(48) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(48) : error C7011: implicit cast from "uint" to "float"
0(49) : error C7548: '>>' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(49) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(49) : error C7011: implicit cast from "uint" to "float"
0(57) : error C7532: global variable gl_Position requires "#version 100" or later
0(57) : error C7532: global type vec4 requires "#version 100" or later
0(58) : error C7532: global variable gl_VertexID requires "#version 130" or later
0(58) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
0(58) : error C7532: global type int requires "#version 100" or later
 
Compile failure in vertex shader: 0(2) : error C0201: unsupported version 330
0(2) : error C0000: syntax error, unexpected '(', expecting "::" at token "("
[... 56 lines cut ...]
0(8) : error C7532: global type vec4 requires "#version 100" or later
0(8) : error C5060: out can't be used with non-varying origin_to_geom
0(9) : error C7532: global type vec2 requires "#version 100" or later
0(9) : error C5060: out can't be used with non-varying shift_to_geom
0(10) : error C7532: global type float requires "#version 100" or later
0(10) : error C5060: out can't be used with non-varying scale_to_geom
0(11) : error C7532: global type uint requires "#version 130" or later
0(11) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
0(11) : error C5060: out can't be used with non-varying flags_to_geom
0(12) : error C5060: out can't be used with non-varying pick_to_geom
0(13) : error C5060: out can't be used with non-varying bits_to_geom
0(16) : error C7532: global type mat4 requires "#version 100" or later
0(20) : error C7532: global variable gl_VertexID requires "#version 130" or later
0(20) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
0(20) : error C7532: global type int requires "#version 100" or later
0(21) : error C1503: undefined variable "flags"
0(22) : error C1503: undefined variable "origin"
0(23) : error C1503: undefined variable "shift"
0(24) : error C1503: undefined variable "bits"
0(25) : error C1503: undefined variable "scale"
 
gl error 1281 in ../src/canvas/glyph_renderer.cpp:44
[1]    17553 IOT instruction (core dumped)  dune3d
```
I might look into why this is happening. some preliminary googling suggests that the OpenGL version in the context may not be hinted correctly and defaults to like OpenGL 1.0. seems like a relatively easy fix, just haven't found where yet.

carrotIndustries, 2024-04-10:
Did this get fixed by 14e1a85910a837840b54473394f1c6a7e53f71cb?

nameless-tee, 2024-05-01:
The issue persists. I'm trying with [69e58bc](https://github.com/dune3d/dune3d/commit/69e58bc896aba65f3289b50ecb61afcf0d4e98af), also on Arch, but with the less open Nvidia driver. I tried removing the GTK version check from the commit you mentioned and added `set_required_version(3, 3)`. Neither changed anything.

roobre, 2024-05-05:
I can also reproduce this on 69e58bc, with the closed-source nvidia driver (no `nvidia-run`).

carrotIndustries, 2024-05-05:
> added set_required_version(3, 3)

Where? It needs to be done before creating the GL context, so the same place as `gtk_gl_area_set_allowed_apis` should work.

Try running Dune 3D with the `GDK_DEBUG=opengl` environment variable set and paste the output.

novaplusplus, 2024-05-06:
Running Fedora 39 here and getting the same problem with the flatpak version. Here's the output of trying to run it with the GDK_DEBUG env set:

`EGL API version 1.5 found
 - Vendor: NVIDIA
 - Version: 1.5
 - Device: /dev/dri/renderD128, 226 128
 - Client APIs: OpenGL_ES OpenGL
 - Extensions:
	EGL_ANDROID_native_fence_sync
	EGL_EXT_buffer_age
	EGL_EXT_client_sync
	EGL_EXT_create_context_robustness
	EGL_EXT_image_dma_buf_import
	EGL_EXT_image_dma_buf_import_modifiers
	EGL_EXT_output_base
	EGL_EXT_present_opaque
	EGL_EXT_stream_acquire_mode
	EGL_EXT_sync_reuse
	EGL_IMG_context_priority
	EGL_KHR_config_attribs
	EGL_KHR_context_flush_control
	EGL_KHR_create_context
	EGL_KHR_create_context_no_error
	EGL_KHR_fence_sync
	EGL_KHR_get_all_proc_addresses
	EGL_KHR_gl_colorspace
	EGL_KHR_gl_renderbuffer_image
	EGL_KHR_gl_texture_2D_image
	EGL_KHR_gl_texture_3D_image
	EGL_KHR_gl_texture_cubemap_image
	EGL_KHR_image
	EGL_KHR_image_base
	EGL_KHR_image_pixmap
	EGL_KHR_no_config_context
	EGL_KHR_partial_update
	EGL_KHR_reusable_sync
	EGL_KHR_stream
	EGL_KHR_stream_attrib
	EGL_KHR_stream_consumer_gltexture
	EGL_KHR_stream_cross_process_fd
	EGL_KHR_stream_fifo
	EGL_KHR_stream_producer_eglsurface
	EGL_KHR_surfaceless_context
	EGL_KHR_swap_buffers_with_damage
	EGL_KHR_wait_sync
	EGL_MESA_image_dma_buf_export
	EGL_NV_nvrm_fence_sync
	EGL_NV_output_drm_flip_event
	EGL_NV_post_sub_buffer
	EGL_NV_quadruple_buffer
	EGL_NV_robustness_video_memory_purge
	EGL_NV_stream_attrib
	EGL_NV_stream_consumer_eglimage
	EGL_NV_stream_consumer_gltexture_yuv
	EGL_NV_stream_cross_display
	EGL_NV_stream_cross_object
	EGL_NV_stream_cross_process
	EGL_NV_stream_cross_system
	EGL_NV_stream_dma
	EGL_NV_stream_fifo_next
	EGL_NV_stream_fifo_synchronous
	EGL_NV_stream_flush
	EGL_NV_stream_metadata
	EGL_NV_stream_origin
	EGL_NV_stream_remote
	EGL_NV_stream_reset
	EGL_NV_stream_socket
	EGL_NV_stream_socket_inet
	EGL_NV_stream_socket_unix
	EGL_NV_stream_sync
[... 151 lines cut ...]
0(80) : error C1503: undefined variable "p2"
 
Compile failure in vertex shader: 0(2) : error C0201: unsupported version 330
0(2) : error C0000: syntax error, unexpected '(', expecting "::" at token "("
0(8) : error C7532: global type vec4 requires "#version 100" or later
0(8) : error C5060: out can't be used with non-varying origin_to_geom
0(9) : error C7532: global type vec2 requires "#version 100" or later
0(9) : error C5060: out can't be used with non-varying shift_to_geom
0(10) : error C7532: global type float requires "#version 100" or later
0(10) : error C5060: out can't be used with non-varying scale_to_geom
0(11) : error C7532: global type uint requires "#version 130" or later
0(11) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
0(11) : error C5060: out can't be used with non-varying flags_to_geom
0(12) : error C5060: out can't be used with non-varying pick_to_geom
0(13) : error C5060: out can't be used with non-varying bits_to_geom
0(16) : error C7532: global type mat4 requires "#version 100" or later
0(20) : error C7532: global variable gl_VertexID requires "#version 130" or later
0(20) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
0(20) : error C7532: global type int requires "#version 100" or later
0(21) : error C1503: undefined variable "flags"
0(22) : error C1503: undefined variable "origin"
0(23) : error C1503: undefined variable "shift"
0(24) : error C1503: undefined variable "bits"
0(25) : error C1503: undefined variable "scale"
`

carrotIndustries, 2024-05-06:
> edit: I just compiled it myself and it actually launched without issue, so evidently something between the latest release and the latest commit solved it at least for my situation.

This makes it look more like a flatpak problem then. There are no relevant commits since version 1.1: https://github.com/dune3d/dune3d/commits/main/

nameless-tee, 2024-05-06:
On my machine it works when I launch it with `GDK_DEBUG="gl-prefer-gl"`. Updating GTK to `4.14.4` did nothing.

This may be an issue with GTK (like most of my computer issues for some reason). When I request an OpenGL 3.3 context in a minimal GTK4 application, it will create an openGL ES 3.2 context instead and replace the GLArea with an error message in the GUI:
https://gist.github.com/nameless-tee/985b7712a66b6e6cda577f8cc9c9d223

There is already an issue on the GTK issue tracker:
https://gitlab.gnome.org/GNOME/gtk/-/issues/6423

darkbox, 2024-05-06:
Just came here to confirm the same issue. Flatpak gives a compile error:
```
Compile failure in vertex shader: 0(2) : error C0201: unsupported version 330
0(3) : error C7532: global type vec2 requires "#version 100" or later
0(4) : error C7532: global type vec3 requires "#version 100" or later
0(4) : error C5060: out can't be used with non-varying color_to_fragment
0(9) : error C7532: global variable gl_VertexID requires "#version 130" or later
0(9) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
0(10) : error C7532: global function pow requires "#version 100" or later
0(15) : error C7532: global variable gl_Position requires "#version 100" or later
...
```

On Ubuntu 24.04 - Wayland.

Adding `GDK_DEBUG=gl-prefer-gl` as @nameless-tee fix the issue for me.

carrotIndustries, 2024-05-06:
Can someone of you try the above branch? This moves the call to `gtk_gl_area_set_allowed_apis` to the ctor, maybe the current location was too late.

carrotIndustries, 2024-05-06:
Now with some more debug output on that branch.

nameless-tee, 2024-05-06:
> Now with some more debug output on that branch.

`get_context()` returns a `std::shared_ptr` pointing at null, causing it to crash before outputting anything new.

```
realize
Creating EGL context version 3.2 (debug:no, forward:no, legacy:no, es:no)
Creating EGL context version 3.0 (debug:no, forward:no, legacy:yes, es:no)

Thread 1 "dune3d" received signal SIGSEGV, Segmentation fault.
0x00007ffff6d787d8 in Gdk::GLContext::is_legacy() const () from /usr/lib/libgtkmm-4.0.so.0
(gdb) bt
#0  0x00007ffff6d787d8 in Gdk::GLContext::is_legacy() const () at /usr/lib/libgtkmm-4.0.so.0
#1  0x00005555556cfdd5 in dune3d::Canvas::on_realize (this=0x555556bc9b10) at ../src/canvas/canvas.cpp:662
#2  0x00007ffff6d09c2f in Gtk::Widget_Class::realize_callback(_GtkWidget*) () at /usr/lib/libgtkmm-4.0.so.0
#3  0x00007ffff6f848d3 in ??? () at /usr/lib/libgobject-2.0.so.0

```

nameless-tee, 2024-05-06:
Here is my truly terrible workaround in case anyone is desperate enough:
https://gist.github.com/nameless-tee/da554375b2a9c9daf9f36db68e4bf9c5

carrotIndustries, 2024-05-07:
Might be a tall order, but if I had a suitable nvidia machine at hand I'd figure out what exactly that environment variable is doing to fix it by debugging gtk/adding printfs. Just staring at the gtk source code didn't get me any further.

Either there's some API of GtkGLArea we're not using correctly or a bugfix for gtk will fall out of it. Wouldn't be the first time that one of my CAD applications resulted in patches in Gtk's OpenGL machinery...

luigim81, 2024-05-10:
Same error here on Fedora 40 with GTX1080 nvidia propitary drivers...

```
User Logs:
--Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(46) : error C7548: '>>' requires "#extension GL_EXT_gpu_shader4 : enable" before use
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(46) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(46) : error C7011: implicit cast from "uint" to "float"
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(47) : error C7548: '>>' requires "#extension GL_EXT_gpu_shader4 : enable" before use
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(47) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(47) : error C7011: implicit cast from "uint" to "float"
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(48) : error C7548: '>>' requires "#extension GL_EXT_gpu_shader4 : enable" before use
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(48) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(48) : error C7011: implicit cast from "uint" to "float"
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(49) : error C7548: '>>' requires "#extension GL_EXT_gpu_shader4 : enable" before use
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(49) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(49) : error C7011: implicit cast from "uint" to "float"
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(57) : error C7532: global variable gl_Position requires "#version 100" or later
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(57) : error C7532: global type vec4 requires "#version 100" or later
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(58) : error C7532: global variable gl_VertexID requires "#version 130" or later
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(58) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(58) : error C7532: global type int requires "#version 100" or later
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]:
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: Compile failure in vertex shader: 0(2) : error C0201: unsupported version 330
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(2) : error C0000: syntax error, unexpected '(', expecting "::" at token "("
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(6) : error C7532: global type vec4 requires "#version 100" or later
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(6) : error C5060: out can't be used with non-varying p1_to_geom
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(7) : error C5060: out can't be used with non-varying p2_to_geom
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(8) : error C7532: global type uint requires "#version 130" or later
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(8) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(8) : error C5060: out can't be used with non-varying flags_to_geom
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(9) : error C5060: out can't be used with non-varying pick_to_geom
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(12) : error C7532: global type mat4 requires "#version 100" or later
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(14) : error C7532: global type float requires "#version 100" or later
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(28) : error C0000: syntax error, unexpected '(', expecting "::" at token "("
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(34) : error C7532: global type vec3 requires "#version 100" or later
Mai 10 11:36:13 fedora org.gnome.Software.desktop[65934]: 0(36) : error C1503: undefined variable "colors"
```

carrotIndustries, 2024-08-31:
From the [Gtk bugtracker](https://gitlab.gnome.org/GNOME/gtk/-/issues/6423#note_2211856)

> If this still occurs with 4.15.x, please reopen.

Can someone check if that's the case?

roobre, 2024-09-04:
Will try to check as soon as 4.15 lands on the arch repos, for now we're still on 4.14:
```
16:52:59 ~ #> pacman -Syi gtk4 | grep Version
Version         : 1:4.14.5-1
```

roobre, 2024-09-29:
Just tested 19254cc355b481f90e45a449966d44d117c6c198 (via https://aur.archlinux.org/packages/dune3d, haven't checked if the aur package has any relevant patches) with gtk4 4.16.2-1, same thing unfortunately. The `GDK_DEBUG=gl-prefer-gl` keeps working just fine thankfully ✨

[dune3d.log](https://github.com/user-attachments/files/17178781/dune3d.log)

vmoriaux, 2024-11-09:
Same issue with Linux Mint 22, Dune 3D flatpak v1.2.0.

jfsgithub2020, 2025-01-13:
In case it is still of interest, Debian Bookworm shows the same issue, with the same possible fix. 
GDK_DEBUG=gl-prefer-gl

padcom, 2025-03-29:
1.3.0 on Linux mint with NVidia driver 570 - still the same problem, `GDK_DEBUG=gl-prefer-gl` still works

carrotIndustries, 2025-03-30:
Can someone confirm that this still happens with Gtk 4.18?

nadiamoe, 2025-04-13:
I can still reproduce this with Gtk 4.18:
```
16:24:35 ~ #> pacman -Qi gtk4
Name            : gtk4
Version         : 1:4.18.4-1
Description     : GObject-based multi-platform GUI toolkit
Architecture    : x86_64
```

[dune3d.log](https://github.com/user-attachments/files/19725747/dune3d.log)

carrotIndustries, 2025-04-16:
From the Gtk bugtracker: https://gitlab.gnome.org/GNOME/gtk/-/issues/6423#note_2414627

can someome give this a try? https://discourse.gstreamer.org/t/glshader-with-glsl-version-330/3227/2

jue89, 2025-04-24:
> can someome give this a try? https://discourse.gstreamer.org/t/glshader-with-glsl-version-330/3227/2

When you create a branch with that change, I can try it!

carrotIndustries, 2025-04-25:
Thanks, try this branch: https://github.com/dune3d/dune3d/tree/glsl-core

jue89, 2025-04-28:
> Thanks, try this branch: https://github.com/dune3d/dune3d/tree/glsl-core

Different error:

```
jue@baxxter ~> nix run github:dune3d/dune3d/glsl-core

(dune3d:5174): Gtk-WARNING **: 19:15:54.832: Unknown key Settings in /home/jue/.config/gtk-4.0/settings.ini

(dune3d:5174): Gtk-WARNING **: 19:15:54.834: Theme parser error: gtk.css:5:1-133: Failed to import: Error opening file /nix/store/3cj6gg0pzh786iq1lp0yy2rca6ryyl64-gnome-themes-extra-3.28/share/themes/Adwaita-dark/gtk-4.0/gtk.css: No such file or directory
connect failed: No such file or directory
realize
Compile failure in fragment shader: 0(108) : error C7531: global variable gl_FragDepth requires "#extension GL_NV_fragdepth : enable" before use
 
Compile failure in fragment shader: 0(14) : error C7531: global variable gl_FragDepth requires "#extension GL_NV_fragdepth : enable" before use
 
Compile failure in fragment shader: 0(22) : error C7531: global function dFdx requires "#extension GL_OES_standard_derivatives : enable" before use
 
gl error 1.281 in ../src/canvas/glyph_renderer.cpp:44
fish: Job 1, 'nix run github:dune3d/dune3d/gl…' terminated by signal SIGABRT (Abort)

```

Sanghalap, 2025-06-15:
For now I'm using a workaround by adding --env=GDK_DEBUG=gl-prefer-gl to the flatpak command :

`/usr/bin/flatpak run --branch=stable --arch=x86_64 --env=GDK_DEBUG=gl-prefer-gl --command=dune3d --file-forwarding org.dune3d.dune3d @@u %U @@`

carrotIndustries, 2026-02-01:
> Confirmed on just compiled "einstein" release.

This in a Gtk bug. So watch the issue over in the Gtk bugtracker: https://gitlab.gnome.org/GNOME/gtk/-/issues/6423