dune3d/dune3d #88: Quits before anything displayed running 1.0.0 Win 10 Pro 22H2

garyphilip, 2024-03-28:
I Installed today using the MSI.

When I run it, it gets as far as briefly creating a taskbar button and showing up in Task Manager for about a second, then quits.

I restarted Windows, uninstalled and reinstalled; same behavior. Tried running from a command prompt, and running as admin.

Perhaps if there is some debug flag I could add to the command I might get more info?

Was really looking forward to trying this out!

carrotIndustries, 2024-03-28:
Are you running it in a VM by any chance? Things like this usually happen when it's not happy with the OpenGL implementation.

To get further debug output, run Dune 3D from an MSYS2 shell.

garyphilip, 2024-03-28:
OpenSCAD gives me this message at start-up:

> Warning: Missing OpenGL capabilities for OpenCSG - OpenCSG has been disabled.
> 
> It is highly recommended to use OpenSCAD on a system with OpenGL 2.0 or later.
> Your renderer information is as follows:
> GLEW version 1.12.0
> GDI Generic (Microsoft Corporation)
> OpenGL version 1.1.0

But it is still entirely usable.

I tried running dune3d from a MSYS2 shell, as suggested, I received the following output:

> Gary@NEW-MAIN UCRT64 /c/Program Files/Dune 3D
> # ./dune3d.exe
> 
> ** (process:13476): WARNING **: 07:26:28.881: Can't set the global locale to the user's preferred locale.
>    locale::facet::_S_create_c_locale name not valid
>    The environment variable LANG may be wrong.
> 
> 
> ** (process:13476): WARNING **: 07:26:29.461: Can't make the global C++ locale equal to the C locale.
>    locale::facet::_S_create_c_locale name not valid
>    C locale = English_United States.1252
> 
> 
> (dune3d.exe:13476): GLib-GIO-WARNING **: 07:26:29.465: win32 session dbus binary not found
> Implementation unexpectedly missing WGL_ARB_extensions_string.  Probably a libepoxy bug.
> Implementation unexpectedly missing WGL_ARB_extensions_string.  Probably a libepoxy bug.
> Implementation unexpectedly missing WGL_ARB_extensions_string.  Probably a libepoxy bug.
> Implementation unexpectedly missing WGL_ARB_extensions_string.  Probably a libepoxy bug.
> 
> (dune3d.exe:13476): GLib-WARNING **: 07:26:29.655: GError set over the top of a previous GError or uninitialized memory.
> This indicates a bug in someone's code. You must ensure an error is NULL before it's set.
> The overwriting error message was: WGL version 1.1 is too low, need at least 3.0
> Gsk-Message: 07:26:29.656: Failed to realize renderer of type 'GskGLRenderer' for surface 'GdkWin32Toplevel': libEGL not available
> 
> realize
> Segmentation fault
> 
> Gary@NEW-MAIN UCRT64 /c/Program Files/Dune 3D
>

bob-u, 2024-03-28:
What GPU?  Check in device manager if your GPU is working properly.

garyphilip, 2024-03-28:
Hardware's just too old I suppose. 
[image]

carrotIndustries, 2024-03-28:
> WGL version 1.1 is too low, need at least 3.0

There's your problem. Dune 3D (and Gtk?) require at least OpenGL 3.0.

bob-u, 2024-03-29:
You have a very old Intel HD 2000 GPU. Good news is that it supports OpenGL3.1 bad news is that Intel released drivers for Windows 8, not 10.  You'll need to figure out if it's possible to install Intel drivers in your OS before trying Dune3D.

Release notes:
file:///C:/Users/Jamie/Downloads/releasenotes_gfx_15.28.24.4229.pdf
Drivers:
https://www.intel.com/content/www/us/en/download/17608/intel-graphics-driver-for-windows-15-28.html

garyphilip, 2024-03-29:
Thanks to you both for the prompt and kind attention!
yeah, it's an old Optiplex 990. I downloaded:
win64_152824.zip
Windows 8.1, 64-bit*, Windows 8, 64-bit*, Windows 7, 64-bit*
Size: 122.3 MB
From https://www.intel.com/content/www/us/en/download/17608/intel-graphics-driver-for-windows-15-28.html.
I unzipped and installed from Device Manager by browsing to and opening Graphics/igdlh64.inf. It said driver updated successfully.
Even after a reboot, the change did not show up.
Then I downloaded the CAB from http://download.windowsupdate.com/d/msdownload/update/driver/drvs/2016/05/200018908_7759dfe62847d7301f772b3405e49df16d841f7b.cab, installed, again I’m still at OpenGL version 1.1.0.

OpenSCAD says:
Warning: Missing OpenGL capabilities for OpenCSG - OpenCSG has been disabled.
It is highly recommended to use OpenSCAD on a system with OpenGL 2.0 or later.
Your renderer information is as follows:
GLEW version 1.12.0
GDI Generic (Microsoft Corporation)
OpenGL version 1.1.0

Oh, just checked back in device mgr, it says Windows has stopped this device because it has reported problems. (Code 43)

Time to give up. I think, with this hardware.

carrotIndustries, 2024-04-10:
Closing since this a hardware/driver issue.

closed 2024-04-10 by carrotIndustries