dune3d/dune3d #92: flatpak missing requirements libgtkmm-4.0, libsigc++-3.0, and opencascade-7.7

combs, 2024-05-05:
Hello! When I install the flatpak on Ubuntu 22.04, it fails to find a number of required libraries.

I found that it was missing libgtkmm-4.0 and was able to hack my way through compiling it with this guide (notice the errata disclosed in comments)

https://terminalroot.com/how-to-install-gtkmm-4-on-ubuntu-2204/

It is also missing libsigc-3.0.so.0, installable via the deb here:
https://launchpad.net/ubuntu/mantic/amd64/libsigc++-3.0-0/3.4.0-7

And then it reports a missing libTKernel.so.7.7, which is part of OpenCascade, I think. I tried to compile 7.7.0, the only listed 7.7.x version on the download section of OpenCascade, and it has compiler errors due to missing/mistaken imports for <limits>. There is a hidden 7.7.2 version available for download here:
https://github.com/Open-Cascade-SAS/OCCT/archive/refs/tags/V7_7_2.tar.gz

I am not that familiar with flatpaks, so am not sure of the best way to add these missing libraries as proper flatpak requirements.

combs, 2024-05-05:
after getting through all the requirements... there's some glibc problems with the flatpak binary:

```
./dune3d: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by ./dune3d)
./dune3d: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by ./dune3d)
./dune3d: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.31' not found (required by ./dune3d)
./dune3d: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by ./dune3d)
```

combs, 2024-05-05:
huh! I think my method of launching the binary failed to include some required paths. I've learned about the flatpak exported binary location:

```
/var/lib/flatpak/exports/bin/org.dune3d.dune3d
```

when I run that, it seems to find the libraries, glibc, etc., just fine, but yacks out with lots of shader errors (I am using nvidia RTX 3060):

```

realize
Compile failure in vertex shader: 0(2) : error C0201: unsupported version 330
0(3) : error C7532: global type vec2 requires "#version 100" or later
0(4) : error C7532: global type vec3 requires "#version 100" or later
0(4) : error C5060: out can't be used with non-varying color_to_fragment
0(9) : error C7532: global variable gl_VertexID requires "#version 130" or later
0(9) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
0(10) : error C7532: global function pow requires "#version 100" or later
0(15) : error C7532: global variable gl_Position requires "#version 100" or later
0(15) : error C7532: global type vec4 requires "#version 100" or later
 
Compile failure in vertex shader: 0(2) : error C0201: unsupported version 330
0(3) : error C7532: global type vec3 requires "#version 100" or later
0(7) : error C5060: out can't be used with non-varying normal_to_fragment
0(8) : error C5060: out can't be used with non-varying color_to_fragment
0(9) : error C5060: out can't be used with non-varying pos_to_fragment
0(11) : error C7532: global type mat4 requires "#version 100" or later
0(14) : error C7532: global type mat3 requires "#version 100" or later
0(21) : error C7532: global function isnan requires "#version 130" or later
0(23) : error C7532: global type vec4 requires "#version 100" or later
0(26) : error C7532: global variable gl_Position requires "#version 100" or later
0(28) : error C7532: global function normalize requires "#version 100" or later
 
Compile failure in vertex shader: 0(2) : error C0201: unsupported version 330
0(3) : error C7532: global type vec3 requires "#version 100" or later
0(4) : error C7532: global type uint requires "#version 130" or later
0(4) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
0(5) : error C7532: global type mat4 requires "#version 100" or later
0(7) : error C7532: global type float requires "#version 100" or later
0(9) : error C5060: out can't be used with non-varying pick_to_frag
0(10) : error C5060: out can't be used with non-varying color_to_frag
0(24) : error C0000: syntax error, unexpected '(', expecting "::" at token "("
0(32) : error C1503: undefined variable "colors"
0(32) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '|' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '|' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '|' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '|' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '|' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(32) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(46) : error C7548: '>>' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(46) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(46) : error C7011: implicit cast from "uint" to "float"
0(47) : error C7548: '>>' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(47) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(47) : error C7011: implicit cast from "uint" to "float"
0(48) : error C7548: '>>' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(48) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(48) : error C7011: implicit cast from "uint" to "float"
0(49) : error C7548: '>>' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(49) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
0(49) : error C7011: implicit cast from "uint" to "float"
0(57) : error C7532: global variable gl_Position requires "#version 100" or later
0(57) : error C7532: global type vec4 requires "#version 100" or later
0(58) : error C7532: global variable gl_VertexID requires "#version 130" or later
[... 59 lines cut ...]
Compile failure in vertex shader: 0(2) : error C0201: unsupported version 330
0(2) : error C0000: syntax error, unexpected '(', expecting "::" at token "("
0(8) : error C7532: global type vec4 requires "#version 100" or later
0(8) : error C5060: out can't be used with non-varying origin_to_geom
0(9) : error C7532: global type vec2 requires "#version 100" or later
0(9) : error C5060: out can't be used with non-varying shift_to_geom
0(10) : error C7532: global type float requires "#version 100" or later
0(10) : error C5060: out can't be used with non-varying scale_to_geom
0(11) : error C7532: global type uint requires "#version 130" or later
0(11) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
[... cut ...]
0(20) : error C7532: global variable gl_VertexID requires "#version 130" or later
0(20) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
0(20) : error C7532: global type int requires "#version 100" or later
0(21) : error C1503: undefined variable "flags"
0(22) : error C1503: undefined variable "origin"
0(23) : error C1503: undefined variable "shift"
0(24) : error C1503: undefined variable "bits"
0(25) : error C1503: undefined variable "scale"
 
gl error 1,281 in ../src/canvas/glyph_renderer.cpp:44
```

carrotIndustries, 2024-05-05:
>  it was missing libgtkmm-4.0 a

It is not: https://github.com/flathub/org.dune3d.dune3d/blob/master/gtkmm.json#L6

> And then it reports a missing libTKernel.so.7.7,

It's also included https://github.com/flathub/org.dune3d.dune3d/blob/master/org.dune3d.dune3d.json#L56

Seems like something's off with your flatpak installation.

> when I run that, it seems to find the libraries, glibc, etc., just fine, but yacks out with lots of shader errors (I am using nvidia RTX 3060):

That seems to be the same issues as in https://github.com/dune3d/dune3d/issues/87

combs, 2024-05-05:
> It is not: https://github.com/flathub/org.dune3d.dune3d/blob/master/gtkmm.json#L6
>
>
> It's also included https://github.com/flathub/org.dune3d.dune3d/blob/master/org.dune3d.dune3d.json#L56
>
> Seems like [...]

Yes, I think you're right. Sorry for the trouble.

closed 2024-05-05 by combs