dune3d/dune3d #95: Gtk 4.8 compatibility (Debian bookworm)

Tjoppen, 2024-05-06:
Hi. The build instructions say to install libgtkmm-4.0-dev but not which version. Debian bookwork ships 4.8.0-3, but Gtk::FileDialog and Gtk::AlertDialog are only available in version 4.10 which are currently limited to Debian testing (and unstable). So I'm humbly requesting Debian bookworm support.

Replacing Gtk::FileDialog with Gtk::FileChooserDialog on 4.8 would help. Not sure what Gtk::AlertDialog could be replaced with as I am not versed in Gtk.

Gtk::StringList::create() needs to be replaced with Gtk::StringList::create(std::vector<Glib::ustring>()).

carrotIndustries, 2024-05-06:
It was a deliberate decision to use the non-deprecated new dialog APIs. When starting a new project I use the latest and greatest of what's available at that point in time. The debian stable problem will fix itself over time.

In the meanwhile, you can use the flatpak.

closed 2024-05-06 by carrotIndustries

Tjoppen, 2024-05-07:
Fair enough