dune3d/dune3d #96: Point rendering issue w/ v1.1.0 Flatpak build on ChromeOS

aboyett, 2024-05-07:
## DescriptionN

When running Dune 3D v1.1.0 via Flatpak `flatpak run org.dune3d.dune3d` on ChromeOS the green rectangles for points on features (ie. line ends, center of arcs) do not render.

This makes it impossible to complete the [Constrain section](https://docs.dune3d.org/en/latest/simple-part.html#constrain) of the "Designing a Simple Part" tutorial as the center of the arc cannot be selected.

## Observed behavior
[image]

## Desired behavior
Green feature points are rendered and selectable.
[image]

## Environment observed
Dune 3D 1.1.0 is running via Flatpak:
```
$ flatpak info org.dune3d.dune3d

Lukas K. - Parametric 3D CAD

          ID: org.dune3d.dune3d
         Ref: app/org.dune3d.dune3d/x86_64/stable
        Arch: x86_64
      Branch: stable
     Version: 1.1.0
     License: GPL-3.0+
      Origin: flathub
  Collection: org.flathub.Stable
Installation: user
   Installed: 95.5 MB
     Runtime: org.gnome.Platform/x86_64/46
         Sdk: org.gnome.Sdk/x86_64/46

      Commit: 9894f7da277199d29fc43e690c84ceb0fa7d2251a4a9634080717b1871817272
      Parent: 2cf1665c9df05beb8ca5ec89eced8fb61c4ca8508b86166d65504d2fb890db9e
     Subject: version 1.1.0 (e1e82edf)
        Date: 2024-05-01 17:55:37 +0000
```

### Host OS
Google Pixelbook (Eve)
Intel Kabylake i5-7Y57
ChromeOS version 24.0.6367.95

### Additional information
#### Container (Crostini)
ChromeOS permits Linux apps in a container in a VM, using a [nested Wayland compositor](https://chromium.googlesource.com/chromiumos/platform2/+/HEAD/vm_tools/sommelier/README.md) and the [VirGL driver](https://docs.mesa3d.org/drivers/virgl.html) to provide a virtual GPU to permit Linux apps to leverage GPU acceleration from the host.
The container is running a Debian 12 (Bookworm) build on kernel 6.6.

```
$ uname -r
6.6.21-01327-gc9107f2d4947
```

##### glx-info

```
$ glxinfo -B
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Mesa/X.org (0x1af4)
    Device: virgl (Mesa Intel(R) HD Graphics 615 (KBL GT2)) (0x1010)
    Version: 22.3.6
    Accelerated: yes
    Video memory: 0MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.3
    Max compat profile version: 4.3
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL vendor string: Mesa/X.org
OpenGL renderer string: virgl (Mesa Intel(R) HD Graphics 615 (KBL GT2))
OpenGL core profile version string: 4.3 (Core Profile) Mesa 22.3.6
OpenGL core profile shading language version string: 4.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.3 (Compatibility Profile) Mesa 22.3.6
OpenGL shading language version string: 4.30
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile

OpenGL ES profile version string: OpenGL ES 3.2 Mesa 22.3.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

< GLX visual tables omitted from output>
```

carrotIndustries, 2024-05-07:
Oh, I'm surprised that it even works at all on that setup.

The green boxes are actually rendered as points with a suitably large point size. Looking really close at your screenshot you can in fact make out a faint small green dot at the center of the arc.

Apparently your GL implementation doesn't support point sizes larger than one.

What are the point size ranges listed in `eglinfo  -l`?

A workaround that I may implement is rendering points as two triangles instead of an actual point.

aboyett, 2024-05-18:
Looks like you identified the problem
>  GL_ALIASED_POINT_SIZE_RANGE = 1, 255
    GL_SMOOTH_POINT_SIZE_RANGE = 1, 1

Full `eglinfo -l` info in [eglinfo-l.log](https://github.com/dune3d/dune3d/files/15366891/eglinfo-l.log)

commit f923ce0d (2024-08-09): use icons for points

changelog: Enhancemts/Editor: use different icons for points

fixes #96
closed 2024-08-09 by carrotIndustries