dune3d/dune3d #97: Almost all UI text is missing on Windows

RagingRoosevelt, 2024-05-08:
OS: Windows 11 build 22631
Dune3d Release: 1.1.0 Beryllium

Steps to reproduce: 
* Download MSI from [release page](https://github.com/dune3d/dune3d/releases/tag/v1.1.0)
* Install using the custom option (only change from defaults was skipping desktop icon)
* Launch

I tried toggling dark theme in preferences to see if that would trigger a redraw (it did, with no resolution to the text issue).

Text shows up in the following places:
* all the tooltips I've seen
* inside the hamburger menu at the top right
* some parts of the settings window

[image]

[image]

[image]

Additional notes:

I noticed that, after changing a setting, the contents of `user_config.json` in `%LOCALAPPDATA%\dune3d` was simply `null`, which seemed odd.  I wouldn't rule out that I screwed something up, but I can't think of anything I did (literally installed and launched).

I tried removing and reinstalling using the "typical" option without any change.

carrotIndustries, 2024-05-08:
> I noticed that, after changing a setting, the contents of user_config.json

That's only used fro the recent files. The preferences are saved to `prefs.json`. 

You could try setting the environment variable `GTK_DEBUG=interactive` and change font to something else in Global→Settings.

Apart from that, the window's header bar also looks a bit off, so there's something else amiss too.

carrotIndustries, 2024-05-09:
Doesn't seem to be a general problem. Works on a friend's Windows 11 machine.

Is it also broken when building from source: https://docs.dune3d.org/en/latest/build-win32.html

RagingRoosevelt, 2024-05-09:
> You could try setting the environment variable `GTK_DEBUG=interactive` and change font to something else in Global→Settings.

> Is it also broken when building from source: https://docs.dune3d.org/en/latest/build-win32.html

I'll try to test both when I get home this evening.

carrotIndustries, 2024-05-09:
> I'll try to test both when I get home this evening.

When you've got MSYS2 installed, you can run `gtk4-demo` to see if it's also broken.

RagingRoosevelt, 2024-05-20:
OK, I was able to change the debug env var and change the font.  Looks like it worked to change the font but that didn't clear up the issue.
[image]

carrotIndustries, 2024-05-20:
The fonts are also broken in the inspector, so this is is Gtk bug.

RagingRoosevelt, 2024-05-24:
👍SG.  I'll see if there's anything open there already.

closed 2024-05-24 by RagingRoosevelt

RagingRoosevelt, 2024-09-15:
Tried setting `GSK_RENDERER=cairo` like you mentioned in #121 and that resolved the issue for me