dune3d/dune3d #100: Improve navigation system.

matiasdelellis, 2024-05-09:
I could never get used to FreeCAD, and with this application I achieved what I needed with a couple of hours of work. Thanks a lot!. 😄 

In principle, the only feature request I can ask of you is to improve the navigation system. It cost me a lot, but mainly because I am working with the notebook's touchpad. 🤦🏻‍♂️

About FreeCAD, the only thing missing is the navigation cube, which helps a lot when you work on the Cartesian axes, but in principle a small change, which would be very useful, would be to be able to navigate with the arrow keys. 🤔

matiasdelellis, 2024-05-12:
I'm familiarizing myself with the code to see what I can contribute.. I can't do much yet, but I found that the cube is part of opencascade.. 😬 

* https://dev.opencascade.org/doc/refman/html/class_a_i_s___view_cube.html
* https://draw.sview.ru/visualization-viewcube/

matiasdelellis, 2024-05-24:
Just commenting that I am using this and it is sufficient for my needs, but obviously I am doing something wrong.
I have no experience in developing and using 3D tools so I still don't understand the use of the camera and perspectives.  In particular seems that 90 degrees are being applied to a point that is moving and therefore rotating 4 times does not return exactly to the same camera view.

https://github.com/matiasdelellis/dune3d/commit/fe58180b094205fadd0cb624a3dd98b3b8010ab3

matiasdelellis, 2024-05-24:
Whel, It seems to be working fine, except that when sending two rotations quickly, when starting the second it cancels the first rotation and the second rotation is applied from where the first was canceled.

I don't know how to improve this, but would you accept a pull request in this state?

SebastianCardenasEscoto, 2024-09-08:
> Dune 3D isn't using opencascade's visualization features at all. We need to implement this in 2D using cairo.

I've been working on this. I can render the cube using cairo, but I've been having issues passing a mouse click into the Cairo environment. Gtk4 documentation is pretty sparse on this, so I was wondering if anyone had suggestions on how to do this?

carrotIndustries, 2024-09-09:
> I've been working on this. I can render the cube using cairo

Neat!

In Gtk4 it works like this: You create an `EventController`/`Gesture` of the right kind and hook it up to the widget. See https://github.com/dune3d/dune3d/blob/main/src/canvas/canvas.cpp#L229 for an example.

Figuring what the x/y coordinates correspond to in what you're drawing then is your business.

SebastianCardenasEscoto, 2024-09-10:
This looks a lot like what I currently have:
`AxesLollipop::AxesLollipop()
{
    create_layout();
    set_content_height(100);
    set_content_width(120);
    set_draw_func(sigc::mem_fun(*this, &AxesLollipop::render));

    auto click_control = Gtk::GestureClick::create();
    click_control->signal_pressed().connect([this,click_control](int n_press, double x, double y) {
        std::cout << "button" << std::endl;
    });
    click_control->set_button(0);
    add_controller(click_control);
}`
However, I fail to see a `button` print statement when clicking the drawing area. I used a different print statement and I found that the click signal is being detected within to the `Canvas` object in canvas.cpp even when clicking in the drawing area. Is the ideal behavior that both of these objects should be receiving the click signal, or do I need to prioritize `AxesLollipop` somehow?

carrotIndustries, 2024-09-10:
Good that you're bringing this up since that could have been a tough one:

In the Cambalanche UI definition of the main window, the box the axis lollipop gets packed into is set to `can-target = false` so that its transparent for pointer events. Since it wasn't interactive that was right thing so far.

If you don't want to touch cambalanche, you can call `lollipop_box->set_can_target(true);` in https://github.com/dune3d/dune3d/blob/main/src/dune3d_appwindow.cpp#L153

SebastianCardenasEscoto, 2025-01-03:
Happy New Year!

Sorry this has taken me a while to get back to. I think I currently have something that's close to being a PR. I do need help with one more thing: would you have a recommended way of modifying the canvas camera from within `axes_lollipop.cpp`?

carrotIndustries, 2025-01-03:
> would you have a recommended way of modifying the canvas camera from within axes_lollipop.cpp

Add a signal to `AxesLollipop`   (or use the `Changeable` base class) and wire it up to the canvas.