dune3d/dune3d #101: Shader Compilation Errors on Ubuntu 24.04 with Nvidia A4000

martinkielhorn-hexagon, 2024-05-13:
After successfully compiling the dune3d software following the provided guide, running the generated binary results in multiple shader compilation errors related to OpenGL version requirements and unsupported syntax.

I built the software on Ubuntu 24.04 with Nvidia A4000. 

I followed the guide:
```
git clone https://github.com/dune3d/dune3d
meson setup build
The Meson build system
Version: 1.3.2
Source dir: /home/kiel/src/dune3d
Build dir: /home/kiel/src/dune3d/build
Build type: native build
Project name: dune3d
Project version: 1.1.0
C compiler for the host machine: cc (gcc 13.2.0 "cc (Ubuntu 13.2.0-23ubuntu4) 13.2.0")
C linker for the host machine: cc ld.bfd 2.42
C++ compiler for the host machine: c++ (gcc 13.2.0 "c++ (Ubuntu 13.2.0-23ubuntu4) 13.2.0")
C++ linker for the host machine: c++ ld.bfd 2.42
Host machine cpu family: x86_64
Host machine cpu: x86_64
Found pkg-config: YES (/usr/bin/pkg-config) 1.8.1
Run-time dependency gtk4 found: YES 4.14.2
Run-time dependency gtkmm-4.0 found: YES 4.10.0
Run-time dependency epoxy found: YES 1.5.10
Run-time dependency eigen3 found: YES 3.4.0
Run-time dependency glm found: YES 0.9.9.8
Found CMake: /usr/bin/cmake (3.28.3)
Run-time dependency opencascade found: YES 7.6.3
Run-time dependency spnav found: YES 1.1
Program python3 found: YES (/usr/bin/python3)
Message: including git commit
Build-time dependency gio-2.0 found: YES 2.80.0
Program /usr/bin/glib-compile-resources found: YES (/usr/bin/glib-compile-resources)
Fetching value of define "__GLIBCXX__" : 20240412 
Message: using libstdc++
Library stdc++ found: YES
Library stdc++fs found: YES
Run-time dependency uuid found: YES 2.39.3
Library m found: YES
Dependency eigen3 found: YES 3.4.0 (cached)
Build targets in project: 8
NOTICE: Future-deprecated features used:
 * 0.56.0: {'gui_app arg in executable'}

Found ninja-1.11.1 at /usr/bin/ninja

meson compile -C build
```

The application fails to start, outputting multiple shader compilation errors, indicating issues such as unsupported shader versions

```
./dune3d
realize
Compile failure in vertex shader: 0(2) : error C0201: unsupported version 330
0(3) : error C7532: global type vec2 requires "#version 100" or later
0(4) : error C7532: global type vec3 requires "#version 100" or later
0(4) : error C5060: out can't be used with non-varying color_to_fragment
0(9) : error C7532: global variable gl_VertexID requires "#version 130" or later
0(9) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
0(10) : error C7532: global function pow requires "#version 100" or later
0(15) : error C7532: global variable gl_Position requires "#version 100" or later
0(15) : error C7532: global type vec4 requires "#version 100" or later
 
Compile failure in vertex shader: 0(2) : error C0201: unsupported version 330
0(3) : error C7532: global type vec3 requires "#version 100" or later
0(7) : error C5060: out can't be used with non-varying normal_to_fragment
0(8) : error C5060: out can't be used with non-varying color_to_fragment
0(9) : error C5060: out can't be used with non-varying pos_to_fragment
[... 106 lines cut ...]
0(8) : error C5060: out can't be used with non-varying origin_to_geom
0(9) : error C7532: global type vec2 requires "#version 100" or later
0(9) : error C5060: out can't be used with non-varying shift_to_geom
0(10) : error C7532: global type float requires "#version 100" or later
0(10) : error C5060: out can't be used with non-varying scale_to_geom
0(11) : error C7532: global type uint requires "#version 130" or later
0(11) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
0(11) : error C5060: out can't be used with non-varying flags_to_geom
0(12) : error C5060: out can't be used with non-varying pick_to_geom
0(13) : error C5060: out can't be used with non-varying bits_to_geom
0(16) : error C7532: global type mat4 requires "#version 100" or later
0(20) : error C7532: global variable gl_VertexID requires "#version 130" or later
0(20) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
0(20) : error C7532: global type int requires "#version 100" or later
0(21) : error C1503: undefined variable "flags"
0(22) : error C1503: undefined variable "origin"
0(23) : error C1503: undefined variable "shift"
0(24) : error C1503: undefined variable "bits"
0(25) : error C1503: undefined variable "scale"
 
gl error 1,281 in ../src/canvas/glyph_renderer.cpp:44
Aborted (core dumped)
```

I am not entirely sure what could be causing these shader compilation issues. My initial thought was that it could be related to my Linux setup or perhaps some incompatibility with my hardware, but I am unable to pinpoint the exact problem. I am willing to try any suggested fixes or workarounds to resolve this issue. Any guidance or suggestions from the community or developers would be greatly appreciated, as I am keen to get this software running smoothly on my system.

carrotIndustries, 2024-05-13:
Duplicate of #87

closed 2024-05-13 by carrotIndustries

martinkielhorn-hexagon, 2024-05-14:
Thanks that link helped.

The bug can be fixed by setting GDK_DEBUG to either gl-prefer-gl, gl-glx, or gl-disable-gles, but won't go away with gl-disable-gl, gl-egl.

Forcing OpenGL Core Profile: Setting the GDK_DEBUG environment variable [1] to gl-prefer-gl, gl-glx, or gl-disable-gles directs GTK, the toolkit dune3d uses for its graphical interface, to prioritize the full OpenGL Core Profile instead of the default ES profile. This enables the shaders to compile successfully as they have access to the required features.

[1] https://docs.gtk.org/gtk4/running.html#gdk_debug