dune3d/dune3d #106: +/- buttons for polar array element count broken

Imarok, 2024-05-23:
**Reproduce:**
1. Create a polar array
2. Change the count with the +/-

**Result:**
Number of elements in the 3D viewport does not change
When deselecting the polar array the number is reset to the previous value.

**Expected:**
The number of elements should immediately change in the 3D viewport and not be reset.

**Tested on:**
38fb19c
(older commit on main, but the newer commits don't sound like they fixed something in the area)

carrotIndustries, 2024-05-24:
Yeah... you need to press enter to commit the change. This was done so that spinning the mouse wheel over the entry doesn't spam the undo/redo history.

The right thing to do is to apply the change to the document right away and to some sort of debouncing for creating the undo/redo items. That's how its done in Horizon EDA.

carrotIndustries, 2024-06-07:
fixed in above commit

closed 2024-06-07 by carrotIndustries