dune3d/dune3d #107: Feature Suggestion: Clickable axis to look at base planes

Imarok, 2024-05-23:
To quickly change the view to look on the one of the base planes (XY, XZ, YZ) it would be nice to make the blobs of the 3D viewport orientation indicator (the RGB thingy in the top right corner) clickable.
I.e.:
- click on X -> look at YZ plane
- click on Y -> look at XZ plane
- click on Z -> look at XY plane

carrotIndustries, 2024-05-24:
Might make sense to combine this with replacing the axis lollipop with a cube as suggested in https://github.com/dune3d/dune3d/issues/100

Imarok, 2024-05-24:
> Might make sense to combine this with replacing the axis lollipop with a cube as suggested in #100

Sounds like a good idea If that is planned. (Though I like the current look of the lollipop ;)

andersmmg, 2025-11-24:
I've been working a bit on a cube to replace the lollipops which will be clickable, here's how it currently looks:
[image]
Each of the faces, edges, and corners are clickable but currently just print to stdout, they would move to that perspective eventually. I'd like to still have the lollipops somehow, maybe as an option or smaller in another corner or something, @carrotIndustries what are your thoughts

carrotIndustries, 2025-11-24:
Looking great!

Freecad has the axes integrated with the cube: 

[image]

Plus another lollipop-style thing in the bottom-right corner.
If we don't want to integrate them, we could have smaller axes lollipop in the bottom-right corner as well.