dune3d/dune3d #117: Feature request: named parameters

Skrapion, 2024-05-25:
Dune 3D does not support named parameters, which makes it challenging to work with parametric models.

Ideally there would be a property window to define parameters with names. This could be similar to the "Manage Buses" window in Horizon EDA. Then dimension and angle constraints could use these names instead of direct measurements.

carrotIndustries, 2024-06-07:
That goes hand in hand with #83

dgramop, 2024-09-06:
I occasionally help out on this feature for solvespace's "parameters" branch. Since Dune3D works with the same solver, I wonder if we can/should use some of the code or lessons learned from solvespace and apply them here.

carrotIndustries, 2024-09-25:
Any major takeaways from the discussions over at solvespace that'd apply to this proposal? https://github.com/dune3d/dune3d/discussions/83#discussioncomment-8751497

dgramop, 2024-10-01:
There are definitely some, but I there's still some details to hammer out about scope rules. I would wait to see a concrete implementation of this in solvespace before taking those solution seriously, so we can learn from those lessons.

I'll keep this issue in mind, and if I happen to be the one who contributes that code to solvespace I'll try to summarize my changes/thinking/suggestions here (or maybe it'll be in the PR writeup)