dune3d/dune3d #121: [Win] GUI elements missing on v1.1.0

DavorSikic, 2024-06-14:
Only few boxes are displayed on fresh typical install of "dune3d-1.1.0-x64.msi" (May 1).

[image]

Everything works well on "dune3d-1.0.0-x64.msi" (Feb 1).

[image]

Windows 10 (64-bit)
Intel HD Graphics for 4th Generation Intel Processors - Pentium G3250
Driver version: 20.19.15.4835 - latest (10/16/2017)

carrotIndustries, 2024-06-16:
The 1.0 release looks broken too, the "open recent" box should have an opaque background.

Must be some weird Gtk / GPU driver interaction.

carrotIndustries, 2024-08-06:
I encountered that problem as well, setting the environment variable `GSK_RENDERER=cairo` fixes it.

RagingRoosevelt, 2024-09-15:
looks a lot like #97