dune3d/dune3d #122: Feature request: transparency for imported STEP models

jue89, 2024-06-17:
For complex assemblies it happens that imported STEP models hide parts of the designed body. It would be nice to specify transparency for STEP models or at least hide them completely.

carrotIndustries, 2024-06-17:
Encountered that as well. Some thoughts on the implementation: Transparency is fairly hard to implement since currently elements aren't rendered in any particular order. What's possible to implement is hiding the step model entirely or only showing a wireframe.

We could implement this at various levels of granularity and persistence:
 - As a property of the imported STEP model, will be saved in the document
 - As a property of the current group, to be saved in the view state once we have that
 - As a property of the current body, to be saved in the view state once we have that
 - As a property of the current document, to be saved in the view state once we have that

jue89, 2024-06-18:
> Encountered that as well. Some thoughts on the implementation: Transparency is fairly hard to implement since currently elements aren't rendered in any particular order. What's possible to implement is hiding the step model entirely or only showing a wireframe.

That would help a lot - both. But I'd say its important to keep the selected anchors visible.

> We could implement this at various levels of granularity and persistence:
>
> * As a property of the imported STEP model, will be saved in the document
> * As a property of the current group, to be saved [...]

In this particular case, I have two step models in one group. I'd love to hide them individually. If the this is saved in the document, I'm completely fine with that.

carrotIndustries, 2024-06-27:
When selecting a STEP model, there's now an option to hide the solid part of it. Anchors remain visible.

[image]

So far, it's not saved as all, but will be saved in the view state soon.

closed 2024-06-27 by carrotIndustries

jue89, 2024-07-04:
Thank you a lot! Very nice :)

carrotIndustries, 2024-07-04:
> So far, it's not saved as all, but will be saved in the view state soon.

Done in eeeca50546caba676ddaaf2dd5c74acc29bad16c.

commit 4fca12b1 (2024-06-27): add EntityView

just for STEP entities for now

changelog: Enhancements/Editor: add option for hiding STEP solid model

closes #122