dune3d/dune3d #129: Dependency "OpenCASCADE" not found

oleka2, 2024-09-03:
I've just tried to compile Dune 3D, on my Slackware Linux 64, but my opencascade was not recognized properly.

```
Run-time dependency opencascade found: NO (tried cmake)
meson.build:19:14: ERROR: Dependency "OpenCASCADE" not found, tried cmake
```

I had the same problem with Horizon-eda a couple of years ago. 
Appending -lTKG3d to the LDFLAGS_OCE variable in the Makefile helped in this matter.

carrotIndustries, 2024-09-03:
Never used slackware before Ran the `ghcr.io/aclemons/slackware:15.0` container to try and get a working slackware setup. Installed git and tried to clone the repo and got stuck at `fatal: unable to access 'https://github.com/dune3d/dune3d/': SSL certificate problem: unable to get local issuer certificate
`

Can you give detailed instruction how to get from that empty container image to where you got stuck at building dune3d?

Depending on where opencascade is installed, (doesn't seem to be in the repos), you may need to point meson to it as such: `meson setup ../build --cmake-prefix-path=/usr/local/lib`

oleka2, 2024-09-04:
I've never used things like the ghcr.io/aclemons/slackware:15.0 containers.
Maybe the opencascade (zipped) that I compiled myself will help ?
Unpacking it (or only viewing its structure in an archiver) should show where it goes after installation in the system.
I removed some of the biggest files in this zip, because git allows only 25MB as an attachment.
[opencascade-tree-directory.zip](https://github.com/user-attachments/files/16863293/opencascade-tree-directory.zip)

carrotIndustries, 2024-09-04:
>  should show where it goes after installation in the system.

Based on that I'd try  `--cmake-prefix-path=/usr/lib64`

oleka2, 2024-09-04:
Not luck as of now, see the screen...
[image]

carrotIndustries, 2024-09-04:
Then I'm out of ideas unless I can reproduce it locally.

If you can't build it, you could use the flatpak instead.

closed 2024-09-04 by carrotIndustries

oleka2, 2024-09-17:
This code :
```
cd /usr/lib64/cmake/opencascade
    sed -i -e 's/\\\${OCCT_INSTALL_BIN_LETTER}//' *.cmake
```
resolved the problem, Dune3D compiles well now.

Did you try the VariCAD software ? It maybe interesting.