dune3d/dune3d #135: Feature request: Expose length ratio constraints

viernullvier, 2024-10-21:
I noticed that some familiar constraint types from Solvespace are still missing:
- Line-to-line length ratio
- Line-to-arc length ratio
- Arc-to-arc length ratio

The implementation is already there in the Solvespace folder, it just needs to be integrated into Dune3D.

carrotIndustries, 2024-10-21:
Yeah, never had any use for them. They're probably a good candidate for someone else to implement.

viernullvier, 2024-10-22:
Fair enough. I realized that I _am_ someone else, so I took a first look at the effort required - a lot of things can be taken from the length constraint implementation, but there are still some details to watch out for (picking the right constrain implementation depending on selection, suppressing the constraint if it's redundant, the related measurement functionality…) in order to make it work as nicely as the other constraints.

That said, I managed to find a fully geometric solution to the problem I was facing without having to specify a ratio at all. Considering that I am not really familiar with C++, I think I have to pass this time.