dune3d/dune3d #136: Export STL file after update 1.1 -> 1.2 failed

adrianhoehne, 2024-10-24:
A drawing was created and exported to STL with Dune3d Version 1.1. After update to 1.2 and opening the working file, Dune3d claims that this file was created by an older version and needs to be saved. Click save, does not remove the message but gives also no info about failure or success. Assuming that the file was saved anyway, I tried to export to STL again. The dialog for saving opens, and I choose the old STL file, which then fills the name automatically into the necessary field. Clicking on save, asks me if that file shall be overwritten and I choose YES. Then Dune3d crashes. 

I also tried with Save as to save the working file, that seems to save the d3dddoc file successfully since the message on the top disappear, but the export to STL still doesn't work.

I tried to export STL with a new file name, that doesn't work either.

System:
Pop!_OS 22.04 with Wayland
Gnome: 42.9
Kernel: 6.9.3-76060903-generic
[Dune3dFehlerLog.txt](https://github.com/user-attachments/files/17505795/Dune3dFehlerLog.txt)

adrianhoehne, 2024-10-24:
After saving the working file and restarting the application, the export was successful again. And it was not reproducible anymore. Also, not with other not migrated files. 
It seems that this is a one time issue that maybe appear only on first start after update.

closed 2024-10-24 by adrianhoehne

carrotIndustries, 2024-10-24:
> was created by an older version and needs to be saved.

That's not quite what it says. You can view the file as much as you like but when you save it the next time it will be upgraded to the latest file format.

> Click save, does not remove the message but gives also no info about failure or success.

From version 1.2 onwards the save button is always enabled to save the workspace views. However the document is only saved if it got modified. That's why the info bar stays there.

Do you still have the original file? Might try reproducing with that one.

So far I don't have a clue why this happend and would like to reproduce it.

adrianhoehne, 2024-10-28:
> > was created by an older version and needs to be saved.
> 
> That's not quite what it says. You can view the file as much as you like but when you save it the next time it will be upgraded to the latest file format.
> 
Yes, you’re right. That was how I understood it.

> > Click save, does not remove the message but gives also no info about failure or success.
> 
> From version 1.2 onwards the save button is always enabled to save the workspace views. However the document is only saved if it got modified. That's why the info bar stays there.

Makes sense. Thank you.
> 
> Do you still have the original file? Might try reproducing with that one.
> 
> So far I don't have a clue why this happend and would like to reproduce it.

I zipped them because I was not able to add them here. I made small changes after it was saved, but maybe they help anyway.

[BatteriehalterD3d.zip](https://github.com/user-attachments/files/17544043/BatteriehalterD3d.zip)