dune3d/dune3d #138: core dump (reproducible)

splace, 2024-10-24:
click on selection up arrow, when there are no selections.

```
Process 93957 (dune3d) of user 1000 dumped core.

Stack trace of thread 2:
#0  0x00007fda157aff68 n/a (n/a + 0x0)
#1  0x00007fda17766eba n/a (n/a + 0x0)
#2  0x00007fda17741f3f n/a (n/a + 0x0)
#3  0x00007fda1774c385 n/a (n/a + 0x0)
#4  0x00007fda176e73aa n/a (n/a + 0x0)
#5  0x00007fda176e84af n/a (n/a + 0x0)
#6  0x00007fda178638e8 n/a (n/a + 0x0)
#7  0x000055d2a8e8cea0 n/a (n/a + 0x0)
#8  0x000055d2a8e8b342 n/a (n/a + 0x0)
#9  0x000055d2a8eadc7f n/a (n/a + 0x0)
#10 0x000055d2a8e36a08 n/a (n/a + 0x0)
#11 0x000055d2a8f0577c n/a (n/a + 0x0)
#12 0x000055d2a8db8bf6 n/a (n/a + 0x0)
#13 0x000055d2a8f8b86f n/a (n/a + 0x0)
#14 0x000055d2a8f793cc n/a (n/a + 0x0)
#15 0x00007fda1a226754 n/a (n/a + 0x0)
#16 0x00007fda1b30991a n/a (n/a + 0x0)
#17 0x00007fda1b31e91d n/a (n/a + 0x0)
#18 0x00007fda1b320071 n/a (n/a + 0x0)
#19 0x00007fda1b325e01 n/a (n/a + 0x0)
#20 0x00007fda1b325ec3 n/a (n/a + 0x0)
#21 0x00007fda1a8e69fc n/a (n/a + 0x0)
#22 0x00007fda1b309b32 n/a (n/a + 0x0)
#23 0x00007fda1b3201c0 n/a (n/a + 0x0)
#24 0x00007fda1b325e01 n/a (n/a + 0x0)
#25 0x00007fda1b325ec3 n/a (n/a + 0x0)
#26 0x00007fda1a9a70a3 n/a (n/a + 0x0)
#27 0x00007fda1b30cde5 n/a (n/a + 0x0)
#28 0x00007fda1b309b32 n/a (n/a + 0x0)
#29 0x00007fda1b3201c0 n/a (n/a + 0x0)
#30 0x00007fda1b325e01 n/a (n/a + 0x0)
#31 0x00007fda1b325ec3 n/a (n/a + 0x0)
#32 0x00007fda1a9a3db0 n/a (n/a + 0x0)
#33 0x00007fda1a9a566b n/a (n/a + 0x0)
#34 0x00007fda1a9a88a5 n/a (n/a + 0x0)
#35 0x00007fda1a971c1c n/a (n/a + 0x0)
#36 0x00007fda1aadaa96 n/a (n/a + 0x0)
#37 0x00007fda1aae2a18 n/a (n/a + 0x0)
#38 0x00007fda1a9eae6a n/a (n/a + 0x0)
#39 0x00007fda1a9eb4a3 n/a (n/a + 0x0)
#40 0x00007fda1ac82f5e n/a (n/a + 0x0)
#41 0x00007fda1ad1e6fa n/a (n/a + 0x0)
#42 0x00007fda1b309b32 n/a (n/a + 0x0)
#43 0x00007fda1b31f3e3 n/a (n/a + 0x0)
#44 0x00007fda1b325e01 n/a (n/a + 0x0)
#45 0x00007fda1b325ec3 n/a (n/a + 0x0)
#46 0x00007fda1ad22745 n/a (n/a + 0x0)
#47 0x00007fda1ac9a9a6 n/a (n/a + 0x0)
#48 0x00007fda1a2b4077 n/a (n/a + 0x0)
#49 0x00007fda1a2b6387 n/a (n/a + 0x0)
#50 0x00007fda1a2b6b73 n/a (n/a + 0x0)
#51 0x00007fda1a4a13c5 n/a (n/a + 0x0)
#52 0x000055d2a8d5af6d n/a (n/a + 0x0)
#53 0x00007fda1602b188 n/a (n/a + 0x0)
#54 0x00007fda1602b24b n/a (n/a + 0x0)
#55 0x000055d2a8d650c5 n/a (n/a + 0x0)
ELF object binary architecture: AMD x86-64

```

carrotIndustries, 2024-10-24:
> click on selection up arrow

Which button are you referring to?

Also, a backtrace without functions names isn't all that useful. Either get the debug symbols or build from source to get a useful backtrace.

splace, 2024-10-28:
[image]

splace, 2024-10-28:
>  Also, a backtrace without functions names isn't all that useful. Either get the debug symbols or build from source to get a useful backtrace.

thats what i got, take it or leave it, up to you.

carrotIndustries, 2024-10-28:
Doesn't reproduce for me. Please provide detailed reproduction instructions starting from a new document or an attached one.

splace, 2024-10-28:
[Screencast From 2024-10-28 22-36-31.webm](https://github.com/user-attachments/assets/9790e27e-b4dc-4bde-abeb-be4a3798ee52)

carrotIndustries, 2024-10-28:
Ok, I assume that's the document from the dune3d-samples repo the last group is a chamfer group, so moving it one up could cause some issues in opencascade. However, it doesn't crash for me, using latest main and opencascade 7.8.1.

Without a usable backtrace from your system, there's not much that can be done to debug this any further.

I assume you're using version 1.2. What's your version of opencascade?

splace, 2024-10-28:
this is latest in flatpak.

> What's your version of opencascade?

means nothing to me.

carrotIndustries, 2024-10-28:
> this is latest in flatpak.

Thanks, that was the missing piece. There, it reproduces for me as well.

carrotIndustries, 2024-10-30:
This has been fixed via https://github.com/flathub/org.dune3d.dune3d/commit/7b5db6b4c1c11fb76ed94f348099b744c4c11476 an updated version (with the same number) is available on flathub.

closed 2024-10-30 by carrotIndustries