dune3d/dune3d #143: Cannot run on Linux with nvidia driver wayland and flatpak

yodatak, 2024-12-05:
Hello here my log i got i cannot run it for now because of this error how can i debug it ?
 flatpak run org.dune3d.dune3d
realize
dune3d: ../src/dispatch_common.c :872 : epoxy_get_proc_address:  l'assertion « 0 && "Couldn't find current GLX or EGL context.\n" » a échoué.

nvidia-smi
Fri Dec  6 00:20:31 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 565.57.01              Driver Version: 565.57.01      CUDA Version: 12.7     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3060        Off |   00000000:09:00.0  On |                  N/A |
| 37%   34C    P5             30W /  170W |    1020MiB /  12288MiB |      0%      Default |
|                                         |                        |                  N/A |

# System Details Report
---

## Report details
- **Date generated:**                              2024-12-06 00:23:13

## Hardware Information:
- **Hardware Model:**                              ASUSTeK COMPUTER INC. PRIME B450M-K
- **Memory:**                                      48.0 GiB
- **Processor:**                                   AMD Ryzen™ 5 2600 × 12
- **Graphics:**                                    NVIDIA GeForce RTX™ 3060
- **Disk Capacity:**                               (null)

## Software Information:
- **Firmware Version:**                            2409
- **OS Name:**                                     Bazzite 41 (FROM Fedora Silverblue)
- **OS Build:**                                    Stable (F41.20241125)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               47
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.11.9-303.bazzite.fc41.x86_64

carrotIndustries, 2024-12-05:
Do the OpenGL demos from `gtk4-demo` work on your machine?

yodatak, 2024-12-06:
Yes i run the gears and open gl area but not shadertoy

not sure if its linked 

[image]

yodatak, 2024-12-06:
> Yes i run the gears and open gl area but not shadertoy
>
> not sure if its linked [...]

https://gitlab.gnome.org/GNOME/gtk/-/issues/6772

carrotIndustries, 2024-12-06:
That looks related. Try the environment variables suggested in the Gtk issue, such as `GDK_DEBUG=gl-prefer-gl`. Also see #87

tallbiscuits, 2024-12-28:
> That looks related. Try the environment variables suggested in the Gtk issue, such as `GDK_DEBUG=gl-prefer-gl`. Also see #87

I have the same issue and can confirm running your suggested environment variable like `flatpak run --env=GDK_DEBUG=gl-prefer-gl org.dune3d.dune3d` works and make Dune 3D run.

carrotIndustries, 2025-04-11:
More likely than not duplicate of #87

closed 2025-04-11 by carrotIndustries