dune3d/dune3d #154: Crash when linking document

ii8, 2025-02-18:
To reproduce:
1. Open [left.txt](https://github.com/user-attachments/files/18844238/left.txt) (Changed extension to .txt so github allows it, just change back to .d3ddoc before opening)
2. Use "Link document" tool and choose to link: [cfg.txt](https://github.com/user-attachments/files/18844333/cfg.txt) (same story)

<details>

<summary>Backtrace</summary>

```
Thread 1 "dune3d" hit Catchpoint 1 (exception thrown), 0x00007ffff3af4d20 in __cxa_throw () from /usr/lib64/libstdc++.so.6
#0  0x00007ffff3af4d20 in __cxa_throw () from /usr/lib64/libstdc++.so.6
#1  0x00007ffff3ac38da in std::__throw_out_of_range(char const*) () from /usr/lib64/libstdc++.so.6
#2  0x0000555555634070 in std::map<dune3d::UUID, std::unique_ptr<dune3d::Group, std::default_delete<dune3d::Group> >, std::less<dune3d::UUID>, std::allocator<std::pair<dune3d::UUID const, std::unique_ptr<dune3d::Group, std::default_delete<dune3d::Group> > > > >::at (this=0x55555907f698, __k=...) at /usr/include/c++/13.2/bits/stl_map.h:553
#3  0x000055555563393d in dune3d::Document::get_group<dune3d::Group> (this=0x55555907f630, uu=...) at ../src/document/document.hpp:143
#4  0x00005555559e4b03 in dune3d::WorkspaceBrowser::select_group (this=0x555556e2af80, uu=...) at ../src/editor/workspace_browser.cpp:1003
#5  0x00005555559dc4c4 in dune3d::WorkspaceBrowser::update_current_group (this=0x555556e2af80, doc_views=std::map with 3 elements = {...}) at ../src/editor/workspace_browser.cpp:311
#6  0x00005555559db2d8 in dune3d::WorkspaceBrowser::update_documents (this=0x555556e2af80, doc_views=std::map with 3 elements = {...}) at ../src/editor/workspace_browser.cpp:189
#7  0x000055555583a0ff in operator() (__closure=0x555556d58960) at ../src/editor/editor.cpp:91
#8  0x000055555585dd4b in std::__invoke_impl<void, dune3d::Editor::init()::<lambda()>&>(std::__invoke_other, struct {...} &) (__f=...) at /usr/include/c++/13.2/bits/invoke.h:61
#9  0x000055555585addf in std::__invoke<dune3d::Editor::init()::<lambda()>&>(struct {...} &) (__fn=...) at /usr/include/c++/13.2/bits/invoke.h:96
#10 0x00005555558568c8 in std::invoke<dune3d::Editor::init()::<lambda()>&>(struct {...} &) (__fn=...) at /usr/include/c++/13.2/functional:113
#11 0x0000555555850674 in sigc::adaptor_functor<dune3d::Editor::init()::<lambda()> >::operator()<>(void) const (this=0x555556d58960) at /usr/include/sigc++-3.0/sigc++/adaptors/adaptor_trait.h:100
#12 0x00005555558506a3 in sigc::internal::slot_call<dune3d::Editor::init()::<lambda()>, void>::call_it(sigc::internal::slot_rep *) (rep=0x555557373520) at /usr/include/sigc++-3.0/sigc++/functors/slot.h:154
#13 0x00005555557d2f0d in sigc::internal::signal_emit<void, void>::emit (impl=std::shared_ptr<sigc::internal::signal_impl> (use count 2, weak count 3) = {...}) at /usr/include/sigc++-3.0/sigc++/signal.h:366
#14 0x00005555557d2fc6 in sigc::signal_with_accumulator<void, void>::emit (this=0x555556df80b8) at /usr/include/sigc++-3.0/sigc++/signal.h:454
#15 0x0000555555949957 in dune3d::Core::add_document (this=0x555556df8060, path=filesystem::path "/home/murray/c/pp-cad/cfg.d3ddoc" = {...}, uu=...) at ../src/core/core.cpp:59
#16 0x0000555555848b63 in dune3d::Editor::open_file (this=0x555556df7d78, path=filesystem::path "/home/murray/c/pp-cad/cfg.d3ddoc" = {...}) at ../src/editor/editor.cpp:1458
#17 0x00005555558493e2 in dune3d::Editor::load_linked_documents (this=0x555556df7d78, uu_doc=...) at ../src/editor/editor.cpp:1499
#18 0x00005555558bfba6 in operator() (__closure=0x555556f752c0) at ../src/editor/editor_action.cpp:272
#19 0x00005555558cfcef in std::__invoke_impl<void, dune3d::Editor::init_actions()::<lambda()>&>(std::__invoke_other, struct {...} &) (__f=...) at /usr/include/c++/13.2/bits/invoke.h:61
#20 0x00005555558cf7b7 in std::__invoke<dune3d::Editor::init_actions()::<lambda()>&>(struct {...} &) (__fn=...) at /usr/include/c++/13.2/bits/invoke.h:96
#21 0x00005555558ce650 in std::invoke<dune3d::Editor::init_actions()::<lambda()>&>(struct {...} &) (__fn=...) at /usr/include/c++/13.2/functional:113
#22 0x00005555558cc9ba in sigc::adaptor_functor<dune3d::Editor::init_actions()::<lambda()> >::operator()<>(void) const (this=0x555556f752c0) at /usr/include/sigc++-3.0/sigc++/adaptors/adaptor_trait.h:100
#23 0x00005555558cc9e9 in sigc::internal::slot_call<dune3d::Editor::init_actions()::<lambda()>, void>::call_it(sigc::internal::slot_rep *) (rep=0x555556f75280) at /usr/include/sigc++-3.0/sigc++/functors/slot.h:154
#24 0x00005555557d2f0d in sigc::internal::signal_emit<void, void>::emit (impl=std::shared_ptr<sigc::internal::signal_impl> (use count 2, weak count 7) = {...}) at /usr/include/sigc++-3.0/sigc++/signal.h:366
#25 0x00005555557d2fc6 in sigc::signal_with_accumulator<void, void>::emit (this=0x555556df80d8) at /usr/include/sigc++-3.0/sigc++/signal.h:454
#26 0x000055555594c843 in dune3d::Core::rebuild_finish (this=0x555556df8060, from_undo=false, comment="tool") at ../src/core/core.cpp:620
#27 0x000055555594c773 in dune3d::Core::rebuild_internal (this=0x555556df8060, from_undo=false, comment="tool") at ../src/core/core.cpp:606
#28 0x000055555594bfe8 in dune3d::Core::maybe_end_tool (this=0x555556df8060, r=...) at ../src/core/core.cpp:538
#29 0x000055555594c289 in dune3d::Core::tool_update (this=0x555556df8060, args=...) at ../src/core/core.cpp:583
#30 0x00005555558e735d in dune3d::Editor::tool_update_data (this=0x555556df7d78, data=std::unique_ptr<dune3d::ToolData> = {...}) at ../src/editor/editor_tool.cpp:58
#31 0x00005555559b2f39 in operator() (__closure=0x555559b81bc0, result=std::shared_ptr<Gio::AsyncResult> (use count 1, weak count 0) = {...}) at ../src/core/tools/tool_link_document.cpp:39
#32 0x00005555559b3f5c in std::__invoke_impl<void, dune3d::ToolLinkDocument::begin(const dune3d::ToolArgs&)::<lambda(Glib::RefPtr<Gio::AsyncResult>&)>&, std::shared_ptr<Gio::AsyncResult>&>(std::__invoke_other, struct {...} &) (__f=...) at /usr/include/c++/13.2/bits/invoke.h:61
#33 0x00005555559b3e8e in std::__invoke<dune3d::ToolLinkDocument::begin(const dune3d::ToolArgs&)::<lambda(Glib::RefPtr<Gio::AsyncResult>&)>&, std::shared_ptr<Gio::AsyncResult>&>(struct {...} &) (__fn=...) at /usr/include/c++/13.2/bits/invoke.h:96
#34 0x00005555559b3d93 in std::invoke<dune3d::ToolLinkDocument::begin(const dune3d::ToolArgs&)::<lambda(Glib::RefPtr<Gio::AsyncResult>&)>&, std::shared_ptr<Gio::AsyncResult>&>(struct {...} &) (__fn=...) at /usr/include/c++/13.2/functional:113
#35 0x00005555559b3b7e in sigc::adaptor_functor<dune3d::ToolLinkDocument::begin(const dune3d::ToolArgs&)::<lambda(Glib::RefPtr<Gio::AsyncResult>&)> >::operator()<std::shared_ptr<Gio::AsyncResult>&>(void) const (this=0x555559b81bc0) at /usr/include/sigc++-3.0/sigc++/adaptors/adaptor_trait.h:100
#36 0x00005555559b3bc4 in sigc::internal::slot_call<dune3d::ToolLinkDocument::begin(const dune3d::ToolArgs&)::<lambda(Glib::RefPtr<Gio::AsyncResult>&)>, void, std::shared_ptr<Gio::AsyncResult>&>::call_it(sigc::internal::slot_rep *, sigc::type_trait_take_t) (rep=0x555556fab0a0, a_#0=...) at /usr/include/sigc++-3.0/sigc++/functors/slot.h:154
#37 0x00007ffff6b8ef85 in gdkmm_SignalProxy_async_callback () from /usr/lib64/libgtkmm-4.0.so.0
#38 0x00007ffff709a493 in g_task_return_now (task=task@entry=0x555559bb04a0) at ../gio/gtask.c:1361
#39 0x00007ffff709b2a3 in g_task_return (type=<optimized out>, task=0x555559bb04a0) at ../gio/gtask.c:1430
#40 g_task_return (task=0x555559bb04a0, type=<optimized out>) at ../gio/gtask.c:1387
#41 0x00007ffff754b502 in response_cb (task=0x555559bb04a0, response=-3) at ../gtk/gtkfiledialog.c:773
#42 0x00007ffff7e96a02 in g_cclosure_marshal_VOID__BOOLEANv (closure=<optimized out>, return_value=<optimized out>, instance=<optimized out>, args=<optimized out>, marshal_data=<optimized out>, n_params=<optimized out>, param_types=0x5555598d4500) at ../gobject/gmarshal.c:272
#43 0x00007ffff7e94761 in _g_closure_invoke_va (closure=0x555559bb0600, return_value=0x0, instance=0x5555588493c0, args=0x7fffffffc4f0, n_params=1, param_types=0x5555598d4500) at ../gobject/gclosure.c:896
#44 0x00007ffff7eaab5f in signal_emit_valist_unlocked (instance=instance@entry=0x5555588493c0, signal_id=signal_id@entry=281, detail=detail@entry=0, var_args=var_args@entry=0x7fffffffc4f0) at ../gobject/gsignal.c:3423
#45 0x00007ffff7eb0932 in g_signal_emit_valist (instance=0x5555588493c0, signal_id=281, detail=0, var_args=0x7fffffffc4f0) at ../gobject/gsignal.c:3262
[... cut ...]
#88 g_main_context_iterate_unlocked (context=context@entry=0x555556b60f40, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../glib/gmain.c:4273
#89 0x00007ffff6ef84dc in g_main_context_iteration (context=context@entry=0x555556b60f40, may_block=may_block@entry=1) at ../glib/gmain.c:4338
#90 0x00007ffff70c97ed in g_application_run (application=0x555556a6f210, argc=1, argv=<optimized out>) at ../gio/gapplication.c:2715
#91 0x000055555580935c in main (argc=1, argv=0x7fffffffdfe8) at ../src/main.cpp:13
```

</details>

void-linux, musl, dune3d @ 90513bb8de89c20eda9260a434141415a792b703

carrotIndustries, 2025-02-18:
Slighly offtopic: Since you're using void linux, how about packaging dune3d for it?

closed 2025-02-18 by carrotIndustries

ii8, 2025-02-18:
Already on it: https://github.com/void-linux/void-packages/pull/54357

commit 09b14b12 (2025-02-18): don't throw exception when opening more than one document

got broken in 95a2cd7e50d082d49d24d017ea4fa86776cf0469

fixes #154