dune3d/dune3d #156: Hang and crash when importing large STEP file.

ii8, 2025-02-28:
When I import a moderately large [step file](https://www.meanwell.com/upload/pdf/RSP-200/RSP-200-3D.zip), 5.5M in this case, Dune 3D will first freeze for about 10 seconds and then quit and show the message: `Gdk-Message: 22:55:42.997: Lost connection to Wayland compositor.`

I can cause the same behaviour in other ways, for example: sketch rectangle, extrude to make a box, make polar array and increase the count to a somewhat high number like 20, try to move it, hang then crash. On faster machines presumably it takes a lot more than 20, but mine is aged and slow.

It seems anything that takes a long time to compute will block the same thread that the wayland event loop is on and cause a timeout.

void-linux, wlroots based compositor, [latest master](https://github.com/dune3d/dune3d/commit/d74b1b8174a9d9bd610be07d1bd5e585039c9373), blabla

Peta-T, 2025-03-14:
I have imported with Opencascade 8.9

[image]

ii8, 2025-03-15:
I think that is because you have a faster machine than me, so it loads before the timeout.

Peta-T, 2025-03-15:
[image]

ii8, 2025-03-15:
Yea your PC is much faster than mine, you can try the other method of producing the bug described in the OP, except make the number 20 higher until it gets really slow, eventually you should see the crash. Obviously you have to be running on wayland natively too.