dune3d/dune3d #158: [Nitpick] Some icons are missing on Mac

viernullvier, 2025-03-10:
Nothing workflow-breaking, but since Dune3D is getting more and more mature, it might be worth addressing.

Some (but not all) icons are not displayed on MacOS, a placeholder icon is used instead. Here are the ones I could identify so far:
 - Create new document
 - Undo
 - Redo
 - "New here?"
 - Customize color theme

[image]

I was able to reproduce this both on bare-metal and nix-based builds. Could it just be a matter of a missing dependency?

carrotIndustries, 2025-03-10:
That's far from nitpicking! You're probably missing the adwaite icon theme. On Arch, it's called adwaita-icon-theme.

Re mac: is that the only thing that's broken so far?

viernullvier, 2025-03-11:
I confirmed that `adwaita-icon-theme` was already installed via brew, but even after recompiling the icons were still missing. It's a bit strange though that _some_ icons do work after all. In my screenshot, `dialog-information-symbolic` and `document-open-symbolic` are displayed correctly, while `document-new-symbolic` and `help-browser-symbolic` are missing.

I'm not sure if this is a coincidence, but the working icons are also referenced from code (`set_icon_name` / `set_from_icon_name` / `set_image_from_icon_name`), while the missing icons are only referenced in `window.ui` and `window.cmb`.

As for the Mac build in general, I'm considering it very stable these days. Haven't seen a crash in a long time, and also no further unexpected behaviors. I think it's reasonable to start looking into creating a distributable app bundle.

carrotIndustries, 2025-03-11:
If you add some code that sets the icon name explicitly, does this make the icons appear?
You could also use the mac equivalent of `strace` to figure out where it's looking for icons.

skurfuerst, 2025-03-15:
Hey,

FYI: for me the icons work nicely on Mac OS - see screenshot:

[image]

All the best,
Sebastian
PS: I followed the instructions for OSX with homebrew, and this worked for me...

viernullvier, 2025-03-15:
Alright, so this issue is getting weirder the more I investigate and I'm starting to suspect that it's just my system that is cursed. Loading the icons from code was a dead end, after all some of the working icons were in fact *not* referenced from code, while loading the missing icons from code did not make them appear.

I found out though that all icons are in fact working when running the binary with `sudo`. I double-checked the permissions of my `adwaita-icon-theme` installation only to find out that all file permissions are *identical*, there's no difference in permissions for working and non-working icons.

It's a bit strange that I've been seeing the same issue on two different machines (Intel and M3), but it's working just fine for others. I guess the issue can be closed for now.

viernullvier, 2025-03-28:
After all, the issue was caused by my local nix installation overriding `$XDG_DATA_DIRS` without including the (working) defaults. Adding the right path (`$HOMEBREW_PREFIX/share` for homebrew) back into the environment fixed it.

closed 2025-03-28 by viernullvier